Joshua-Riek/ubuntu-rockchip #237: Any support for PREEMPT-RT kernel ?

anslex, 2023-06-30:
Hello,

It's more a question rather than an issue. 
How can one compile RT image ? 

Thanks!

Joshua-Riek, 2023-06-30:
I don't know what an RT kernel is. Real-Time?

Joshua-Riek, 2023-07-02:
I looked a little into Real-Time kernels, and this is not something I would work on. The current kernel is a mess forked from Android. You will need to wait for the mainline kernel to be fully supported before someone can attempt this with reasonable effort.

closed 2023-07-02 by Joshua-Riek

anslex, 2023-07-06:
Hi, 
I totally understand, thank you for your response!
I will try to install RT patch when I receive the hardware

Best,

wanyingwe, 2024-09-24:
> Hi, I totally understand, thank you for your response! I will try to install RT patch when I receive the hardware
> 
> Best,

Hello, Did you succeed? I have the same question.

MattiaGrigoli, 2024-10-31:
@anslex have you done it? i'm working on it and i'm not finding anything that works