Joshua-Riek/ubuntu-rockchip #238: Orange Pi 5 Plus PWM Enabled by default?

smjltd, 2023-07-01:
As per the documentation below

[orangepiwiki](http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_5_Plus#PWM_test_method)

it suggest to enable pwm by using `sudo orangepi-config`

is there a way to enable pwm, i can see there are overlays present in this path.
`ls /boot/firmware/dtbs/overlays`

but missing more than pwmchip
`ls /sys/class/pwm/`

Joshua-Riek, 2023-07-01:
I don't have a config tool like orangepi-config yet. You must specify the device tree overlays related to PWM in the file /boot/firmware/ubuntuEnv.txt. Please see the below wiki page for an example and more information.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

ferrellsl, 2023-07-01:
Correct me if I'm wrong, but isn't the orangepi-config just a script? If so, it should be trivial to copy or replicate it. 

 
 

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

closed 2023-07-02 by Joshua-Riek