Joshua-Riek/ubuntu-rockchip #239: Boot on usb

Gazgiza, 2023-07-01:
is it possible to boot ubuntu on an usb on my orange pi 5?

Joshua-Riek, 2023-07-01:
Yes, you can boot from a USB device. But it does not work with USB to SSD converters. 

Look at the wiki on how to install the bootloader to the SPI NOR flash, which will allow you to boot from a USB or NVMe device.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5#install-the-bootloader-to-the-spi-nor-flash

closed 2023-07-02 by Joshua-Riek

bagong, 2023-11-15:
Is it possible that USB comes after NVME in the boot sequence? I can't get it to boot from USB (and have a working nvme install which would be quite a hassle to remove...). It does seem logical to me that as the removable drive USB should come before NVME (and after SD). Is there a way to change the boot sequence, e.g. by editing ubuntuEnv.txt?