Joshua-Riek/ubuntu-rockchip #243: Updated the kernel in version 1.18 and the 1.19 version is quite different.

ironiris, 2023-07-03:
Updated the kernel in version 1.18 and the 1.19 version is quite different.

If OPI5Plus upgraded the kernel to version 1.18, you should see that 
From the beginning of the installation, in the Settings - Audio tab - HDMI-In Audio is enabled for Mic/Aux Audio in the Global Audio Devices list.
This means that OBS audio mixer will be able to monitor OPI5Plus's microphone/Aux and record without any configuration.
However, in 1.19, it is possible to record HDMI Input via Gstreamer Source, but it is not visible in the settings or in the Audio Mixer.
There is a difference.

Also, when streaming YouTube in version 1.18, the speed is initially around 40 fps, but it soon picks up to 50 fps or more, whereas in version 1.19 it stays around 40 fps.
I don't know if this is the cause, but when streaming videos, 1.19 is constantly skipping a few seconds after a certain amount of time.

Joshua-Riek, 2023-07-03:
On a fresh install of v1.18 there are no GPU-related package updates, so performance should be the same with chromium and YouTube. Maybe a program like tracker miner fs is running in the background, causing fps issues using many CPU resources.

```
ubuntu@ubuntu-desktop:~$ sudo apt update
[sudo] password for ubuntu: 
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease           
Hit:4 https://ppa.launchpadcontent.net/jjriek/rockchip/ubuntu jammy InRelease  
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease            
Hit:6 https://ppa.launchpadcontent.net/liujianfeng1994/panfork-mesa/ubuntu jammy InRelease
Hit:7 https://ppa.launchpadcontent.net/liujianfeng1994/rockchip-multimedia/ubuntu jammy InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
30 packages can be upgraded. Run 'apt list --upgradable' to see them.
ubuntu@ubuntu-desktop:~$ apt list --upgradable 
Listing... Done
accountsservice/jammy-updates,jammy-security 22.07.5-2ubuntu1.4 arm64 [upgradable from: 22.07.5-2ubuntu1.3]
bind9-host/jammy-updates,jammy-security 1:9.18.12-0ubuntu0.22.04.2 arm64 [upgradable from: 1:9.18.12-0ubuntu0.22.04.1]
bind9-libs/jammy-updates,jammy-security 1:9.18.12-0ubuntu0.22.04.2 arm64 [upgradable from: 1:9.18.12-0ubuntu0.22.04.1]
cups-bsd/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-client/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-common/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 all [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-core-drivers/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-daemon/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-ipp-utils/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-ppdc/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
cups-server-common/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 all [upgradable from: 2.4.1op1-1ubuntu4.2]
cups/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
gdm3/jammy-updates 42.0-1ubuntu7.22.04.3 arm64 [upgradable from: 42.0-1ubuntu7.22.04.2]
gir1.2-accountsservice-1.0/jammy-updates,jammy-security 22.07.5-2ubuntu1.4 arm64 [upgradable from: 22.07.5-2ubuntu1.3]
gir1.2-gdm-1.0/jammy-updates 42.0-1ubuntu7.22.04.3 arm64 [upgradable from: 42.0-1ubuntu7.22.04.2]
libaccountsservice0/jammy-updates,jammy-security 22.07.5-2ubuntu1.4 arm64 [upgradable from: 22.07.5-2ubuntu1.3]
libcups2/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
libcupsimage2/jammy-updates,jammy-security 2.4.1op1-1ubuntu4.4 arm64 [upgradable from: 2.4.1op1-1ubuntu4.2]
libgdm1/jammy-updates 42.0-1ubuntu7.22.04.3 arm64 [upgradable from: 42.0-1ubuntu7.22.04.2]
libmm-glib0/jammy-updates 1.20.0-1~ubuntu22.04.2 arm64 [upgradable from: 1.20.0-1~ubuntu22.04.1]
libx11-6/jammy-updates,jammy-security 2:1.7.5-1ubuntu0.2 arm64 [upgradable from: 2:1.7.5-1]
libx11-data/jammy-updates,jammy-security 2:1.7.5-1ubuntu0.2 all [upgradable from: 2:1.7.5-1]
libx11-dev/jammy-updates,jammy-security 2:1.7.5-1ubuntu0.2 arm64 [upgradable from: 2:1.7.5-1]
libx11-xcb1/jammy-updates,jammy-security 2:1.7.5-1ubuntu0.2 arm64 [upgradable from: 2:1.7.5-1]
linux-headers-5.10.160-rockchip/jammy 5.10.160-5 arm64 [upgradable from: 5.10.160-3]
linux-image-5.10.160-rockchip/jammy 5.10.160-5 arm64 [upgradable from: 5.10.160-3]
linux-libc-dev/jammy-updates,jammy-security 5.15.0-76.83 arm64 [upgradable from: 5.15.0-73.80]
modemmanager/jammy-updates 1.20.0-1~ubuntu22.04.2 arm64 [upgradable from: 1.20.0-1~ubuntu22.04.1]
u-boot-orangepi5plus-rk3588/jammy 2017.09+20230525.gitac3b1ca9-2 arm64 [upgradable from: 2017.09+20230525.gitac3b1ca9-1]
ubuntu-drivers-common/jammy-updates 1:0.9.6.2~0.22.04.4 arm64 [upgradable from: 1:0.9.6.2~0.22.04.3]
```

Joshua-Riek, 2023-07-04:
@ironiris I ran the command `sudo alsa force-reload` then opened the sound settings and can see HDMI-In as an input device. I will look into this further, but please try with the command and let me know if it works.

ironiris, 2023-07-05:
> @ironiris I ran the command `sudo alsa force-reload` then opened the sound settings and can see HDMI-In as an input device. I will look into this further, but please try with the command and let me know if it works.

After installing version 1.19, I completed the software update that popped up automatically.
and ran 'sudo alsa force-reload', 'apt update && apt install obs-studio obs-gstreamer'
In the settings window of Ubuntu, I could see that the audio was coming in from the HDMI input,
and the Mic/Aux was also enabled in obs-studio, so I was able to monitor it by default.

This is a very desirable result, and streaming through the orangepi5plus became very smooth, and the skipping during the video disappeared.

Thank you.

Joshua-Riek, 2023-07-05:
I'm glad to hear it's working! I need to figure out how to make the make this default so there is no need for the 'sudo alsa force-reload' command.

closed 2025-02-17 by github-actions[bot]