Joshua-Riek/ubuntu-rockchip #245: Post-setup boot issues

Natisfaction, 2023-07-03:
I just reinstalled the OS, this time with 1.19 version, I did exactly the same setup procedure and went fine, but then when it should reboot to finish the installation and bring me to the Desktop, it just doesn't boot.

Joshua-Riek, 2023-07-03:
Did you manually reboot the system yourself? If so, there could have been cached data not flushed to the disk, leaving essential boot and configuration files missing. Also, what board did you experience this with?

Joshua-Riek, 2023-07-03:
I found a way to ensure all data is written to the disk before the install interface was complete. I will preform some testing tonight to make sure this fixes the issue.

Natisfaction, 2023-07-03:
Sorry for the late response. The OS just rebooted after the final installation cleanup. I'm using OrangePi 5 board.

Joshua-Riek, 2023-07-04:
The system should not reboot after the installation. I updated the flash-kernel package, which is in charge of updating important boot files to sync all data to the disk after installation. Try and re-downloading the latest release; it should work with no issues.

closed 2023-07-04 by Joshua-Riek

Natisfaction, 2023-07-04:
I tried another time, but no success.

Maybe I should've mention that I'm using the OrangePi 5 with the NVMe SSD. Don't know if this matters tho.

Joshua-Riek, 2023-07-04:
I double-checked with a fresh install on my Orange Pi 5 with an NVMe SSD, and the installation works without issue. Make sure you are using a 5v4a power supply. Additionally, the system does not need to be connected to the internet for setup.

Btw there is no need to re-install between v1.17 and v1.19, all important changes can be downloaded with a simple apt update command.

Natisfaction, 2023-07-04:
Perfect, thanks for let me know that. But I did not reinstalled the OS from the image for this reason, I actually did that because when I booted up the OPi5, it just kept rebooting itself for like 5 times, then the desktop was loaded for 1 minute and then kept crashing for no reason.

Joshua-Riek, 2023-07-04:
Weird, are you using a 5v4a power supply? The software side of things is quite stable, this might be a power-related issue.

Natisfaction, 2023-07-04:
Ok, for some strange reason changing my old cable, which was btw 5v4a, made it work... I don't really know why but I'm glad it worked lmao. Thanks so much for your hard work btw, and for the support you provide. This release of Ubuntu is really stable, and makes a really good alternative to other OPi5 supported OS (which I pretty much don't really like, especially the official Ubuntu for OrangePi 5. I really appreciate that, thank you!

Joshua-Riek, 2023-07-04:
Haha, I'm glad it's working now! Thanks, I'm happy to see people enjoying my work; I have learned quite a lot from this project so far and only plan to improve for a better user experience.

Natisfaction, 2023-07-04:
I know this isn't the right section to talk about this, but in older versions I noticed smt strange: when I've got Wayland enabled, some programs works unexpectedly. Then when it's disabled those programs works perfectly but having multiple instances of them opened at the same time causes tons of performance decreasing. This is reproducible by disabling Wayland and then opening 2 instances of chromium. If you want I can open another issue, but before doing then I'll personally check tomorrow if now it's that way too.

Joshua-Riek, 2023-07-04:
Wayland has the best performance right now with the GPU drivers and is the most stable. Disabling it and moving to x11 would introduce many issues and bugs. I would test the most recent image and see if this also occurs. However, I don't think this is something I can fix, It sounds like a bug with wayland or the GPU drivers.

Natisfaction, 2023-07-05:
Oh ok, that's not a problem btw, so don't worry too much about that. Thanks so much for the support you provide. Have a good day!