Joshua-Riek/ubuntu-rockchip #250: M.2 Crashes

pbattin, 2023-07-06:
Have an Orange Pi 5 Plus and I cannot get it to boot from the m.2 drive. 

During start it freezes and crashes a few times before loading the set up screen. But then it will also crash when setting up.

If it gets passed set up phase, once ot restarts it will not boot properly and is showing errors on boot. 

The SD card flashed the same way loads just fine though. Tried different ways to install it and no luck. 

Any ideas what's special about installing the image on a m.2?

Joshua-Riek, 2023-07-06:
It sounds like your board might not be getting enough power. Are you using a 5v 4a power supply?

pbattin, 2023-07-06:
> It sounds like your board might not be getting enough power. Are you using a 5v 4a power supply?

Ahh, seems like I'm only using a 3.6A power supply right now. I'll try one with more than 4A

pbattin, 2023-07-06:
> > It sounds like your board might not be getting enough power. Are you using a 5v 4a power supply?
> 
> Ahh, seems like I'm only using a 3.6A power supply right now. I'll try one with more than 4A

Updated the power supply and it resolved my issues it seems. Thanks for the quick reply!

closed 2023-07-06 by pbattin

Joshua-Riek, 2023-07-06:
Awesome! Glad to hear it's working!