Joshua-Riek/ubuntu-rockchip #252: [Orange Pi 5+]  HDMI IN - Not a video capture device

ggdk123, 2023-07-08:
Testing the HDMI In with ffmpeg on ubuntu server. Below is the error message.

```
ubuntu@ubuntu:~$ sudo v4l2-ctl --list-devices
rk_hdmirx (fdee0000.hdmirx-controller):
        /dev/video0

ubuntu@ubuntu:~$ ffmpeg -i /dev/video0 ~/output.mkv
ffmpeg version 4.4.2-0ubuntu0.22.04.1+rkmpp20230327 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.3.0-1ubuntu1~22.04)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1+rkmpp20230327 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --arch=arm64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libdrm --enable-rkmpp --enable-version3 --disable-libopenh264 --disable-vaapi --disable-vdpau --disable-decoder=h264_v4l2m2m --disable-decoder=vp8_v4l2m2m --disable-decoder=mpeg2_v4l2m2m --disable-decoder=mpeg4_v4l2m2m --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[video4linux2,v4l2 @ 0xaaaae42d1930] Not a video capture device.
/dev/video0: No such device
```

Joshua-Riek, 2023-07-09:
You need to install some packages for /dev/video0 to appear. Try to install the below packages then reboot your system.

```bash
sudo apt-get update
sudo apt-get install -y mali-g610-firmware rockchip-multimedia-config
```

ggdk123, 2023-07-10:
> You need to install some packages for /dev/video0 to appear. Try to install the below packages then reboot your system.
> 
> ```shell
> sudo apt-get update
> sudo apt-get install -y mali-g610-firmware rockchip-multimedia-config
> ```

It seems i'm getting the same error after having installed these 2 packages.

Joshua-Riek, 2023-07-10:
Can you share the output of `ls -la /dev/video0`?

ggdk123, 2023-07-10:
> Can you share the output of `ls -la /dev/video0`?

```
ubuntu@ubuntu:~$ ls -la /dev/video0
crw-rw---- 1 root video 81, 0 Jul 10 16:07 /dev/video0
```

ggdk123, 2023-07-11:
tried the gstreamer example script, below is the output

```
ubuntu@ubuntu:~$ ./testhdmiin.sh
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstAudioSrcClock
Redistribute latency...
Redistribute latency...
Redistribute latency...
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 11968' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
Redistribute latency...
Redistribute latency...
WARNING: from element /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: Can't record audio fast enough
Additional debug info:
../gst-libs/gst/audio/gstaudiobasesrc.c(847): gst_audio_base_src_create (): /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0:
Dropped 64386 samples. This is most likely because downstream can't keep up and is consuming samples too slowly.
WARNING: from element /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: Can't record audio fast enough
Additional debug info:
../gst-libs/gst/audio/gstaudiobasesrc.c(847): gst_audio_base_src_create (): /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0:
Dropped 210798 samples. This is most likely because downstream can't keep up and is consuming samples too slowly.
WARNING: from element /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: Can't record audio fast enough
```

ggdk123, 2023-07-11:
> I think this is an issue with using ffmpeg directly

is this an orange pi 5 specific issue? or all the rk3588 boards are all designed to be so? like Radxa ROCK 5B

Joshua-Riek, 2023-07-13:
It looks like the above example failed on your end because you are using the server image, I think a desktop environment needs to be started for it to work.

ggdk123, 2023-07-14:
> It looks like the above example failed on your end because you are using the server image, I think a desktop environment needs to be started for it to work.

You are right. It works now on the desktop. But latency is very visible. In my view, gstreamer is not exactly equivalent to ffmpeg. ffmpeg is absolutely needed for hdimi in on rk3588 board.

Joshua-Riek, 2023-07-14:
I agree, but Rockchip's work on ffmpeg has been abandoned. The open-source community must create a fork to add more support and capabilities. There has been some work already done, but it's not something I have looked at yet https://github.com/Joshua-Riek/ubuntu-rockchip/issues/246.

closed 2023-08-20 by Joshua-Riek

StuartIanNaylor, 2023-10-23:
https://github.com/rigaya/rkmppenc

bagong, 2023-10-30:
"Recording" with "ffmpeg -i /dev/video0 ./output.mkv" and "playback" with "mpv /dev/video0" work right away (2023/10/30 (latest of everything, but not yet rl 1.29)), which is way cool! I'd love to use it in OBS or VLC, but seem not to be able to do what can be done with ffmpeg and mpv...

Joshua-Riek, 2023-10-30:
OBS should work fine to record, but to get OBS to work properly there are a few extra steps. If I recall some env vars need to be set and extra packages. But that's great to hear!

bagong, 2023-10-31:
OBS works fine if you install qt6 platform plugin (hidden in the package qt6-wayland, and yes, qt6, not 5 as everybody assumes ;-) ).

Problem is I see nothing if I define a v4l2 capture device, although OBS does seem to see the hdmiin device:

````
$: v4l2-ctl --list-devices
rk_hdmirx (fdee0000.hdmirx-controller):
	/dev/video
````

And this is how OBS sees the device (without producing an image):

[image]

Is there something I can do to understand the problem, solve it?

bagong, 2023-10-31:
Vlc gives this output at an attempt to play "video capture device" /dev/video0:

````
[0000ffff44001150] v4l2 demux error: cannot select input 0: Inappropriate ioctl for device
Using mplane plugin for capture 
[0000ffff44001150] v4l2 demux error: not a radio tuner device
Using mplane plugin for capture 
[0000ffff440033c0] v4l2 stream error: cannot select input 0: Inappropriate ioctl for device
````

mpv also complained about "Inapropriate ioctl for device" but still produced a video then ;-)

StuartIanNaylor, 2023-10-31:
With `rkmppenc --input-format v4l2 -i /dev/video0 \
  --input-option channel:0 --input-option ignore_input_error:1 --input-option ts:abs \
  --audio-source "hw:<n>:format=alsa/codec=aac;enc_prm=aac_coder=twoloop;bitrate=192" \
  -o out.ts`
  
  Does it work because they have included `--input-option ignore_input_error:1`

bagong, 2023-11-02:
It doesn't work for me, the out.ts isn't written (I didn't know how to stop the recording, so I hit ctrl-c, maybe that was wrong). I did however try to identify the term that makes "Inappropriate ioctl for device" go away, and it's not the "ignore_input_error" but "input-option channel:0".

esolk, 2024-01-15:
> "Recording" with "ffmpeg -i /dev/video0 ./output.mkv" and "playback" with "mpv /dev/video0" work right away (2023/10/30 (latest of everything, but not yet rl 1.29)), which is way cool! I'd love to use it in OBS or VLC, but seem not to be able to do what can be done with ffmpeg and mpv...

I'm running on Ubuntu22.04 and I did 'apt upgrade', but still get `Not a video capture device.` error while running FFmpeg. MPV shows the same errro. Gstreamer can work with laggy.
How do you make ffmpeg and mpv work?

StuartIanNaylor, 2024-01-17:
Anyone tried the Gstreamer input for Obs instead?
https://github.com/fzwoch/obs-gstreamer

Retsel023, 2024-01-23:
i got my hdmi input to work with ffmpeg. I tweaked it for my nintendo switch. ffmpeg -f v4l2 -video_size 1920x1080 -i /dev/video0 -pix_fmt bgr24 -vf "format=rgb24" -framerate 80 -video_size 1920x1080 -c:v rawvideo -vsync 1 -f sdl "SDL output"

It took me a couple of days to get any input at all. I gave up on OBS gsteamer and the add capture device option. Only black screen and format errors. I might be able to fix the gsstream errors now with the knowledge i have now but i like the ffmpeg so thats fine. Low latency and decent quality. Note, i do not record my screen i merely wanted to display it on the orangepi and maybe stream it. 
[image]

N0tiK44, 2024-03-18:
> i got my hdmi input to work with ffmpeg. I tweaked it for my nintendo switch. ffmpeg -f v4l2 -video_size 1920x1080 -i /dev/video0 -pix_fmt bgr24 -vf "format=rgb24" -framerate 80 -video_size 1920x1080 -c:v rawvideo -vsync 1 -f sdl "SDL output"
> 
> It took me a couple of days to get any input at all. I gave up on OBS gsteamer and the add capture device option. Only black screen and format errors. I might be able to fix the gsstream errors now with the knowledge i have gained but i like the ffmpeg so thats fine. Low latency and decent quality. Note, i do not record my screen i merely wanted to display it on the orangepi and maybe stream it.
> 
> For this to work i used the ubuntu 22.04 server image + kde plasma. installed the mali drivers, the ffmpeg dependencies, some libraries and a lot of troubleshooting/debugging. ![IMG_20240123_210140](https://private-user-images.githubusercontent.com/8491773/299100317-b6e89e61-405e-4593-a3de-6286c0bb96d4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA3Njc0NTMsIm5iZiI6MTcxMDc2NzE1MywicGF0aCI6Ii84NDkxNzczLzI5OTEwMDMxNy1iNmU4OWU2MS00MDVlLTQ1OTMtYTNkZS02Mjg2YzBiYjk2ZDQuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMThUMTMwNTUzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTgwZGRiZThhNDc0M2Y1YTk3YzFjMzM5N2U1YzRlMTFlNTAwNDk5NGM0NTA4NDgyZjE3ZjU3MDZiOTVkZjg0OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.fBlkCOfeyFokGlX1otLVU9llRqqIZ9r51HTA05ZPehQ)

What were you stats for specifically latency?

kurkovpavel, 2025-08-22:
If you just want to capture hdmi rx for your project (making own code), you can use ioctl for getting low latency (about 30ms at 60Hz) video. example:  [https://github.com/kurkovpavel/orangepi5plus_hdmi_rx](url)

kchicagmb, 2025-10-15:
> ioctl

@kurkovpavel I use your example but I have the next:

./hdmi_capture 
Driver: rk_hdmirx
Card: rk_hdmirx
Bus info: fdee0000.hdmirx-controller
Capabilities: 0x84201000
Current format: BGR3 (3840x2160)
Pixel format: 33524742
Field: 1
Colorspace: 8
Unsupported pixel format: 0x33524742
Failed to capture frame

What version of OS do you have?

kurkovpavel, 2026-01-14:
> Your ffmpeg is too old, apply these patches to support rockchip hdmirx:
>
>    -
>    https://github.com/FFmpeg/FFmpeg/commit/7eb37c142a92b27b0d5f66f9d6ad5905085daa8e.patch
>    -
>    https://github.com/FFmpeg/FFmpeg/commit/cc6fb1643d7e14c6f76a48e0cffad96394cb197c.patch
>
> If your hdmirx driver was from SDK, apply this patch:
>
>    -
>    https://github.com/istoreos/istoreos/raw/refs/heads/istoreos-22.03/target/linux/rockchip/rk35xx/patches-5.10/1030-hdmirx-ioctl.patch

Are you trying this code?  kurkovpavel/orangepi5plus_hdmi_rx: Capturing
from hdmi rx input to cpp pipeline without external libraries
<https://github.com/kurkovpavel/orangepi5plus_hdmi_rx>

It's supports NV24 only,
For RGB you can use this one:
kurkovpavel/orangepi5plus_hdmi_rx_to_framebuffer
<https://github.com/kurkovpavel/orangepi5plus_hdmi_rx_to_framebuffer> (for
OS without GUI)

or add these cases to switch in camera_hdmi.c:
        case V4L2_PIX_FMT_RGB24: {
            cv::Mat rgb(height, width, CV_8UC3, yuv_data);
            cv::cvtColor(rgb, frame, cv::COLOR_RGB2BGR);
            result = true;
            break;
        }
        case V4L2_PIX_FMT_BGR24: {
            frame = cv::Mat(height, width, CV_8UC3, yuv_data);
            result = true;
            break;
        }

I have Linux orangepi5plus 6.1.43-rockchip-rk3588