Joshua-Riek/ubuntu-rockchip #253: Emergency mode with Ubuntu 22.04 on Rock 5B

petewarden, 2023-07-08:
I'm using the Ubuntu 22.04 server image on a Rock 5B, and I've been getting a lot of boots into emergency mode. The system first spends 90 seconds displaying "Starting task 2/5: Setting kernel variables" or something similar, and then tells me I'm in emergency mode. Unfortunately at that point it's not responding to my USB keyboard (I've tried different USB sockets and keyboards) so I can't run any of the FS checking code it recommends. Funnily enough I can press ctrl+alt+del earlier in the process to start a shutdown, so there's some communication with the keyboard. I'm also unable to boot into bios or grub by holding down shift or pressing escape early in the boot, though I'm unsure exactly what the expected keypresses for that kind of mode are (my HDMI display only turns on a few seconds in).

Also oddly, I'm able to boot successfully sometimes, without changing anything. I don't have any peripherals connected, I've tried booting with nothing connected via USB and I still see this occurring. I'm also not able to ssh into the board, presumably because the ssh daemon hasn't been started by the time it fails.

I'm not expecting a fix based on these vague symptoms, but any help you can give me in debugging this would be very much appreciated. Is there a magic key combo I should be using to boot into some kind of safe mode so I can check the file system, for example, or any other way I can interact with the system to get more information when it's wedged like this?

Thanks for any ideas!

Joshua-Riek, 2023-07-08:
I have not encountered this before. But since it seems to be random to a point, my first guess is a power issue. Are you using a 5v 4a power supply? If so try a different USB-C cable or a different 5v 4a power supply.

petewarden, 2023-07-10:
Thanks @Joshua-Riek! I've had problems in the past with underpowered power supplies, and I thought I had a beefy enough one, but checking the fine print it's 5v 3a, not 5v 4a. I've ordered some 5v 3a versions.

Having said that, I was able to get the system to recognize my USB keyboard intermittently, which let me debug a bit deeper. I suspect the root cause *might* be related to a problem mounting the `/boot/firmware` volume. When running `journalctl -b` I see `Timed out waiting for device /dev/disk/by-uuid/C357-CDA8`. When I try `mount /boot/firmware /dev/disk/by-uuid/C357-CDA8` from the emergency mode prompt it works with no delay.

There were also some other jobs (multipathd, apply kernel variables) that were timing out, but I now suspect they were depending on `/boot/firmware` in systemd. For completeness, here's what I tried to sort out various errors I saw in the boot logs:

`systemctl disable multipathd.service`
`sudoedit /etc/sysctl.d/10-ptrace-conf`
Comment out `kernel.yama.ptrace_scope = 1` at the end and save the file.

I also tried disabling the `fsck` on that disk in `/etc/fstab`, but that didn't seem to make a difference. I ran `fsck` manually on `/boot/firmware` from emergency mode once it was manually mounted, it found a dirty bit (because I couldn't shut the board down cleanly, only by pulling the power) but otherwise reported no errors.

Anyway, no action requested, I just wanted to leave a trail of breadcrumbs in case anyone else hits a similar issue. I will update when I've tried the new power supply too. My board has booted cleanly the last few times, so I can no longer reproduce the issue, but I'm sure it will come back, so any suggestions on things to look for are very welcome, as always.

Joshua-Riek, 2023-07-10:
I have 11 RK3588 boards, and all of them have power-related quirks. The Rock 5B is the worst offender, in this case, you must use a fixed 5v 4a power supply with the Rock 5B. 

I strongly think this is a power-related issue, or you have a corrupted OS image.

petewarden, 2023-07-10:
Thanks again @Joshua-Riek, I appreciate you sharing your deep experience with these boards! I'm glad you got me to check my assumptions that a PD-rated power supply would be good enough, and I'm hopeful the new ones I've ordered will help.

closed 2023-07-13 by Joshua-Riek

petewarden, 2023-07-14:
Just to update this thread (again no action needed) but the new power supplies haven't alleviated the issue, we're still seeing it frequently.

One thing I'm now trying to figure out is how to set `grub` configuration parameters, or any other options I have while booting. I hope that I can generate more debug logging information to help track this down. I'm trying to follow the [Ubuntu guide](https://wiki.ubuntu.com/Kernel/KernelBootParameters) but unfortunately I'm not seeing a grub menu when I hold down shift while rebooting, and `/etc/default/grub' doesn't exist, so I'm open to ideas on how to affect the boot process. Apologies for the noob questions, I haven't done this level of debugging before. Thanks for all your help so far.

Joshua-Riek, 2023-07-14:
The grub efi in theory, could be loaded by u-boot. There is support for loading and executing efi applications. However, in my experience, this can introduce weird bugs, such as reboot issues where the system can not properly power cycle. I would recommend staying with u-boot if you prefer reliability and stability.  

Have you hooked up a serial console for debugging the boards? You can post the output here, and I can take a look. 

I do have a random thought about your issue previously mentioned. Since it seems related to some sort of filesystem issue, I recall letting the cloud-init package resize the filesystem during the boot process to fill the remaining disk space with the root filesystem. This is only the case on the ubuntu server images, not the desktop images I have available. Maybe try and run a few desktop images and see if you still have an issue. If the issue looks like it does not happen on the desktop image, we might have found the problem.

petewarden, 2023-07-15:
Thanks @Joshua-Riek, that gave me some good leads to try! I did [nuke cloud-init](https://gist.github.com/zoilomora/f862f76335f5f53644a1b8e55fe98320) and snap while I was at it, since I saw some odd log messages from that. I've confirmed that cloud-init no longer shows up in the logs, so it seems to be gone, but I still see the boot issues.

I didn't actually know uboot was being used instead of grub, though reading up it makes sense. That probably demonstrates my low level of understanding of the booting process, so apologies again for asking such basic questions, but is there a file I can edit to modify the boot command line parameters? Reading [docs](https://u-boot.readthedocs.io/en/latest/develop/devicetree/control.html) seems to indicate I should look for a `.dtb` file that I can edit, but I'm not finding one in `/boot/` or anywhere else.

petewarden, 2023-07-15:
Actually, I just realized I had a mental block because `/boot/firmware` wasn't mounted, but `/boot/firmware/ubuntuEnv.txt` has what I need. I'm still not sure how the file is read when the folder isn't mounted yet, but I'll happily trust the Linux gods if it does work when I change it.

petewarden, 2023-07-15:
Once I learned more about the [magic behind `/boot/firmware`](https://0pointer.net/blog/linux-boot-partitions.html) (my basic understanding is the UEFI has its own stack to read the contents at the start of boot, and the later mounting of it in the main filesystem is more of a convenience for updating than anything that's required for boot) I was able to substantially reduce the issues by just adding `nofail` to the `/etc/fstab` options for the `system-boot` device.

The system still frequently fails to mount the device, but it's no longer blocking the boot, and my applications are able to run without `/boot/firmware` mounted. I'm aware this isn't a long-term solution, and I've still seen one hung reboot (on the "Apply Kernel Variables" task) so I'm not out of the woods, but at least it's unblocked our prototyping.

I'll continue to dig in when I get a chance and update this thread with anything relevant I discover. Thanks again for all your help @Joshua-Riek!

Joshua-Riek, 2023-07-15:
There is a UEFI edk2 port that you might want to look at. Its still very much a work in progress, but might be exactly what you are looking for.
https://github.com/edk2-porting/edk2-rk3588