Joshua-Riek/ubuntu-rockchip #254: Camera ISP Not working On Orange Pi 5B

peytonicmaster6, 2023-07-10:
I am using the official Orange Pi 5 [camera module](https://www.amazon.com/Orange-Pi-Camera-Compatible-Computers/dp/B0BYP55F15) with an Orange Pi 5B and it appears that the ISP is not working properly. The camera feed is entirely green and noisy. On the Orange Pi Arch Linux image, the same camera has a normal looking camera feed and appears to have the ISP running properly. I'm using the exact same gstreamer command as the Arch Linux image, which looks like the following (it's pulled from the test_camera.sh script 

`gst-launch-1.0 v4l2src device=/dev/video42 io-mode=4 ! video/x-raw,format=NV12,width=720,height=576,framerate=15/1 ! xvimagesink > /dev/null 2>&1`

Also, the issue is present if I use OpenCV and python like below, so I think its an issue with the ISP.
```
import cv2

vid = cv2.VideoCapture(42)
while(True):
    ret, frame = vid.read()
    cv2.imshow('frame', frame)
    
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

vid.release()
cv2.destroyAllWindows()
```

peytonicmaster6, 2023-07-14:
I checked the kernel logs and there's one line that appears to be the difference between the ISP running versus not running. On the Arch Linux image, before the camera starts streaming, this line is present in the logs

`rkcif-mipi-lvds2: sditf_reinit_mode, mode->rdbk_mode 0, mode->name rkisp0-vir1, link_mode 1`

On your image, this line is missing from the logs. Also when the system boots on your image, there's a few additional lines in the logs.
```
rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
rkcif_scale_set_fmt: req(80,60) src out(0,0)
rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
rkcif_scale_set_fmt: req(80,60) src out(0,0)
rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
rkcif_scale_set_fmt: req(80,60) src out(0,0)
rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
rkcif_scale_set_fmt: req(80,60) src out(0,0)
```
The `rkcif_scale_set_fmt: req(80,60) src out(0,0)` lines are not present on the Arch Linux image. I also tried using the Orange PI 5 overlay rather than the 5B on your image since that's what the Arch Linux image uses, but the results were the same.

0312birdzhang, 2023-09-06:
camera not working on Orange Pi5 too

dmesg when use `gst-launch-1.0 v4l2src device=/dev/video11 io-mode=4 ! video/x-raw,format=NV12,width=720,height=576,framerate=15/1 ! xvimagesink > /dev/null 2>&1` command
```
[  176.513172] stream_cif_mipi_id0: open video, entity use_countt 1
[  176.513560] stream_cif_mipi_id0: close video, entity use_count 0
[  176.613601] stream_cif_mipi_id1: open video, entity use_countt 1
[  176.613952] stream_cif_mipi_id1: close video, entity use_count 0
[  176.808576] stream_cif_mipi_id2: open video, entity use_countt 1
[  176.808897] stream_cif_mipi_id2: close video, entity use_count 0
[  176.908766] stream_cif_mipi_id3: open video, entity use_countt 1
[  176.909098] stream_cif_mipi_id3: close video, entity use_count 0
[  177.598426] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  177.599386] stream_cif_mipi_id0: s_power 0, entity use_count 0
[  177.698797] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  177.700029] stream_cif_mipi_id0: s_power 0, entity use_count 0
[  177.797054] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  177.797984] stream_cif_mipi_id0: s_power 0, entity use_count 0
[  177.895759] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  177.896925] stream_cif_mipi_id0: s_power 0, entity use_count 0
[  177.996524] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  177.997443] stream_cif_mipi_id0: s_power 0, entity use_count 0
[  178.097575] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  178.098293] stream_cif_mipi_id0: s_power 0, entity use_count 0
[  178.221195] stream_cif_mipi_id0: s_power 1, entity use_count 1
[  178.236603] rkisp_hw fdcb0000.rkisp: set isp clk = 300000000Hz
[  178.251766] rkcif-mipi-lvds: stream[0] start streaming
[  178.251839] rockchip-mipi-csi2 fdd10000.mipi0-csi2: stream on, src_sd: 00000000b083bb71, sd_name:rockchip-csi2-dphy0
[  178.251867] rockchip-mipi-csi2 fdd10000.mipi0-csi2: stream ON
[  178.251894] rockchip-csi2-dphy0: dphy0, data_rate_mbps 600
[  178.252165] rockchip-csi2-dphy csi2-dcphy0: csi2_dphy_s_stream stream on:1, dphy0
[  178.252171] rockchip-csi2-dphy csi2-dcphy0: csi2_dphy_s_stream stream on:1, dphy0
```

0312birdzhang, 2023-09-11:
Just find this https://forum.radxa.com/t/how-to-use-radxa-camera-4k/13031/2 

Will test it then

Joshua-Riek, 2023-09-11:
I updated the camera software packages with commit d206c00d1af117b65d82f7cbf24077423c992c5a and enabled rkisp2 in the kernel. I have the kernel building now on my PPA. When it is finished update your system and enter the below commands to install the new rkaiq and rkisp packages.

```
sudo apt-get update
sudo apt-get upgrade -y
wget https://media.githubusercontent.com/media/Joshua-Riek/ubuntu-rockchip/main/packages/rkaiq/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb?download=true
wget https://media.githubusercontent.com/media/Joshua-Riek/ubuntu-rockchip/main/packages/rkaiq/camera_engine_rkaiq_rk3588_update_arm64.deb?download=true
wget https://media.githubusercontent.com/media/Joshua-Riek/ubuntu-rockchip/main/packages/rkaiq/camera_engine_rkisp-v2.2.0_arm64.deb?download=true
sudo dpkg -i camera_engine_rkaiq_rk3588_1.0.3_arm64.deb
sudo dpkg -i camera_engine_rkaiq_rk3588_update_arm64.deb
sudo dpkg -i camera_engine_rkisp-v2.2.0_arm64.deb
```

peytonicmaster6, 2023-09-12:
Hey, thanks for pushing this fix with the recent commit! Unfortunately, I'm not able to get it working on the Orange Pi 5B. I did a fresh install of release version 1.24 and followed the steps above, and I now have rkisp_3A.service and rkaiq_3A.service, but they are not working properly.

Here's the output of rkisp_3A.service when I try and start it
```Sep 12 00:10:43 jarvis-desktop systemd[1]: Starting rkisp 3A engine...
Sep 12 00:10:43 jarvis-desktop systemd[1]: Started rkisp 3A engine.
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: stream_cif is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: stream_cif_dvp is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: rkisp1_mainpath is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: rkisp1_selfpath is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: rkisp1-isp-subdev is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: rkisp1-input-params is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: media get entity by name: rkisp1-statistics is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]: ERR: Bad media topology
Sep 12 00:10:43 jarvis-desktop rkisp_3A-1[1956]:  error 22, Invalid argument
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: stream_cif is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: stream_cif_dvp is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: rkisp1_mainpath is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: rkisp1_selfpath is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: rkisp1-isp-subdev is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: rkisp1-input-params is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: media get entity by name: rkisp1-statistics is null
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]: ERR: Bad media topology
Sep 12 00:10:43 jarvis-desktop rkisp_3A-4[1962]:  error 22, Invalid argument
Sep 12 00:10:43 jarvis-desktop rkisp_3A.sh[1966]: rkisp_3A_server: no process found
Sep 12 00:10:43 jarvis-desktop systemd[1]: rkisp_3A.service: Deactivated successfully.
```

Similarly, here's the output of rkaiq_3A.service when trying to start it
```
Sep 12 00:28:24 jarvis-desktop systemd[1]: Starting Enable Rockchip camera engine rkaiq...
Sep 12 00:28:24 jarvis-desktop systemd[1]: Started Enable Rockchip camera engine rkaiq.
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-isp-subdev is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-input-params is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-statistics is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp_mainpath is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: Cound not find rkisp dev names, skipped /dev/media0
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: ERR: Bad media topology for: /dev/media0
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-isp-subdev is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-input-params is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-statistics is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp_mainpath is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: Cound not find rkisp dev names, skipped /dev/media1
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: ERR: Bad media topology for: /dev/media1
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-isp-subdev is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-input-params is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp-statistics is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: media get entity by name: rkisp_mainpath is null
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: Cound not find rkisp dev names, skipped /dev/media2
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: ERR: Bad media topology for: /dev/media2
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: DBG: get rkisp-isp-subdev devname: /dev/v4l-subdev9
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: DBG: get rkisp-input-params devname: /dev/video41
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: DBG: get rkisp-statistics devname: /dev/video40
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: DBG: get rkisp_mainpath devname: /dev/video33
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: rkaiq log level ff0
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: 
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: ************************** VERSION INFOS **************************
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: version release date: 2023-07-06
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]:          AIQ:       AIQ v5.0x1.3
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]:    IQ PARSER:       Calib v1.4.8,magicCode:1170944
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: ************************ VERSION INFOS END ************************
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: DBG: get rkisp-isp-subdev devname: /dev/v4l-subdev10
Sep 12 00:28:24 jarvis-desktop rkaiq[3080]: DBG: get rkisp-input-params devname: /dev/video50
[... cut ...]
ERR: Bad media topology for: /dev/media10
ERR: Bad media topology for: /dev/media11
ERR: Bad media topology for: /dev/media12
ERR: Bad media topology for: /dev/media13
ERR: Bad media topology for: /dev/media14
ERR: Bad media topology for: /dev/media15
Segmentation fault (core dumped)
```

This looks to be the same log output as [here](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/288#issuecomment-1694578331) in #288. Any ideas as to what might be going on or if I missed something on the installation?

Joshua-Riek, 2023-09-12:
Can you try with one of the desktop images found below? My instructions may be incorrect, but these new images should work just fine.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6155114077

peytonicmaster6, 2023-09-13:
Tried the desktop image from that link and the rkisp_3A.service and rkaiq_3A.service were present, but had the same error messages as before. I'm using the ov13855 camera from orange pi and I'll get a very dark image and the camera will often freeze up so things definitely are still not working properly.

Joshua-Riek, 2023-09-13:
I tested the Orange Pi 5 Plus again and found that the ov13855 works perfectly, but the ov13850 does not. This is very strange.

peytonicmaster6, 2023-09-14:
That's super odd as I have a ov13855 with an Orange Pi 5B and things aren't working for me. Things run fine for me on the Orange Pi OS. Their rockchip_defconfig only contains the following lines in regards to the camera isp
```
CONFIG_VIDEO_ROCKCHIP_CIF=y
CONFIG_VIDEO_ROCKCHIP_RKISP1=y
CONFIG_VIDEO_ROCKCHIP_ISP=y
CONFIG_VIDEO_ROCKCHIP_ISPP=y
```

I tried setting the defconfig with these parameters and recompiling the kernel, but I got the same results. My guess is has to do with the rkaiq and rkisp packages.

Joshua-Riek, 2023-09-14:
I just did some testing on the Orange Pi 5B and i could not get ov13855 or ov13850 working, i wonder if the device tree overlay has some error for the 5B.

peytonicmaster6, 2023-09-25:
I was finally able to get things working! Turns out the segmentation faults are manly related to invalid configuration files, among other random things. I ended up having to recompile rkaiq_3A_server from the source code from [firefly](https://gitlab.com/firefly-linux/external/camera_engine_rkaiq/-/tree/rk3588/firefly), replace the existing rkaiq_3A_server in /bin/ and then things worked for me. It's a slightly older version, AIQ v5.0x1.2-rc4, rather than AIQ v5.0x1.3. but it uses the same version of the tuning file, so the tuning files from the packages you updated work fine!

Here's a link to my [github](https://github.com/peytonicmaster6/camera-engine-rkaiq.git) with the modified source code (firefly has things set-up for cross-compiling and not compiling naively on the rk3588 chips) and the compiled version if you'd like to give it a try.

peytonicmaster6, 2023-12-16:
I've been playing around with different versions of the camera engine for a while now and have finally figured out how to get things working properly! The issues have actually mainly been with rkaiq_3A_server rather than tuning files or the librkaiq.so.

- v5.0x1.3 introduced a change to rkaiq_3A_sever meant to add support for mutli camera systems, but actually produced a bug that broke compatibility on multi camera systems. This is why you haven't had any issues running the ov13855 on the Orange Pi 5 plus, but failed to get it to work on the Orange Pi 5b. I don't have other boards to test, but my hunch is all boards with multiple camera ports have had issues while those with only one port have been fine.
- The rkaiq_3A_sever built from v5.0x1.2-rc4 does not have the multi-camera support bug and works fine with the librkaiq.so file from v5.0x1.3 up through release 1.27 of your Ubuntu image Not sure what exactly causes it to break on newer releases, but it starts complaining about not being able to find any cameras.
- A new release of the library was just pushed last night by Firefly, v5.0x3.0. This is actually 3 versions ahead of the current rkaiq libraries included with your images, which includes some fixes for the multi cam support, among other changes. I compiled this new version, copied librkaiq.so to /usr/lib and rkaiq_3A_server to /usr/bin, rebooted, and everything works perfectly on the newest release of your Ubuntu image, v1.31. I even updated all packages to their newest versions with apt-get update/upgrade, and it still works!

My suggestion would be to include the newest version of rkaiq in your Ubuntu image. I'm not exactly sure how you're integrating it, but [here's](https://gitlab.com/firefly-linux/external/camera_engine_rkaiq/-/commits/rk3588/linux_release_v1.3.0a) the source code of the most recent release. Attached is an instruction guide for building natively on the Orange Pi as the code is setup for cross-compiling.
As for the ov13850 camera problems, I'm pretty sure that's related to the tuning file. The tuning file format is as follows
`{sensor-model}_{camera-name}_{lens-name}.json`
where camera-name and lens-name come from the rockchip,camera-module-name and rockchip,camera-module-lens-name in the dtoverlay settings. 

Based on your linux-rockchip repo, the naming convention for the tuning file should be `ov13850_CMK-CT0116_default.json`, but the tuning file provided is `ov13850_CMK-CT0116_Largan-50013A1.xml`. Changing the name of that file may get things working, but its an xml file. XML files were used with previous ISP versions, so the tuning parameters may be outdated entirely and backwards-compatibility is essentially nonexistent.

In the repo I linked, there's a newer version of an ov13850 file, found in rkaiq/iqfiles/isp21/ called `ov13850_ZC-OV13850R2A-V1_Largan-50064B31.json`. Renaming it to the correct name might work, but image quality may be messed up as its likely for a different camera-lens combo. Unfortunately I don't have a ov13850 to test and confirm, but I know for sure the tuning file name must be changed.

[native_rkaiq.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13694003/native_rkaiq.zip)

peytonicmaster6, 2023-12-16:
Here are the native binaries needed for anyone who just wants to test without having to compile from source.
[v5.0x3.0-binaries.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/13694014/v5.0x3.0-binaries.zip)

1. Place librkaiq.so in /usr/lib
2. Place rkaiq_3A_sever in /usr/bin
3. Reboot

Sunhill666, 2024-03-07:
@peytonicmaster6 Hello, by using the zip file(native_rkaiq.zip) you given, followed the README file and your instruction, renamed `ov013850_ZC-OV13850R2A-V1_Largan-50064B31.json` to `ov13850_ZC-OV13850R2A-V1_default.json`, build up `librkaiq.so` and `rkaiq_3A_server`, and replaced them to corrsponding folder, after doing abovemention works, i reboo system, and check out by using `systemctl status rkaiq_3A.service` command got:

```text
○ rkaiq_3A.service - Enable Rockchip camera engine rkaiq
     Loaded: loaded (/lib/systemd/system/rkaiq_3A.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Thu 2024-03-07 10:27:23 CST; 12min ago
    Process: 626 ExecStart=/etc/init.d/rkaiq_3A.sh start (code=exited, status=0/SUCCESS)
    Process: 634 ExecStop=/etc/init.d/rkaiq_3A.sh stop (code=exited, status=0/SUCCESS)

Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media8
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media9
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media10
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media11
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media12
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media13
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media14
Mar 07 10:27:23 orange-pi rkaiq[628]: ERR: Bad media topology for: /dev/media15
Mar 07 10:27:23 orange-pi rkaiq_3A.sh[638]: rkaiq_3A_server: no process found
Mar 07 10:27:23 orange-pi systemd[1]: rkaiq_3A.service: Deactivated successfully.
```

and `rkaiq_3A_server` command got:

```text
ERR: Bad media topology for: /dev/media0
ERR: Bad media topology for: /dev/media1
ERR: Bad media topology for: /dev/media2
ERR: Bad media topology for: /dev/media3
ERR: Bad media topology for: /dev/media4
ERR: Bad media topology for: /dev/media5
ERR: Bad media topology for: /dev/media6
ERR: Bad media topology for: /dev/media7
ERR: Bad media topology for: /dev/media8
ERR: Bad media topology for: /dev/media9
ERR: Bad media topology for: /dev/media10
ERR: Bad media topology for: /dev/media11
ERR: Bad media topology for: /dev/media12
ERR: Bad media topology for: /dev/media13
ERR: Bad media topology for: /dev/media14
ERR: Bad media topology for: /dev/media15
```

It looks like not working, btw, my platform is Orange Pi 5 Plus and camera is OV13850.

peytonicmaster6, 2024-03-08:
@Sunhill666 my guess is that it's likely a tuning file error. To confirm this, you would want to run the following command and then rerun the rkaiq_3A_server.
```
export persist_camera_engine_log=0xffffffffff
```
This will turn on logging mode and output more useful information than just bad media topology, since that usually appears regardless.

Looking through the iqfiles folder I only see OV13850 files in the isp2x folder, and unfortunately backwards compatibility basically doesn't exist as they tend to rename fields for no good reason. I'm pretty sure I tried copying the OV13850 file and renaming it with the OV13855 convention to test with my camera and it didn't work either. 

What I'd suggest doing is using the OV13855 tuning file found in isp3x to give the correct format and field names and then copying the corresponding information from the OV13850 file into it. Also, if you're using the OV13850 module from Orange Pi, you would want the file to be named `ov13850_CMK-CT0116_default.json`.

improperaffo, 2024-06-14:
I have a similar issue, my camera feed is very dark and greenish. I have an orange pi 5 pro and an OV13855 camera. I followed the steps in https://github.com/Joshua-Riek/ubuntu-rockchip/issues/254#issuecomment-1858937548, but it did not solve my problems. 

If I execute  `rkaiq_3A_server` I get
```
Cound not find rkisp dev names, skipped /dev/media0
ERR: Bad media topology for: /dev/media0
DBG: get rkisp-isp-subdev devname: /dev/v4l-subdev3
DBG: get rkisp-input-params devname: /dev/video19
DBG: get rkisp-statistics devname: /dev/video18
DBG: get rkisp_mainpath devname: /dev/video11
rkaiq log level ffffffffff
XCORE:K:rk_aiq_init_lib, ISP HW ver: 30

************************** VERSION INFOS **************************
version release date: 2023-09-7
         AIQ:       AIQ v5.0x3.0
   IQ PARSER:       Calib v1.4.8,magicCode:1170944
************************ VERSION INFOS END ************************
ERR: Bad media topology for: /dev/media2
ERR: Bad media topology for: /dev/media3
ERR: Bad media topology for: /dev/media4
ERR: Bad media topology for: /dev/media5
ERR: Bad media topology for: /dev/media6
ERR: Bad media topology for: /dev/media7
ERR: Bad media topology for: /dev/media8
ERR: Bad media topology for: /dev/media9
ERR: Bad media topology for: /dev/media10
ERR: Bad media topology for: /dev/media11
ERR: Bad media topology for: /dev/media12
ERR: Bad media topology for: /dev/media13
ERR: Bad media topology for: /dev/media14
ERR: Bad media topology for: /dev/media15
XCORE:E:access /etc/iqfiles//ov13855_CMK-OT2016-FV1_default.json && /etc/iqfiles//ov13855_CMK-OT2016-FV1_default.bin failed!
XCORE:E:_rkAiqManager init error!
CAMHW:E:can't find sensor 
XCORE:E:camhw deinit error -7
IPCSERVER:K:cid[-1] Deinit success
MessageParser process loop exit!
XCORE:K:cid[1] rk_aiq_uapi_sysctl_deinit_locked success.
Segmentation fault (core dumped)
```

Also, `systemctl status rkaiq_3A.service` returns 
```
Unit rkaiq_3A.service could not be found.
```

I'm running Ubuntu 24.04 server `v2.2.1`

Joshua-Riek, 2024-06-20:
Please see #690 for the latest info on adding support for camera devices. It's taking extra time because i'm implementing it properly.

Joshua-Riek, 2024-06-24:
Closing see https://github.com/Joshua-Riek/ubuntu-rockchip/pull/690

closed 2024-06-24 by Joshua-Riek