Joshua-Riek/ubuntu-rockchip #256: Missing sg module

leimera, 2023-07-11:
### Thank you
Lets start of by saying thank you for this project. I am relatively new to linux and am glad there is a familiar ubuntu image for the orange pi5.

### What I am trying to achieve

use the archegon DVD drive to rip DVD in the MakeMKV docker to automate the process. The docker requires the sg? as a device input but it is missing when running the lsblk command. Only the sr0 file for the DVD drive is showing.

### Setup
Orange Pi 5B
Ubuntu Server
Archagon DVD drive
jlesage / docker-makemkv

### Issue
Can't mount an external dvd drive to a docker image because there is no /dev/sg? for the drive. There is only /dev/sr0. W ith a lot of googeling I found using the command "modprobe sg" I get the following output:

modprobe: FATAL: Module sg not found in directory /lib/modules/5.10.160-rockchip

"lsblk" output:

NAME         MAJ:MIN  RM   SIZE RO TYPE MOUNTPOINTS
sda                    8:0        1        0B  0 disk
sdc                    8:32      1 119.1G  0 disk /mnt/sdcard
sr0                    11:0      1 1024M  0 rom
mmcblk0           179:0    0   58.2G  0 disk
├─mmcblk0p1  179:1    0   512M  0 part /boot/firmware
└─mmcblk0p2  179:2    0   57.7G  0 part /
mmcblk0boot0 179:32   0       4M  1 disk
mmcblk0boot1 179:64   0       4M  1 disk

Is the sg module not implemented yet or am I missing something?

Let me know if  if this is clear enough or if I have missed something obvious. Thanks again!!

Joshua-Riek, 2023-07-11:
I took a quick look in the kernel config, and it looks disabled. Let me enable it and make sure the kernel compiles correctly. 
```
#
# SCSI support type (disk, tape, CD-ROM)
#
CONFIG_BLK_DEV_SD=y
# CONFIG_CHR_DEV_ST is not set
CONFIG_BLK_DEV_SR=y
# CONFIG_CHR_DEV_SG is not set
# CONFIG_CHR_DEV_SCH is not set
# CONFIG_SCSI_CONSTANTS is not set
# CONFIG_SCSI_LOGGING is not set
CONFIG_SCSI_SCAN_ASYNC=y
```

Joshua-Riek, 2023-07-11:
I enabled the drivers with the following commit https://github.com/Joshua-Riek/linux-rockchip/commit/4951f03c5e50842980d97c4b7e22f6bef0a34112. I plan to release a kernel update in the next few days, which will be included. However, I can post some instructions on how you can build the kernel yourself and test the changes.

Joshua-Riek, 2023-07-11:
On your Orange Pi 5, please follow the instructions below. This process can take between 20 and 40 minutes. After it is complete, make sure to run the `sync` command before rebooting your system, this will make sure all data is written to your disk. When you reboot you should be able to see /dev/sgX. 

## Install Required Packages

Please install the below packages before compiling the kernel.
```bash
sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs device-tree-compiler python2 python3 \
python-is-python3 fdisk bc
```

## Compile the Linux Kernel

1. Clone the Linux Kernel repository.
```bash
git clone -b linux-5.10-gen-rkr4 https://github.com/Joshua-Riek/linux-rockchip.git
```
2. Build as a deb package.
```bash
cd linux-rockchip
sudo dpkg-buildpackage -a "$(cat debian/arch)" -d -b -nc -uc

```

3. Install the kernel deb packages.
```bash
sudo dpkg -i ../*.deb
sync
```

leimera, 2023-07-12:
Thank you @Joshua-Riek. Looks like you were able to figure out the solution faster than it took me to figure out what the problem was. And thank you for implementing the module. If I get a sec I will follow your instructions.

Joshua-Riek, 2023-07-12:
If you have difficulty, I plan on releasing a kernel update sometime tomorrow. You will be able to download and install it with a simple `sudo apt update && sudo apt upgrade` command. I will let you know once it is available.

leimera, 2023-07-13:
Sounds good. Thank you!

Joshua-Riek, 2023-07-13:
The update went live last night, update your system and issue a reboot. It should work with no issues.

leimera, 2023-07-14:
Will do.