Joshua-Riek/ubuntu-rockchip #258: NanoPi R6S USB-3 port not working

Labbers, 2023-07-13:
NanoPi R6S USB-3 port is not working. It is working in other ubuntu versions, therefore it is not a hardware issue. 
Also, bluetooth gamepads(ps5) is not forwarded with Moonlight-qt, this also works with armbian and the friendlyelec ubuntu image.

Labbers, 2023-07-16:
Seems like both problems came with the updates in the linux-5.10-gen-rkr4 kernel. By reverting to linux-5.10-gen-rkr3.6 the usb port and the bluetooth gamepad works again.

Joshua-Riek, 2023-07-26:
This has been fixed by efectn. See the pr above.

closed 2023-07-26 by Joshua-Riek

npollonen, 2023-07-28:
Updating still brakes both the HID driver and the USB 3.0. 
The  fix by efectn was only for the USB driver and was maybe not yet added to your version? Latest commit 2 days ago.

Joshua-Riek, 2023-07-28:
It's not in the latest version. The commit was merged a few hours after I released the lastest kernel update.