Joshua-Riek/ubuntu-rockchip #259: add file dtbo overlays error

tickley, 2023-07-14:
reading /dtbs/rk3588-rock-5b.dtb
255544 bytes read in 25 ms (9.7 MiB/s)
** Unable to read file /dtbs/overlays/rock-5b-rock-5ab-camera-imx415-boy.dtbo **
reading /dtbs/overlays/rock-5ab-camera-imx415-boy.dtbo
2780 bytes read in 8 ms (338.9 KiB/s)
Applying device tree overlay: /dtbs/overlays/rock-5ab-camera-imx415-boy.dtbo
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
** Unable to read file /dtbs/overlays/rock-5b-rock-5b-hdmi1-8k.dtbo **
reading /dtbs/overlays/rock-5b-hdmi1-8k.dtbo
1372 bytes read in 8 ms (167 KiB/s)
Applying device tree overlay: /dtbs/overlays/rock-5b-hdmi1-8k.dtbo
failed on fdt_overlay_apply(): FDT_ERR_BADMAGIC
base fdt does did not have a /__symbols__ node
make sure you've compiled with -@
Error applying device tree overlays, restoring original device tree
reading /dtbs/rk3588-rock-5b.dtb
255544 bytes read in 24 ms (10.2 MiB/s)

Joshua-Riek, 2023-07-14:
The overlay `rock-5ab-camera-imx415-boy` does not exist. The available overlays can be found in the kernel source here:
https://github.com/Joshua-Riek/linux-rockchip/tree/linux-5.10-gen-rkr4/arch/arm64/boot/dts/rockchip/overlay

tickley, 2023-07-14:
rock-5b-hdmi1-8k Why does this overlay existence still report errors? Is it affected by rock-5ab-camera-imx415-boy

Joshua-Riek, 2023-07-14:
Oh, you will need to remove `.dtbo` in the overlay name. Below is part of the u-boot script that loads overlays.

```
for overlay_file in ${overlays}; do
    if load ${devtype} ${devnum}:${distro_bootpart} ${fdtoverlay_addr_r} /dtbs/overlays/${overlay_prefix}-${overlay_file}.dtbo; then
        echo "Applying device tree overlay: /dtbs/overlays/${overlay_prefix}-${overlay_file}.dtbo"
        fdt apply ${fdtoverlay_addr_r} || setenv overlay_error "true"
    elif load ${devtype} ${devnum}:${distro_bootpart} ${fdtoverlay_addr_r} /dtbs/overlays/${overlay_file}.dtbo; then
        echo "Applying device tree overlay: /dtbs/overlays/${overlay_file}.dtbo"
        fdt apply ${fdtoverlay_addr_r} || setenv overlay_error "true"
    elif load ${devtype} ${devnum}:${distro_bootpart} ${fdtoverlay_addr_r} /dtbs/overlays/rk3588-${overlay_file}.dtbo; then
        echo "Applying device tree overlay: /dtbs/overlays/rk3588-${overlay_file}.dtbo"
        fdt apply ${fdtoverlay_addr_r} || setenv overlay_error "true"
    fi
done
```

tickley, 2023-07-14:
ok, I see the problem. I'll give it a try. Thank you very much

closed 2023-07-14 by Joshua-Riek