Joshua-Riek/ubuntu-rockchip #260: How can I install system updates?

za-songguo, 2023-07-14:
If there is a system update, how can I install it? Can I install it through apt? Or should I flash a new image to my MicroSD card? 
What’s more, will this project support the next LTS version of Ubuntu?

Joshua-Riek, 2023-07-14:
When there is an update or new release, you can install it through apt—no need to re-install the entire system. 

I would like to support Ubuntu 24.04 when it becomes available. It will take time as a lot of packages that relate to multimedia will need to be ported.

za-songguo, 2023-07-14:
> When there is an update or new release, you can install it through apt—no need to re-install the entire system.
> 
> I would like to support Ubuntu 24.04 when it becomes available. It will take time as a lot of packages that relate to multimedia will need to be ported.

When Ubuntu 24.04 comes out, can I update my system to Ubuntu 24.04 through apt?

Joshua-Riek, 2023-07-14:
> When Ubuntu 24.04 comes out, can I update my system to Ubuntu 24.04 through apt?

I have no idea, It's way too early for me to know. A lot of changes will need to take place for a smooth update process between LTS releases.

za-songguo, 2023-07-14:
> I have no idea, It's way too early for me to know. A lot of changes will need to take place for a smooth update process between LTS releases.

Hope everything is fine.
Thanks!

closed 2023-07-14 by Joshua-Riek