Joshua-Riek/ubuntu-rockchip #261: Support Cephfs

inux39, 2023-07-15:
Already enabled ceph block module.
but cephfs module is still not contained.

```
ubuntu@ubuntu:~$ modinfo ceph
modinfo: ERROR: Module ceph not found.
ubuntu@ubuntu:~$ modinfo libceph
name:           libceph
filename:       (builtin)
license:        GPL
file:           net/ceph/libceph
description:    Ceph core library
author:         Patience Warnick <patience@newdream.net>
author:         Yehuda Sadeh <yehuda@hq.newdream.net>
author:         Sage Weil <sage@newdream.net>
```

```
CONFIG_CEPH_FS=m
CONFIG_CEPH_FSCACHE=y
CONFIG_CEPH_FS_POSIX_ACL=y
CONFIG_CEPH_FS_SECURITY_LABEL=y
```

Joshua-Riek, 2023-07-15:
Thank you for reporting this. I will add this to the kernel config shortly.

Joshua-Riek, 2023-07-15:
Support for Cephfs has been added with the commit https://github.com/Joshua-Riek/linux-rockchip/commit/c431f27b53525c678dd1439448047998ff2e9981.

closed 2023-07-15 by Joshua-Riek

inux39, 2023-07-20:
Thank you very much for the release!
I able to mount cephfs.