Joshua-Riek/ubuntu-rockchip #262: Building error on orangepi-5

hakotani-o, 2023-07-15:
run:  
sudo ./build.sh --board=orangepi-5 -l

error:
 E: Unable to locate package u-boot-orangepi-5
 Error: in ./scripts/config-image.sh on line 87
 Error: in ./build.sh on line 134

package name may be 
u-boot-orangepi5/jammy 2017.09+20230525.gitac3b1ca9-1 arm64
  U-Boot for the Orange Pi 5

Joshua-Riek, 2023-07-15:
I'm renaming some of the packages on Launchpad and the build process. I have not yet been able to upload them to my ppa.

This is to make the board names a little easier to read such as below:
    orangepi5 -> orangepi-5
    orangepi5b -> orangepi-5b
    orangepi5plus -> orangepi-5-plus

hakotani-o, 2023-07-18:
I tried to fix it , Until the Launchpad packages are ready.
It worked fine with orangepi-5
What do you think?

file: scripts/config-image.sh
line: 87

    # Install the bootloader
    if [[ ${LAUNCHPAD}  == "Y" ]]; then
        lp_name=`echo ${BOARD} | sed 's/-5/5/'`
        chroot ${chroot_dir} /bin/bash -c "apt-get -y install u-boot-${lp_name}-rk3588"
    else

Joshua-Riek, 2023-07-19:
The new packages are now available on Launchpad, and this is no longer an issue.
https://launchpad.net/~jjriek/+archive/ubuntu/rockchip

closed 2023-07-19 by Joshua-Riek