Joshua-Riek/ubuntu-rockchip #264: OPi5+ snap error firefox

bdlabitt, 2023-07-17:
Previously I used Debian Bookworm, provided by OPi.  It failed a lot with TLS errors, and Arduino app could not download board support packages due to openssl errors.  curl failed with openssl errors.

Now I have downloaded the ubuntu image from here for OPi5+ and have it setup.  During setup, Ubuntu tried to install firefox and failed.  Unfortunately I could not capture the error.  I attempted to download the snap firefox and I believe this is the error I received:
`Download snap firefox (2910) from channel "stable" (local error:tls: bad record MAC)
 Download snap gnome-3-38-2004 from channel "stable" (local error: tls: bad record MAC)
 dpkg: error processing archive /var/cache/apt/archives/firefox_1%3a1snap1-0ubuntu2_amr64.deb (--unpack)
new firefox package pre-installation script subprocess returned error status 1`

I can understand that some snaps may not be available.  But Ubuntu seems to depend on them.  More concerning are the TLS errors.  

Is this all expected behavior on an OPi5+?  Is there some hardware error or software that's causing this?  Having openssl working and or tls, is necessary these days, right?  If this is expected behavior, I can still return this computer.  It's a pity, since booting from nvme has been awesome.  Runs circles around an RPI.  

If there's something I can try, please let me know.  I do want to be able to run Arduino on the OPi5+.

Joshua-Riek, 2023-07-17:
I have not encountered any TLS or openssl errors, maybe its your network? Ubuntu does not depend on any snap packages, they are optional. Have you installed snapd?

```bash
sudo apt update
sudo apt install snapd
```

bdlabitt, 2023-07-17:
apt is also reporting Hash Sum mismatch on some files
`Err:108 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 libllvm14 arm64 1:14.0.0-1ubuntu1      
  Hash Sum mismatch
  Hashes of expected file:
   - SHA512:18b7b48efa57b7371f01e5b6c17afef88da377b4f010dd9415837429486de1d1ebda8633e4d315bb52ffcf34e84ba879a89b352e12a46688196f41595935d31a
   - SHA256:0bda201c9940dc2e3d681da2c498e7e4ce1b9daab293781f8d2f8a2e558d2996
   - SHA1:34e620226a1f1d35121ea83faac72e4ded4e7088 [weak]
   - MD5Sum:dd072c09ee5a0e38ebef97a5152cb225 [weak]
   - Filesize:22601780 [weak]
  Hashes of received file:
   - SHA512:b9df80ed0727a29a783cb7e473a666cf70fad9c90d82412ee6afcc098b346645391e5b110bcabf65eef24089a2b606ff4ccbeee7e89644f16b6bc969f161002a
   - SHA256:41ed1ba192ca8ef99a8e737771308a3e949f10c1870ce8a164ffb60be168270a
   - SHA1:3f09aef43223c53f269a4ce69c8880a0b8e0a9e9 [weak]
   - MD5Sum:1ab37ef3674351c7dfadf9bf80cbebd6 [weak]
   - Filesize:22601780 [weak]
  Last modification reported: Fri, 25 Mar 2022 02:34:04 +0000
`

Joshua-Riek, 2023-07-17:
Ehhh, can you post the full log from your console?

bdlabitt, 2023-07-17:
Re: network errors, maybe, but my PopOS i7 laptop (an Ubuntu variant) reports no such errors on big downloads.  I am on a decent network now.  Over 500Mbps has been observed on the OPi5+.

bdlabitt, 2023-07-17:
Yes.  
[aptoutput.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12074020/aptoutput.txt)

Joshua-Riek, 2023-07-17:
Hmm, looks like some issue with the package, my guess is the apt mirror is updating right now. Where the hash and package have not both been updated yet.

bdlabitt, 2023-07-17:
> I have not encountered any TLS or openssl errors, maybe its your network? Ubuntu does not depend on any snap packages, they are optional. Have you installed snapd?
> 
> ```shell
> sudo apt update
> sudo apt install snapd
> ```

Didn't know that snapd was optional with Ubuntu, thought it was forced down our throats.  My bad on that.  No experience with Ubuntu on small arm64 platforms.

Joshua-Riek, 2023-07-17:
I would have it installed by default, but there is an issue where snaps can not be installed in a chroot environment. So i would have to come up with some sort of hack to get them installed during the first boot of the system.

bdlabitt, 2023-07-17:
> Hmm, looks like some issue with the package, my guess is the apt mirror is updating right now. Where the hash and package have not both been updated yet.

So, just wait a bit?  Been suspicious of these tls errors.  Made Arduino 1.8.19 unusable, because it couldn't download without tls errors.

I am also getting a warning in apt, at the end, about an OpenSSL vulnerability CVE-2-23-2650, yet there are no openssl updates.  Never seen this advertising in apt before...  New ubuntu thing?

bdlabitt, 2023-07-17:
> I would have it installed by default, but there is an issue where snaps can not be installed in a chroot environment. So i would have to come up with some sort of hack to get them installed during the first boot of the system.

Thanks, then I will install it.  Kind of holding my nose though.  Really don't go for these snaps/flatpacks, they often are huge downloads.  Seems like they take as long to download as to compile.  Some of these downloads are nearly as big as the distro!

bdlabitt, 2023-07-17:
Do I have to install the snap store?  Or something else?  I notice there is no way to install any software, save for apt. (or synaptic, but that's the same thing.)

Still failing firefox install.
`bruce@bopi5p:~/Apps$ sudo apt install snapd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
snapd is already the newest version (2.58+22.04.1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
bruce@bopi5p:~/Apps$ sudo apt install firefox
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  firefox
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 0 B/72.3 kB of archives.
After this operation, 261 kB of additional disk space will be used.
Preconfiguring packages ...
(Reading database ... 185474 files and directories currently installed.)
Preparing to unpack .../firefox_1%3a1snap1-0ubuntu2_arm64.deb ...
=> Installing the firefox snap
==> Checking connectivity with the snap store
==> Installing the firefox snap
error: cannot perform the following tasks:
- Download snap "firefox" (2910) from channel "stable" (local error: tls: bad record MAC)
- Download snap "gnome-3-38-2004" (145) from channel "stable" (local error: tls: bad record MAC)
dpkg: error processing archive /var/cache/apt/archives/firefox_1%3a1snap1-0ubuntu2_arm64.deb (--unpack):
 new firefox package pre-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/firefox_1%3a1snap1-0ubuntu2_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

Joshua-Riek, 2023-07-17:
> So, just wait a bit? Been suspicious of these tls errors. Made Arduino 1.8.19 unusable, because it couldn't download without tls errors.
>
> I am also getting a warning in apt, at the end, about an [...]

You should only need to wait a bit, I have no idea how long but maybe 1-2 hours? I can install a new image and test it out to see if I get the same result later tonight. Right now I'm working on some bootloader and kernel issues that are blocking my next release.

bdlabitt, 2023-07-18:
Finally got firefox to install.
You can close this issue.  I'm getting too many downloading errors and TLS errors on this platform independent of OS to continue with this hardware.  Sorry to bother you.  Approve of your work and your efforts.

closed 2023-07-18 by bdlabitt