Joshua-Riek/ubuntu-rockchip #265: flash-kernel not updating

BooBerry, 2023-07-19:
Hi, I wasn't sure where to post this, but suddenly when checking for updates via apt using this distro on my Orange Pi 5B and attempting to install updates, an error occurs when trying to update flash-kernel...

`Setting up flash-kernel (3.104ubuntu16+17) ...
flash-kernel: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.140ubuntu13.2) ...
Processing triggers for flash-kernel (3.104ubuntu16+17) ...
Can't find /boot/vmlinuz- (see /tmp/flash-kernel-no-kernel-error.log)
dpkg: error processing package flash-kernel (--configure):
 installed flash-kernel package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 flash-kernel
E: Sub-process /usr/bin/dpkg returned an error code (1)`

And the contents of flash-kernel-no-kernel-error.log...

`+ ls -l /boot/vmlinuz-*
ls: cannot access '/boot/vmlinuz-*': No such file or directory
+ ls -l bin boot dev etc home lib lost+found media mnt opt proc root run sbin snap srv swapfile sys tmp usr var
lrwxrwxrwx   1 root root          7 May 19 19:02 bin -> usr/bin
lrwxrwxrwx   1 root root          7 May 19 19:02 lib -> usr/lib
lrwxrwxrwx   1 root root          8 May 19 19:02 sbin -> usr/sbin
-rw-------   1 root root 2147483648 May 19 19:08 swapfile

boot:
total 16
lrwxrwxrwx 1 root root    31 May 19 19:09 config -> config-5.10.110-rockchip-rk3588
drwxr-xr-x 3 root root 16384 Dec 31  1969 firmware
lrwxrwxrwx 1 root root    35 May 20 05:08 initrd.img -> initrd.img-5.10.110-rockchip-rk3588
lrwxrwxrwx 1 root root    35 May 19 19:09 System.map -> System.map-5.10.110-rockchip-rk3588
lrwxrwxrwx 1 root root    32 May 20 05:08 vmlinuz -> vmlinuz-5.10.110-rockchip-rk3588

dev:
total 8
crw-------  1 root    root     10, 124 Jul 17 08:47 ashmem
crw-r--r--  1 root    root     10, 235 Jul 17 08:47 autofs
drwxr-xr-x  2 root    root         320 Jul 19 02:23 block
crw-rw----  1 root    disk     10, 234 Jul 17 08:47 btrfs-control
drwxr-xr-x  3 root    root          60 Dec 31  1969 bus
crw-rw----  1 root    video   248,   0 Jul 17 08:47 cec0
drwxr-xr-x  2 root    root        3660 Jul 17 08:47 char
crw--w----  1 root    tty       5,   1 Jul 17 08:47 console
crw-------  1 root    root     10, 123 Jul 17 08:47 cpu_dma_latency
crw-rw-rw-  1 root    root     10, 125 Jul 17 08:47 crypto
drwxr-xr-x  8 root    root         160 Jul 17 08:47 disk
drwxrwxrwx  2 root    root         160 Dec 31  1969 dma_heap
drwxr-xr-x  3 root    root         140 Jul 17 08:47 dri
crw-------  1 root    root    240,   0 Jul 17 08:47 drm_dp_aux0
crw-rw----  1 root    video    29,   0 Jul 17 08:47 fb0
lrwxrwxrwx  1 root    root          13 Jul 17 08:47 fd -> /proc/self/fd
crw-rw-rw-  1 root    root      1,   7 Jul 17 08:47 full
crw-rw-rw-  1 root    root     10, 229 Jul 17 08:47 fuse
crw-------  1 root    root    254,   0 Jul 17 08:47 gpiochip0
crw-------  1 root    root    254,   1 Jul 17 08:47 gpiochip1
crw-------  1 root    root    254,   2 Jul 17 08:47 gpiochip2
crw-------  1 root    root    254,   3 Jul 17 08:47 gpiochip3
crw-------  1 root    root    254,   4 Jul 17 08:47 gpiochip4
crw-------  1 root    root    254,   5 Jul 17 08:47 gpiochip5
crw-------  1 root    root    236,   0 Jul 17 08:47 hidraw0
crw-------  1 root    root    236,   1 Jul 17 08:47 hidraw1
drwxr-xr-x  2 root    root           0 Jul 17 08:47 hugepages
crw-------  1 root    root     10, 183 Jul 17 08:47 hwrng
crw-rw----  1 root    i2c      89,   0 Jul 17 08:47 i2c-0
crw-rw----  1 root    i2c      89,  10 Jul 17 08:47 i2c-10
crw-rw----  1 root    i2c      89,   2 Jul 17 08:47 i2c-2
crw-rw----  1 root    i2c      89,   6 Jul 17 08:47 i2c-6
crw-rw----  1 root    i2c      89,   7 Jul 17 08:47 i2c-7
crw-rw----  1 root    i2c      89,   9 Jul 17 08:47 i2c-9
crw-------  1 root    root    245,   0 Jul 17 08:47 iio:device0
lrwxrwxrwx  1 root    root          12 Jul 17 08:47 initctl -> /run/initctl
drwxr-xr-x  4 root    root         300 Jul 17 08:47 input
[... 912 lines cut ...]
drwxr-xr-x  57 root root  4096 May 27 09:15 include
drwxr-xr-x 116 root root  4096 Jul 14 04:00 lib
lrwxrwxrwx   1 root root     3 May 19 19:22 lib64 -> lib
drwxr-xr-x  23 root root 12288 Jul 11 04:04 libexec
drwxr-xr-x  10 root root  4096 May 19 19:02 local
drwxr-xr-x   2 root root 20480 Jul 19 05:06 sbin
drwxr-xr-x 292 root root 12288 May 27 09:15 share
drwxr-xr-x   2 root root  4096 Jul 14 04:00 src

var:
total 48
drwxr-xr-x  2 root root     4096 Jul 18 00:00 backups
drwxr-xr-x 19 root root     4096 May 20 11:15 cache
drwxrwsrwt  2 root whoopsie 4096 Jul 17 08:46 crash
drwxr-xr-x 71 root root     4096 May 20 05:27 lib
drwxrwsr-x  2 root staff    4096 Apr 18  2022 local
lrwxrwxrwx  1 root root        9 May 19 19:02 lock -> /run/lock
drwxrwxr-x 15 root syslog   4096 Jul 18 00:00 log
drwxrwsr-x  2 root mail     4096 May 19 19:02 mail
drwxrwsrwt  2 root whoopsie 4096 May 19 19:20 metrics
drwxr-xr-x  2 root root     4096 May 19 19:02 opt
lrwxrwxrwx  1 root root        4 May 19 19:02 run -> /run
drwxr-xr-x  7 root root     4096 Jul  5 16:27 snap
drwxr-xr-x  7 root root     4096 May 19 19:19 spool
drwxrwxrwt 14 root root     4096 Jul 19 05:06 tmp`

Joshua-Riek, 2023-07-20:
It looks like you are on an old release. I messed up the kernel packaging, where the 5.10.110 kernel could not upgrade to 5.10.160. I attempted to fix this a few times and encountered many issues, I'm not sure if this is something I would fix at this point. Either place flash-kernel on hold or re-install your system, I'm sorry about the inconvenience.
```
sudo apt-mark hold flash-kernel
```

BooBerry, 2023-07-20:
Went ahead and reinstalled, all is well again. :)

closed 2023-07-20 by BooBerry