Joshua-Riek/ubuntu-rockchip #266: Color Problem on Wayland

za-songguo, 2023-07-20:
(v1.19)
I found that color has different display effects on Xorg or Wayland, and the effect on Xorg is normal because it looks as same as my phone displays.
I don't know the reason, is it the problem of the GPU driver?

Joshua-Riek, 2023-07-20:
I believe this is an issue with Wayland. For instance, if you try to use the gnome night light feature it will not work at all on Wayland, but will work fine on X11.

za-songguo, 2023-07-20:
The panfork GPU driver doesn't works well on Xorg.
😢

za-songguo, 2023-07-21:
> I believe this is an issue with Wayland. For instance, if you try to use the gnome night light feature it will not work at all on Wayland, but will work fine on X11.

The panfork GPU driver doesn't works well on Xorg.

za-songguo, 2023-07-21:
> I believe this is an issue with Wayland. For instance, if you try to use the gnome night light feature it will not work at all on Wayland, but will work fine on X11.

The panfork GPU driver doesn't work well on Xorg.

Joshua-Riek, 2023-07-21:
Yeah, sadly, I think Xorg will always not perform well on this device. Most arm devices I have been working with are all switching to Wayland which has its own problems.

za-songguo, 2023-08-13:
I think it is a problem with my display, I will close the issue.

closed 2023-08-13 by za-songguo