Joshua-Riek/ubuntu-rockchip #267: hope to introduce rtl8188EUS driver

SESProMax, 2023-07-21:
I have a usb wifi adaptor comfast CF-WU810N. It can not work on this OS. It use rtl8188EUS, so I think its driver can be added in system.

Joshua-Riek, 2023-07-21:
I will take a look, some wifi drivers are easier to implement than others.

Joshua-Riek, 2023-07-21:
I should be able to add this driver to the kernel, and I see no issues at a glance. I will try to add this over the weekend.

SESProMax, 2023-07-21:
Thankyou.

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

Joshua-Riek, 2023-07-23:
I added the driver to the kernel https://github.com/Joshua-Riek/linux-rockchip/commit/555c135525e09e83fcf20ae3655fddeda5c68ac3. It should be working when I release the next kernel update.

closed 2023-07-23 by Joshua-Riek

SESProMax, 2023-07-24:
Does this OS support OTA through apt？

Joshua-Riek, 2023-07-24:
Yes, this OS supports OTA updates.