Joshua-Riek/ubuntu-rockchip #269: Wazuh Install failing (not 64 bit?)

SignalMedic, 2023-07-22:
https://documentation.wazuh.com/current/quickstart.html

Ran this install script and it failed, claiming that it must be run on a 64-bit system. 

Orange Pi 5, Samsung 512GB NVMe, 1.20 server booting from SSD.

Joshua-Riek, 2023-07-22:
Maybe the x86_64 version was installed. Try to download the aarch64 versions for ubuntu.

https://documentation.wazuh.com/current/installation-guide/packages-list.html

Joshua-Riek, 2023-07-22:
I look at the download script; it is only for x86_64 system, not arm64. 

```bash
# ------------ checks.sh ------------ 
function checks_arch() {

    arch=$(uname -m)

    if [ "${arch}" != "x86_64" ]; then
        common_logger -e "Uncompatible system. This script must be run on a 64-bit system."
        exit 1
    fi
}
```

SignalMedic, 2023-07-22:
I'm in idiot. Thanks

Joshua-Riek, 2023-07-22:
Nah, simple mistake!

closed 2023-07-23 by Joshua-Riek