Joshua-Riek/ubuntu-rockchip #270: Youtube Video: Green screen

ggdk123, 2023-07-23:
Some videos on Youtube gives a green screen, audio works fine. a decoder issue?
[image]

Joshua-Riek, 2023-07-23:
Some video resolutions are displayed in green under the panfork mesa driver. The alternative would be to use the Mali blob driver with X11, which introduces many more issues and will provide worse overall performance. 

Currently, the most optimal performance-centered solution is to use mesa panfork with wayland.

ggdk123, 2023-07-24:
> Some video resolutions are displayed in green under the panfork mesa driver.

I see, that is why most of shorts are in green.

digby5, 2023-07-25:
not sure if its just an ppa update but with the latest armbian build, the green screen on sd videos seem to have disappeared

Joshua-Riek, 2023-07-26:
You are probably using x11, as Armbian defaults to this display manager. It has a lot of performance issues compared to wayland.

digby5, 2023-07-27:
23.5 gnome. just checked cpu usage and it looks like they disabled hw video decode tho' hw accel otherwise is enabled. this is wayland btw. chromium gpu section says hw video decode is enabled but the video codec part is blank. i check the udev, egl and other settings to make sure the install from the ppa is the same as prior installs and they look the same. i guess some developer got tired of the green screens tho' wirh this compromise some higher bitrate 4k videos will max the cpu

Joshua-Riek, 2023-07-27:
There are also specific flags chromium must be run with for hw acceleration.

ironiris, 2023-08-07:
chrome://settings/system
System -> Use hardware acceleration when available "off" and Relaunch
and then
System -> Use hardware acceleration when available "on" and Relaunch
Resolved.

Mreddi1, 2023-08-10:
> chrome://settings/system System -> Use hardware acceleration when available "off" and Relaunch and then System -> Use hardware acceleration when available "on" and Relaunch Resolved.

Only turning off the hardware acceleration helped me with green screen

JuneJaJa, 2023-09-15:
This case workaround. Install "enhanced-h264ify" extension and configure options enable Block AV1. Close chomium browser and open it again.

edp1096, 2023-12-19:
[image]

Most videos play well with `h264ify` extension.
I think this issue can be closed.

hubby4, 2024-03-12:
I'm no longer experiencing green video issues recently.

Did switching back to Amazingfate's PPA have something to do with it?

Either way, this is positive development.

nyanmisaka, 2024-03-13:
Because the MPP is patched to use 64-aligned pitch.
https://github.com/nyanmisaka/mpp/commit/1c92144b40d087a9224e7a2784e8170016cff087
https://github.com/nyanmisaka/mpp/commit/78868cb6110afac329c540443c8f4acd8f49d303