Joshua-Riek/ubuntu-rockchip #272: request : multiple: system and build instructions

develperbayman, 2023-07-23:
1.)can we get links on the main github page to build instructions (used to be there?) mainly the required dependencies to build on arm i used to know but forgot and deleted the information like a smart person distro hopping

2.) openGL can you upgrade it to at least 3.2 as standard out of the box? (newer the better) most good emulator programs require it (i tried downloading and building openGL i have failed so far not sure what im doing) .....better video hardware/software alignment i.e chrome green screen but works fine in firefox ...3d printing software works great! thanks

3.)built in microphone still isnt being displayed correctly in settings seems to work spotty at best im trying to load large language models using a custom chat program that uses voice and audio input ....i usually work with this while watching tv its smart enough to know me from the tv it will occasionally pickup the tv but very spotty but will not take my commands ...the only way i can really tell is on debian plasma the mic works great so i can tell because how long iv been doing it

4.)could we barrow the part from the x64 version of the installer to select which disk to format and install on during initial setup and update the spi flash accordingly 

looks great this is still the best distro out there for the opi5 thanks!

Joshua-Riek, 2023-07-23:
1. There is a link to the wiki on the readme document, which has build instructions linked here: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Building
2. We must wait for mainline Linux support for further Mesa or OpenGL updates. The currently supported GPU software is a hack created by icecream95, which provides fundamental support, and that is it. It will be long before we have this stable and usable.
3. I recommend using an external microphone plugged into the audio jack. It should work perfectly for detecting audio input. The onboard microphone needs a major audio config hack to break other audio devices. So, for now, there are no plans to fix this.
4. I have been experimenting with this but have found multiple issues where it might not be feasible until better software support exists. I need a lot of research and development here, as this would essentially be a live iso installer.

closed 2023-08-09 by Joshua-Riek

rcarmo, 2024-02-14:
Re: 4, Armbian has working SPI support. I can boot an SD card, dd the system to an NVMe and have it update the SPI via `armbian-config`.