Joshua-Riek/ubuntu-rockchip #273: gpio operation not permitted since release v1.20

smerik, 2023-07-23:
Using the `v1.20` Ubuntu Server board image [Orange Pi 5](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.20/ubuntu-22.04.2-preinstalled-server-arm64-orangepi-5.img.xz) I faced the following error message:
```sh
ubuntu@ubuntu:~$ sudo gpio readall
wiringPiSetup: mmap (GPIO) failed: Operation not permitted
```

When using the `v1.19` Ubuntu Server board image [Orange Pi 5](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.19/ubuntu-22.04.2-preinstalled-server-arm64-orangepi5.img.xz) the problem is not there:
```sh
ubuntu@ubuntu:~$ sudo gpio readall
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |   47 |   0 |    SDA.5 |     IN | 1 |  3 || 4  |   |        | 5V       |     |      |
 |   46 |   1 |    SCL.5 |     IN | 1 |  5 || 6  |   |        | GND      |     |      |
 |   54 |   2 |    PWM15 |     IN | 1 |  7 || 8  | 0 | IN     | RXD.0    | 3   | 131  |
 |      |     |      GND |        |   |  9 || 10 | 0 | IN     | TXD.0    | 4   | 132  |
 |  138 |   5 |  CAN1_RX |     IN | 1 | 11 || 12 | 1 | IN     | CAN2_TX  | 6   | 29   |
 |  139 |   7 |  CAN1_TX |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
 |   28 |   8 |  CAN2_RX |     IN | 1 | 15 || 16 | 1 | IN     | SDA.1    | 9   | 59   |
 |      |     |     3.3V |        |   | 17 || 18 | 1 | IN     | SCL.1    | 10  | 58   |
 |   49 |  11 | SPI4_TXD |     IN | 1 | 19 || 20 |   |        | GND      |     |      |
 |   48 |  12 | SPI4_RXD |     IN | 1 | 21 || 22 | 1 | IN     | GPIO2_D4 | 13  | 92   |
 |   50 |  14 | SPI4_CLK |     IN | 1 | 23 || 24 | 1 | IN     | SPI4_CS1 | 15  | 52   |
 |      |     |      GND |        |   | 25 || 26 | 1 | IN     | PWM1     | 16  | 35   |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
```

It's not clear to me what is causing this problem. I also have an Orange Pi 5 running with an SSD which was originally installed using the `v1.18` image and is up-to-date with all packages, including the kernel and `gpio` is also working fine on that system.

Joshua-Riek, 2023-07-23:
This is very strange, I will investigate. Thank you for reporting this.

Joshua-Riek, 2023-07-23:
I forgot to update the orangepi-release file when I renamed the board names. WiringOP looks for a specific string when trying to determine the board type. Start a terminal with sudo and enter the below command. 

```bash
echo "BOARD=orangepi5" > /etc/orangepi-release
```

smerik, 2023-07-23:
Ah, interesting. I already found that command in issue #40 but I already saw mine was filled with that value. But apparently I did overlook the dash sign `-` 😅

Your solution solved my problem:
```sh
ubuntu@ubuntu:~$ cat /etc/orangepi-release
BOARD=orangepi-5
ubuntu@ubuntu:~$ sudo gpio readall
wiringPiSetup: mmap (GPIO) failed: Operation not permitted
ubuntu@ubuntu:~$ sudo su -
root@ubuntu:~# echo "BOARD=orangepi5" > /etc/orangepi-release
root@ubuntu:~#
logout
ubuntu@ubuntu:~$ sudo gpio readall
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |   47 |   0 |    SDA.5 |     IN | 1 |  3 || 4  |   |        | 5V       |     |      |
 |   46 |   1 |    SCL.5 |     IN | 1 |  5 || 6  |   |        | GND      |     |      |
 |   54 |   2 |    PWM15 |     IN | 1 |  7 || 8  | 0 | IN     | RXD.0    | 3   | 131  |
 |      |     |      GND |        |   |  9 || 10 | 0 | IN     | TXD.0    | 4   | 132  |
 |  138 |   5 |  CAN1_RX |     IN | 1 | 11 || 12 | 1 | IN     | CAN2_TX  | 6   | 29   |
 |  139 |   7 |  CAN1_TX |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
 |   28 |   8 |  CAN2_RX |     IN | 1 | 15 || 16 | 1 | IN     | SDA.1    | 9   | 59   |
 |      |     |     3.3V |        |   | 17 || 18 | 1 | IN     | SCL.1    | 10  | 58   |
 |   49 |  11 | SPI4_TXD |     IN | 1 | 19 || 20 |   |        | GND      |     |      |
 |   48 |  12 | SPI4_RXD |     IN | 1 | 21 || 22 | 1 | IN     | GPIO2_D4 | 13  | 92   |
 |   50 |  14 | SPI4_CLK |     IN | 1 | 23 || 24 | 1 | IN     | SPI4_CS1 | 15  | 52   |
 |      |     |      GND |        |   | 25 || 26 | 1 | IN     | PWM1     | 16  | 35   |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
```

Thank you for the quick reply.

Joshua-Riek, 2023-07-23:
Thanks again for reporting this. I will edit the latest release info to link this issue with the fix.

Joshua-Riek, 2023-07-23:
### Fix for the Orange Pi 5 / 5B
```bash
 echo "BOARD=orangepi5" | sudo tee /etc/orangepi-release
```
### Fix for the Orange Pi 5 Plus
```bash
 echo "BOARD=orangepi5plus" | sudo tee /etc/orangepi-release
```

closed 2023-07-26 by Joshua-Riek