Joshua-Riek/ubuntu-rockchip #274: rk_vendor_read eth mac address failed since release v1.20

smerik, 2023-07-23:
Using the `v1.20` Ubuntu Server board image [Orange Pi 5](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.20/ubuntu-22.04.2-preinstalled-server-arm64-orangepi-5.img.xz) the system does not seem to be able to detect the MAC address of the network port resulting in the generation of a random MAC address.
This happens on each reboot, which results a different MAC address on each boot. When a DHCP server is depending on assigning an IP address using the MAC address, this is not possible anymore.

This problem does not occur using the `v1.19` Ubuntu Server board image [Orange Pi 5](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.19/ubuntu-22.04.2-preinstalled-server-arm64-orangepi5.img.xz).

Here is some `dmesg` logging by `rk_gmac-dwmac fe1c0000.ethernet` from `v1.20` with the error message at `8.216366`:
```log
[    8.215600] rk_gmac-dwmac fe1c0000.ethernet: IRQ eth_lpi not found
[    8.215779] rk_gmac-dwmac fe1c0000.ethernet: Looking up phy-supply from device tree
[    8.215791] rk_gmac-dwmac fe1c0000.ethernet: Looking up phy-supply property in node /ethernet@fe1c0000 failed
[    8.215805] rk_gmac-dwmac fe1c0000.ethernet: no regulator found
[    8.215815] rk_gmac-dwmac fe1c0000.ethernet: clock input or output? (output).
[    8.215824] rk_gmac-dwmac fe1c0000.ethernet: TX delay(0x42).
[    8.215833] rk_gmac-dwmac fe1c0000.ethernet: Can not read property: rx_delay.
[    8.215841] rk_gmac-dwmac fe1c0000.ethernet: set rx_delay to 0xffffffff
[    8.215854] rk_gmac-dwmac fe1c0000.ethernet: integrated PHY? (no).
[    8.215864] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock mac_clk_rx
[    8.215873] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock mac_clk_tx
[    8.215892] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock clk_mac_speed
[    8.216117] rk_gmac-dwmac fe1c0000.ethernet: init for RGMII_RXID
[    8.216266] rk_gmac-dwmac fe1c0000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    8.216277] rk_gmac-dwmac fe1c0000.ethernet: 	DWMAC4/5
[    8.216285] rk_gmac-dwmac fe1c0000.ethernet: DMA HW capability register supported
[    8.216294] rk_gmac-dwmac fe1c0000.ethernet: RX Checksum Offload Engine supported
[    8.216303] rk_gmac-dwmac fe1c0000.ethernet: TX Checksum insertion supported
[    8.216311] rk_gmac-dwmac fe1c0000.ethernet: Wake-Up On Lan supported
[    8.216347] rk_gmac-dwmac fe1c0000.ethernet: TSO supported
[    8.216356] rk_gmac-dwmac fe1c0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    8.216366] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: rk_vendor_read eth mac address failed (-1)
[    8.216380] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: generate random eth mac address: 82:4a:22:95:fd:af
[    8.216390] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: rk_vendor_write eth mac address failed (-1)
[    8.216400] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: id: 1 rk_vendor_read eth mac address failed (-1)
[    8.216410] rk_gmac-dwmac fe1c0000.ethernet: rk_get_eth_addr: mac address: 82:4a:22:95:fd:af
[    8.216420] rk_gmac-dwmac fe1c0000.ethernet: device MAC address 82:4a:22:95:fd:af
[    8.216430] rk_gmac-dwmac fe1c0000.ethernet: Enabled Flow TC (entries=2)
[    8.216438] rk_gmac-dwmac fe1c0000.ethernet: TSO feature enabled
[    8.216446] rk_gmac-dwmac fe1c0000.ethernet: Using 32 bits DMA width
[   15.530676] rk_gmac-dwmac fe1c0000.ethernet eth0: PHY [stmmac-1:01] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[   15.533271] rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
[   15.533283] rk_gmac-dwmac fe1c0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   15.533513] rk_gmac-dwmac fe1c0000.ethernet eth0: registered PTP clock
[   15.533669] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[   18.571089] rk_gmac-dwmac fe1c0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

Compared to `v1.19`:
```log
[    7.186635] rk_gmac-dwmac fe1c0000.ethernet: IRQ eth_lpi not found
[    7.186780] rk_gmac-dwmac fe1c0000.ethernet: Looking up phy-supply from device tree
[    7.186791] rk_gmac-dwmac fe1c0000.ethernet: Looking up phy-supply property in node /ethernet@fe1c0000 failed
[    7.186804] rk_gmac-dwmac fe1c0000.ethernet: no regulator found
[    7.186814] rk_gmac-dwmac fe1c0000.ethernet: clock input or output? (output).
[    7.186824] rk_gmac-dwmac fe1c0000.ethernet: TX delay(0x42).
[    7.186832] rk_gmac-dwmac fe1c0000.ethernet: Can not read property: rx_delay.
[    7.186840] rk_gmac-dwmac fe1c0000.ethernet: set rx_delay to 0xffffffff
[    7.186853] rk_gmac-dwmac fe1c0000.ethernet: integrated PHY? (no).
[    7.186864] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock mac_clk_rx
[    7.186873] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock mac_clk_tx
[    7.186892] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock clk_mac_speed
[    7.187118] rk_gmac-dwmac fe1c0000.ethernet: init for RGMII_RXID
[    7.187260] rk_gmac-dwmac fe1c0000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    7.187271] rk_gmac-dwmac fe1c0000.ethernet: 	DWMAC4/5
[    7.187280] rk_gmac-dwmac fe1c0000.ethernet: DMA HW capability register supported
[    7.187289] rk_gmac-dwmac fe1c0000.ethernet: RX Checksum Offload Engine supported
[    7.187298] rk_gmac-dwmac fe1c0000.ethernet: TX Checksum insertion supported
[    7.187306] rk_gmac-dwmac fe1c0000.ethernet: Wake-Up On Lan supported
[    7.187341] rk_gmac-dwmac fe1c0000.ethernet: TSO supported
[    7.187349] rk_gmac-dwmac fe1c0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    7.187359] rk_gmac-dwmac fe1c0000.ethernet: Enabled Flow TC (entries=2)
[    7.187367] rk_gmac-dwmac fe1c0000.ethernet: TSO feature enabled
[    7.187375] rk_gmac-dwmac fe1c0000.ethernet: Using 32 bits DMA width
[   14.509051] rk_gmac-dwmac fe1c0000.ethernet eth0: PHY [stmmac-1:01] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[   14.512106] rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
[   14.512118] rk_gmac-dwmac fe1c0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   14.512269] rk_gmac-dwmac fe1c0000.ethernet eth0: registered PTP clock
[   14.512652] rk_gmac-dwmac fe1c0000.ethernet eth0: configuring for phy/rgmii-rxid link mode
[   19.599302] rk_gmac-dwmac fe1c0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```

Joshua-Riek, 2023-07-23:
I recently talked with some other developers, and some weird stuff is happening with MAC addresses on these RK3588 boards. For example, some boards don't have MAC addresses assigned to their ethernet ports which is crazy.

The bootloader can influence MAC address assignments, and I believe the following commit is the cause of the issue https://github.com/orangepi-xunlong/u-boot-orangepi/commit/1f70ac3a968d121d86350092a7eff0dbd56114b7. I will create a patch to revert this commit and let you know the result.

smerik, 2023-07-23:
> I recently talked with some other developers, and some weird stuff is happening with MAC addresses on these RK3588 boards. For example, some boards don't have MAC addresses assigned to their ethernet ports which is crazy.

Aye, this seems to be quite an interesting issue.

> The bootloader can influence MAC address assignments, and I believe the following commit is the cause of the issue [orangepi-xunlong/u-boot-orangepi@1f70ac3](https://github.com/orangepi-xunlong/u-boot-orangepi/commit/1f70ac3a968d121d86350092a7eff0dbd56114b7). I will create a patch to revert this commit and let you know the result.

That sounds promising. I'm looking forward to the result.

Joshua-Riek, 2023-07-23:
Yeah, it was the bootloader. I will be updating the bootloader on my apt ppa shortly and posting instructions on how to fix it once the package is uploaded.

Joshua-Riek, 2023-07-23:
The package build has finished, and waiting for it to be published.

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip/+build/26444502

Joshua-Riek, 2023-07-23:
To fix this issue, users must first update their system with apt as shown below, then update the bootloader depending on their system configuration.
```bash
sudo apt-get update 
sudo apt-get upgrade
```
### Install the bootloader to the SPI NOR flash
```
sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
sync
```

### Install the bootloader to your SD Card
```
dd if=/lib/u-boot-orangepi-rk3588/idbloader.img of=/dev/mmcblk1 seek=64 conv=notrunc
dd if=/lib/u-boot-orangepi-rk3588/u-boot.itb of=/dev/mmcblk1 seek=16384 conv=notrunc
sync
```

### Install the bootloader to the eMMC
```
dd if=/lib/u-boot-orangepi-rk3588/idbloader.img of=/dev/mmcblk0 seek=64 conv=notrunc
dd if=/lib/u-boot-orangepi-rk3588/u-boot.itb of=/dev/mmcblk0 seek=16384 conv=notrunc
sync
```

smerik, 2023-07-24:
Fix confirmed.

Joshua-Riek, 2023-07-24:
Weird, I did not test this out on the opi5+. I will take a look later today. If you had a previous bootloader in the spi nor and booted from something like the nvme this issue should not occur.

The bootloader package for the opi5+ is u-boot-orangepi-5-plus.

Joshua-Riek, 2023-07-24:
Oh, the old u-boot package is still cached on the ppa, which is why you can download that package. It has the previous bootloader revision without the MAC address issue. But either way, the new package is named `u-boot-orangepi-5-plus` and is preinstalled. 

ppa: https://launchpad.net/~jjriek/+archive/ubuntu/rockchip/+packages

ewaldc, 2023-07-24:
Yes, something I wrote can not be correct :-(
When I look now at both systems (one with SD card and one with NVME), the installed packages are  `u-boot-tools` and `u-boot-orange5-pi-plus`, so the correct(ed) ones.  When writing the post, I did a `apt-cache search` which also turned up the `u-boot-orangepi-5-plus-rk3588`, so I must have copied the wrong line.
Quite sure though that neither package was installed, because I looked in /lib and none of the files were there.  Hence, why I installed it.
But I hate to waste more of your time:  I will check again tomorrow and reinstall the image.    In any case, it always great news: it's either fixed or (for some maybe) not a problem.   If there is anything I can do to make it up, let me know!

closed 2023-07-26 by Joshua-Riek