Joshua-Riek/ubuntu-rockchip #275: Startup encounters installer crash

Esxxlx, 2023-07-24:
Hello, I fork your project to adapt my board "rk3588-wkp-r4.dts". You use the construction script to complete the compilation according to the t6 adaptation method, and it runs smoothly.
[image]

Now the problem is: after flashing the machine, start and run the installer, and after a while, it prompts that the installer crashes. I have tried various methods and have not found a solution. I hope you can help me solve the problem, 
thank you

Joshua-Riek, 2023-07-24:
The model name defined in your device tree needs to be added to the flash-kernel package. I would be happy to add it if you can tell me the model name.

Esxxlx, 2023-07-25:
Thank you very much, here are 2 devices in my hand.
```
	model = "WUKONG WRP R4";
	compatible = "wukong,wkp-r4", "rockchip,rk3588";
```
```
	model = "HINLINK H88K MAX";
	compatible = "hinlink,h88k-max", "rockchip,rk3588";
```

Esxxlx, 2023-07-25:
Thank you so much! Report back to you after waiting a few hours and testing again

Esxxlx, 2023-07-25:
Problem solved, thanks again for your help!

closed 2023-07-25 by Esxxlx