Joshua-Riek/ubuntu-rockchip #276: NanoPC-T6 won't boot from NVMe SSD

Stigby, 2023-07-24:
Hi! I have a problem where I cannot boot from the NVMe drive. I have followed all the steps on the wiki page, including installing the bootloader onto the SPI NOR flash, downloading the latest NanoPC-T6 image, and installing the image onto the NVMe drive. After this process, I turned off the NanoPC, removed the SD card, and tried to start up the NanoPC. The system LED is solid, with no flashing.

U-Boot flashing
```
ubuntu@ubuntu:~$ sudo dd if=/lib/u-boot-friendlyelec-rk3588/rkspi_loader.img of=/dev/mtdblock0 conv=notrunc
32768+0 records in
32768+0 records out
16777216 bytes (17 MB, 16 MiB) copied, 1.11745 s, 15.0 MB/s
ubuntu@ubuntu:~$ sync
```
Downloaded the latest NanoPC-T6 image and checked that the NVMe SSD is detected
```
ubuntu@ubuntu:~$ ls
ubuntu-22.04.2-preinstalled-server-arm64-nanopc-t6.img.xz
ubuntu@ubuntu:~$ lsblk /dev/nvme0n1
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
nvme0n1     259:0    0 931.5G  0 disk
├─nvme0n1p1 259:1    0   512M  0 part
└─nvme0n1p2 259:2    0   5.5G  0 part
```

Disk partition table
```
ubuntu@ubuntu:~$ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
GPT PMBR size mismatch (12521471 != 1953525167) will be corrected by write.
The backup GPT table is not on the end of the device.

Disk /dev/nvme0n1: 931.51 GiB, 1000204886016 bytes, 1953525168 sectors
Disk model: WD Blue SN570 1TB
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: E9240F7D-D01B-4614-8A6B-43C3FFA656AF

Device           Start      End  Sectors  Size Type
/dev/nvme0n1p1   32768  1081343  1048576  512M Linux extended boot
/dev/nvme0n1p2 1081344 12519423 11438080  5.5G Linux filesystem

Disk /dev/mmcblk1: 7.45 GiB, 7994343424 bytes, 15613952 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: E9240F7D-D01B-4614-8A6B-43C3FFA656AF

Device           Start      End  Sectors  Size Type
/dev/mmcblk1p1   32768  1081343  1048576  512M Linux extended boot
/dev/mmcblk1p2 1081344 15613918 14532575  6.9G Linux filesystem
```

Writing the NanoPC-T6 image to the NVMe drive (/dev/nvme0n1)
```
ubuntu@ubuntu:~$ xz -dc ubuntu-22.04.2-preinstalled-server-arm64-nanopc-t6.img.xz | sudo dd of=/dev/nvme0n1 bs=4k
1565184+0 records in
1565184+0 records out
6410993664 bytes (6.4 GB, 6.0 GiB) copied, 52.4243 s, 122 MB/s
ubuntu@ubuntu:~$ sync
```

perfectsyrup, 2023-07-24:
ls /dev/mtdblock* command will not help?

I also plan to use nvme on NanoPC-T6. Currently use it with eMMC, but want to switch to nvme disk.

Joshua-Riek, 2023-07-24:
I might have a patch for this issue, but I need help testing it. Please run the below commands on your NanoPC-T6 and let me know if `/dev/mtdblk*` exist after reboot.

```
wget https://github.com/Joshua-Riek/ubuntu-rockchip/files/12152840/rk3588-nanopc-t6.zip
unzip rk3588-nanopc-t6.zip 
sudo cp rk3588-nanopc-t6.dtb /boot/firmware/rk3588-nanopc-t6.dtb
sync
sudo reboot now
```
[rk3588-nanopc-t6.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12152840/rk3588-nanopc-t6.zip)

Joshua-Riek, 2023-07-24:
Here is the code change. I map out fspim1 according to the schematics.
```diff
diff --git a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
index 47a1030a4c73..5a62a440c243 100644
--- a/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3588-nanopc-t6.dts
@@ -478,6 +478,37 @@ &pcie3x4 {
        status = "okay";
 };
 
+&sfc {
+       status = "okay";
+       max-freq = <50000000>;
+       #address-cells = <1>;
+       #size-cells = <0>;
+       pinctrl-names = "default";
+       pinctrl-0 = <&fspim1_pins>;
+
+       spi_flash: spi-flash@0 {
+               #address-cells = <1>;
+               #size-cells = <0>;
+               compatible = "jedec,spi-nor";
+               reg = <0x0>;
+               spi-max-frequency = <50000000>;
+               spi-tx-bus-width = <1>;
+               spi-rx-bus-width = <4>;
+               status = "okay";
+
+               partitions {
+                       compatible = "fixed-partitions";
+                       #address-cells = <1>;
+                       #size-cells = <1>;
+
+                       loader@0 {
+                               label = "loader";
+                               reg = <0x0 0x1000000>;
+                       };
+               };
+       };
+};
+
 &pinctrl {
        gpio-leds {
                sys_led_pin: sys-led-pin {
@@ -537,6 +568,25 @@ pcie_m2_1_pwren: pcie-m21-pwren {
                };
        };
 
+       fspi {
+               /omit-if-no-ref/
+               fspim1_pins: fspim1-pins {
+                       rockchip,pins =
+                               /* fspi_clk_m1 */
+                               <2 RK_PB3 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_cs0n_m1 */
+                               <2 RK_PB4 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d0_m1 */
+                               <2 RK_PA6 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d1_m1 */
+                               <2 RK_PA7 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d2_m1 */
+                               <2 RK_PB0 2 &pcfg_pull_up_drv_level_2>,
+                               /* fspi_d3_m1 */
+                               <2 RK_PB1 2 &pcfg_pull_up_drv_level_2>;
+               };
+    };
+
        sdmmc {
                sd_s0_pwr: sd-s0-pwr {
                        rockchip,pins = <4 RK_PA5 RK_FUNC_GPIO &pcfg_pull_down>;
```

Stigby, 2023-07-25:
> I might have a patch for this issue, but I need help testing it. Please run the below commands on your NanoPC-T6 and let me know if `/dev/mtdblk*` exist after reboot.
>
> ```
> wget [...]

Morning. I ran the patch, but ` /dev/mtdblk*` still doesn't exist, even after rebooting the T6.

Before patch:
```
ubuntu@ubuntu:~$ ls /dev/nvme0n1
/dev/nvme0n1
ubuntu@ubuntu:~$ ls /dev/mtdblk*
ls: cannot access '/dev/mtdblk*': No such file or directory
```

Command:
```
ubuntu@ubuntu:~$ wget https://github.com/Joshua-Riek/ubuntu-rockchip/files/12152840/rk3588-nanopc-t6.zip
...
Length: 50907 (50K) [application/zip]
Saving to: ‘rk3588-nanopc-t6.zip’

rk3588-nanopc-t6.zip    100%[==============================>]  49.71K  --.-KB/s    in 0.1s

2023-07-25 11:08:48 (458 KB/s) - ‘rk3588-nanopc-t6.zip’ saved [50907/50907]

ubuntu@ubuntu:~$ unzip rk3588-nanopc-t6.zip
Archive:  rk3588-nanopc-t6.zip
  inflating: rk3588-nanopc-t6.dtb
ubuntu@ubuntu:~$ sudo cp rk3588-nanopc-t6.dtb /boot/firmware/rk3588-nanopc-t6.dtb
ubuntu@ubuntu:~$ sync
ubuntu@ubuntu:~$ sudo reboot now
```

After patch and reboot:
```
ubuntu@ubuntu:~$ ls /dev/mtdblk*
ls: cannot access '/dev/mtdblk*': No such file or directory
ubuntu@ubuntu:~$ sudo ls /dev/mtdblk*
ls: cannot access '/dev/mtdblk*': No such file or directory
```

Stigby, 2023-07-25:
> ls /dev/mtdblock* command will not help?
> 
> I also plan to use nvme on NanoPC-T6. Currently use it with eMMC, but want to switch to nvme disk.

Tried, didn't work.

Joshua-Riek, 2023-07-25:
I see, i will need the hardware to do more testing and come up with a proper patch for this. But you can still write the bootloader into the spi nor using the RKDevTool. That should allow you to boot from the nvme.

Stigby, 2023-07-25:
> I see, i will need the hardware to do more testing and come up with a proper patch for this. But you can still write the bootloader into the spi nor using the RKDevTool. That should allow you to boot from the nvme.

Okay. I'm pretty new to this whole SPI-NOR flashing thing, and flashing in general, so apologies if this sounds like a dumb question 😅. 

I've followed FriendlyWrt's and Raxdxa's documentation on flashing, but I still want to double-check with you that I'm not doing anything incorrectly (to avoid bricking my new T6). I've now installed Rockship's Driver Assistant (v5.1.1) and RKDevTool (v2.86) for Windows. In RKDevTool, am I supposed to place the bootloader image `...\u-boot-friendlyelec-rk3588\rkspi_loader.img` in the `loader` partition table and click Run, or is there anything else I should do still?

[image]

lee86, 2023-07-25:
[image]

1、Chose `Download Image`
2、Chose loadr bin file
(1)Click the blue area at the tail,Chose the file `loader.bin`
(2)Address 0x00000000
(3)Storage: blank
3、Chose UBOOT file
(1)Click the blue area at the tail,Chose the file `rkspi_loader.img`
(2)Address 0x00000000
(3)Storage:SPINOR
4、Chick `Write by Address`
5、Run will download loader.bin,and then write rkspi_loader.img into SPINOR

@Stigby 

What's more,if you want to test device,you also need to download the boot file `loader.bin`
[image]

1、Chose `Advanced Function`
2、Chose Boot file and Download
(1)Click the blue frame to chose the boot file
(2)Download
3、After Download Boot file,you can test your device

Stigby, 2023-07-25:
Hi, and thanks for the quick and detailed response @lee86! 

Where can I locate the `MiniLoaderAll.bin` file which is seen in your screenshots? The only files I have are `idbloader.img`, `rkspi_loader.img`, and `u-boot.itb`, which I copied from the `u-boot-friendlyelec-rk3588` directory.

lee86, 2023-07-26:
> Hi, and thanks for the quick and detailed response @lee86!
> 
> Where can I locate the `MiniLoaderAll.bin` file which is seen in your screenshots? The only files I have are `idbloader.img`, `rkspi_loader.img`, and `u-boot.itb`, which I copied from the `u-boot-friendlyelec-rk3588` directory on the T6.

My Orange pi 5 plus and T6 are both compatible with this file, you can try to download it to T6.
[MiniLoaderAll.bin.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12167301/MiniLoaderAll.bin.zip)
It may take effect.

Stigby, 2023-07-26:
> My Orange pi 5 plus and T6 are both compatible with this file, you can try to download it to T6. [MiniLoaderAll.bin.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12167301/MiniLoaderAll.bin.zip) It may take effect.

Thanks for the bootloader, @lee86 Works like a charm! I was able to test out the different options e.g. `TestDevice` and the `ListStorage` options in the Advanced Functions tab. Could you still provide a newer version RKDevTool (version 3.15)? I can't seem to find any newer versions besides versions 2.58 and 2.86. Version 2.86 and below are missing the `Storage` row.

[image]
[image]

lee86, 2023-07-26:
> Thanks for the bootloader, @lee86 Works like a charm! I was able to test out the different options e.g. `TestDevice` and the `ListStorage` options in the Advanced Functions tab. Could you still [...]

[RKDevTool_v3.15_for_window.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12171462/RKDevTool_v3.15_for_window.zip)
I have set the language to English.

Stigby, 2023-07-26:
> [RKDevTool_v3.15_for_window.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12171462/RKDevTool_v3.15_for_window.zip) I have set the language to English.

Thanks again, @lee86! You da real MVP. I successfully flashed the SPI-NOR, and my T6 is now booting straight from the NVMe. For anyone else wanting to do the same, I've uploaded a zip folder containing all of the files I used during flashing process. Thanks again, @lee86 and @Joshua-Riek.

ZIP file: [NanoPC-T6 files for flashing.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12172020/NanoPC-T6.files.for.flashing.zip)

The zip folder contains:

- DriverAssistant (v.5.1.1) [(from FriendlyElec)](https://download.friendlyelec.com/NanoPC-T6)
- RKDevTool (v3.15) (from @lee86)
- MiniLoaderAll.bin (from @lee86)
- U-boot-friendlyelec-rk3588 [(copied from Ubuntu Server 22.04.2 LTS, ubuntu-rockchip v.1.20)](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.20)
  - idbloader.img
  - rkspi_loader.img
  - u-boot.itb
 
[image]

lee86, 2023-07-26:
> > [RKDevTool_v3.15_for_window.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12171462/RKDevTool_v3.15_for_window.zip) I have set the language to English.
> 
> Thanks again, @lee86! You da real MVP. I successfully flashed the SPI-NOR, and my T6 is now booting straight from the NVMe. For anyone else wanting to do the same, I've uploaded a zip folder containing all the files I used during flashing process. Thanks again, @lee86 and @Joshua-Riek.
> 
> ZIP file: [NanoPC-T6 files for flashing.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12172020/NanoPC-T6.files.for.flashing.zip)
> 
> The zip folder contains:
> 
> * DriverAssistant (v.5.1.1) [(from FriendlyElec)](https://download.friendlyelec.com/NanoPC-T6)
> * RKDevTool (v3.15) (from @lee86)
> * MiniLoaderAll.bin (from @lee86)
> * U-boot-friendlyelec-rk3588 [(copied from Ubuntu Server 22.04.2 LTS, ubuntu-rockchip v.1.20)](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.20)
>   
>   * idbloader.img
>   * rkspi_loader.img
>   * u-boot.itb
> 
> [image]

Congratulations, great to hear your T6 is working, and these two files are provided by RockChip.   :)

Your job may help others, nice work！

Joshua-Riek, 2023-07-26:
Very good information here. I will try and document this process for users to follow easily.

closed 2023-08-01 by Joshua-Riek