Joshua-Riek/ubuntu-rockchip #279: Bluetooth disappears eventually

vladkozlov69, 2023-07-26:
I am using rtl8822ce WiFi+Bluetooth card. Sometimes hci0 device disappears from the system. Here is some info from dmesg:

```
[269003.692495] usb 4-1: USB disconnect, device number 2
[269003.692868] Bluetooth: hci0: failed to disable LE scan: status 0x1f
[269003.892128] usb 4-1: new full-speed USB device number 3 using ohci-platform
[269004.071974] usb 4-1: device descriptor read/64, error -62
[269004.371728] usb 4-1: device descriptor read/64, error -62
[269004.668228] usb 4-1: new full-speed USB device number 4 using ohci-platform
[269004.861325] usb 4-1: device descriptor read/64, error -62
[269005.161243] usb 4-1: device descriptor read/64, error -62
[269005.267931] usb usb4-port1: attempt power cycle
[269005.470924] usb 4-1: new full-speed USB device number 5 using ohci-platform
[269005.884087] usb 4-1: device not accepting address 5, error -62
[269006.060667] usb 4-1: new full-speed USB device number 6 using ohci-platform
[269006.473702] usb 4-1: device not accepting address 6, error -62
[269006.473785] usb usb4-port1: unable to enumerate USB device
```

Unfortunately only reboot helps.
On classic intel PC, where usb controller is connected via pci bus I am able to restart USB without rebooting system.
However on Rockchip SOC command "lspci | grep USB" returns nothing.

Is there a way to restart/reenumerate USB automatically? Maybe there could be a fix for this?

Joshua-Riek, 2023-07-31:
Try to remove the rtl8822ce module from your board then insert it again. But make sure it is properly inserted and secure. This looks like it the module may be loose?

vladkozlov69, 2023-08-03:
No, actually I observe the same behaviour on three PCs, so it seems to be rtl8822ce USB issue.

vladkozlov69, 2023-08-04:
Today it happened again - hci0 disappeared on one of my T6s. Guess what? I just turned my WiFi router off and on so rtl8822ce had to reconnect. Magically hci0 appeared in the system after WiFi reconnected. So it is definitely not a bad electrical contact, and looks like an rtl8822ce (in dmesg I see that firmware blob was reloaded in the wifi chip).

Will keep my eye on it, and maybe next time I'll try to reproduce the issue this way.

vladkozlov69, 2023-08-06:
> If this also happens on a normal pc, it looks like an issue with your dongle.

Sorry, when I said "PC" I meant Nano**PC**-T6 :) Three NanoPCs each equipped with rtl8822ce card.

Joshua-Riek, 2023-08-20:
Hey, sorry for only getting back to this now. When this issue happens, can you please put dmesg output into a file and upload it here? I have a few rtl8822ce cards I have been running with no issues so far.

vladkozlov69, 2023-08-31:
[dmesg-bt-failure.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12484959/dmesg-bt-failure.txt)

Here it is. The Home Assistant is running Xiaomi BLE integration using RTL8822CE and after several days it ends up with "usb usb4-port1: unable to enumerate USB device" and hci0 disappears completely until reboot.