Joshua-Riek/ubuntu-rockchip #280: Enhancement request: ability to run cinnamon

ewaldc, 2023-07-27:
I installed `cinnamon` for some development work on a clean v1.20 install with all upgrades performed .  Cinnamon installs without errors, but fails to start the cinnamon window manager with either gdm3 and/or lightdm display manager.  When starting cinnamon manually as `cinnamon --replace`, the error recieved is:
`Clutter-CRITICAL: Unable to initialize Clutter: Unable to initialize Clutter backend: no available drivers found.
Cinnamon error: Unable to initialize Clutter`.

However, when starting `cinnamon` as root using `sudo cinnamon --replace`, cinnamon starts fine.  The cinnamon apps (e.g. nemo, gedit) run without issues as regular user. Hence, it must be some small issue...
I have not been able to find any other related errors in Xorg.log or journalctl which could point to a permission or other issue.

There is a working Cinnamon Armbian Ubuntu version which I am using for now for Cinnamon development work , but it does not run as smootly as yours!  
Happy to figure it out myself, but I was hoping you might have some suggestions...
Thanks in advance

Joshua-Riek, 2023-07-27:
Currently, I have the system optimized out of the box for the Wayland display manager as it is the most mature desktop environment and the best-performing environment for the available GPU drivers. The Mesa Panfork GPU drivers will most likely need to be uninstalled entirely for this to work, which again would be another performance hit. 

I have been against supporting anything other than Gnome Wayland due to all the GPU and performance issues. However, I must say that I am curious about why Cinnamon is having some issues, I can take a look but please do not expect a smooth experience compared to Gnome.

ewaldc, 2023-07-27:
100% agreed, no expectation of any level of support, nor smooth performance.  Today I have a dedicated flash card with Armbian Ubuntu Cinnamon and it is not smooth, but I am using it only to test development builds of Cinnamon.
Everything else (development, compilation, running ppa etc.), happens on the v1.20 instance with NVME already.
It was an attempt to get rid of that extra installation.  I can already run and test all Cinnamon tools (gedit, nemo, menu editor) directly on Gnome 3 Wayland and they run very smooth !  
Only with the Cinnamon Window Manager there is somewhere a permission issue, just could not find any indication where it's going wrong.

Joshua-Riek, 2023-08-20:
Sorry for only getting back to this now, but I have limited time at the moment and do not have any plans in the future to add support for cinnamon.

closed 2023-08-20 by Joshua-Riek