Joshua-Riek/ubuntu-rockchip #282: Orange Pi 5 Plus with Intel AX210 wifi/Bluetooth Card & Video Play Back (Youtube and Netflix)

va3byw, 2023-07-28:
Thank you for the OS image. It works very well but with some issues on SBC.

1. Bluetooth firmware cannot be loaded and below is the related error message in Logs. WIFI works fine though.
    Bluetooth: hci0: Failed to load Intel firmware file (-2).
2. When playing Youtube videos, it shows "green" screen when some of videos are in av1 or avc1 (not always), but audios are normal.
3. When playing Netflix, it says "This title is not available to watch instantly. Please try another title." with Error Code M7111-1331. I tried the fixes provided on Netflix web site but no luck.

Regards,

Bob

Joshua-Riek, 2023-07-28:
1. Probably missing firmware files; please try the below then reboot your system to load the firmware.
```
sudo wget -P /lib/firmware/intel https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/intel/ibt-0041-0041.sfi
sudo wget -P /lib/firmware/intel https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/intel/ibt-0041-0041.ddc
```
2. There is nothing I can do about this. The current GPU drivers are a reverse-engineered hack and will not be further improved or developed. We will need to wait for mainline Linux support and Mesa GPU drivers to be available. I expect this to work in a _user-friendly_ state in 1-2 years. From my understanding, the green screen is due to unsupported video resolutions. 
3. You must spoof the browser, open a terminal, and launch chromium with `chromium-browser --user-agent="Mozilla/5.0 (X11; CrOS aarch64 15236.80.0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/110.0.0.0 Safari/537.36"`

bagong, 2023-11-24:
I assume that until the rk35888 boards can be considered fully Linux-supported in a user-friendly way, we will have to go through two further phases: introduction of the forthcoming 6.1 kernel from rockchip and it's stabilisation/enhancement, and then the introduction of mainline-support. Both "new kernels" will be followed by transitional periods in which bugs are identified and fixed or worked around based on user-feedback. I think after the introduction of the new 6.1 kernel it would be lovely to start collecting "tips"/"workarounds" and the like that are currently spread over quite a few issues (like here), in the Wiki. Maybe at some point we could discuss a workflow how such "solutions" could be centrally collected in a colaborative way.