Joshua-Riek/ubuntu-rockchip #283: ax210  wireless card not work on rock-5b

iamryanchia, 2023-07-29:
Hi, Is ax210 supported？I found my ax210 does not work. Here is the output:

```
sudo iwconfig
wlP2p33s0  IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=-2147483648 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on

sudo iwlist wlP2p33s0 scanning
wlP2p33s0  Interface doesn't support scanning : Invalid argument
```

I have ran these and reboot, but it does not help.

```
#wifi
sudo wget -P /lib/firmware https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/iwlwifi-ty-a0-gf-a0-59.ucode
sudo mv /lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm /lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm.bak

#bt
sudo wget -P /lib/firmware/intel https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/intel/ibt-0041-0041.sfi
sudo wget -P /lib/firmware/intel https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/intel/ibt-0041-0041.ddc
```

If it is not supported yet, have any plan to support? Thank you!

Joshua-Riek, 2023-07-30:
I assume you are using the server version which uses Netplan. Please follow the below steps to connect to wifi. Bluetooth firmware is missing but does work after it is added and you reboot the system.

Edit the Netplan configuration file:
```
sudo nano /etc/netplan/50-cloud-init.yaml
```
Ensure that the wifis block is aligned with the above ethernets or version block if present. The entire configuration file may look similar to the one below:
```
# This file is generated from information provided by the datasource.  Changes
# to it will not persist across an instance reboot.  To disable cloud-init's
# network configuration capabilities, write a file
# /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg with the following:
# network: {config: disabled}
network:
    ethernets:
        eth0:
            dhcp4: true
            optional: true
    version: 2
    wifis:
        wlP2p33s0:
            optional: true
            access-points:
                "SSID-NAME-HERE":
                    password: "PASSWORD-HERE"
            dhcp4: true
```
Once ready, apply the changes and connect to your wireless interface by executing the below command:
```
sudo netplan apply
```
Finally reboot and wifi should work.

iamryanchia, 2023-07-30:
I am using preinstalled dekstop version. Does it manage network by netplan? no file under /etc/netplan.

iamryanchia, 2023-07-30:
Than you! I create a new file and put these contents. wifi works now.

Joshua-Riek, 2023-07-30:
Awesome, I will fix the Bluetooth and add the proper firmware later today.

Joshua-Riek, 2023-07-30:
I updated the firmware package on my ppa to include the Bluetooth firmware for the AX210. 

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip

closed 2023-07-30 by Joshua-Riek