Joshua-Riek/ubuntu-rockchip #285: orange Pi-5Plus can't run  with Desktop in EMMC

aisyun, 2023-07-29:
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi-rk3588 (Jul 23 2023 - 21:23:18)
Trying to boot from MMC1
spl: mmc init failed with error: -123
Trying to boot from MMC2
Trying fit image at 0x4000 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(d21d14728d...) + OK
## Checking uboot 0x00200000 ... sha256(1a67e315fb...) + OK
## Checking fdt 0x003129a0 ... sha256(d2dda3ef7c...) + OK
## Checking atf-2 0x000f0000 ... sha256(c00c7fd75b...) + OK
## Checking atf-3 0xff100000 ... sha256(71c3a5841b...) + OK
## Checking atf-4 0xff001000 ... sha256(2301cf73be...) + OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 513.190 ms

5.228463] debugfs: File 'Capture' in directory 'dapm' already present!
[    5.251034] ES8323 7-0011: ASoC: error at soc_component_write_no_lock on ES8323.7-0011: -5
[    5.355846] debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
[    5.364021] rockchip-dmc dmc: failed to get vop bandwidth to dmc rate
[    5.369668] rockchip-dmc dmc: failed to get vop pn to msch rl
[    5.378209] rockchip-dmc dmc: could not find power_model node
[    5.390817] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdab0000-0xfdabffff]
[    5.393953] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdac0000-0xfdacffff]
[    5.396956] RKNPU fdab0000.npu: can't request region for resource [mem 0xfdad0000-0xfdadffff]
[    5.405509] debugfs: Directory 'fdab0000.npu-rknpu' with parent 'vdd_npu_s0' already present!
[    5.432614] RKNPU fdab0000.npu: failed to find power_model node
[    5.435517] RKNPU fdab0000.npu: RKNPU: failed to initialize power model
[    5.438382] RKNPU fdab0000.npu: RKNPU: failed to get dynamic-coefficient
[    7.076032] rk-pcie fe150000.pcie: PCIe Link Fail
[    7.078506] rk-pcie fe150000.pcie: failed to initialize host
[    7.122716] rk-pcie fe170000.pcie: PCIe Link Fail
[    7.125170] rk-pcie fe170000.pcie: failed to initialize host

Joshua-Riek, 2023-07-29:
This should be working fine, but i will double check.

Joshua-Riek, 2023-07-29:
I checked this out and saw no problem. Please ensure you have a proper 5v 4a power supply or try to write the os to your emmc again.

closed 2023-07-29 by Joshua-Riek

aisyun, 2023-07-29:
but the serial port  infomation is stop in:

```
 7.076032] rk-pcie fe150000.pcie: PCIe Link Fail
[ 7.078506] rk-pcie fe150000.pcie: failed to initialize host
[ 7.122716] rk-pcie fe170000.pcie: PCIe Link Fail
[ 7.125170] rk-pcie fe170000.pcie: failed to initialize host
```

and no HDMI video output.
test in v.1.20-v1.21 with the same problem.
so the Destop is not run successful

Joshua-Riek, 2023-07-29:
There are EDID issues with many monitors, which is a kernel issue. Please try with a 1080p, 2k, or 4k monitor. What is the resolution of your monitor?