Joshua-Riek/ubuntu-rockchip #286: Orangepi 5 plus Bluetooth scan returns no devices

ndrewpj, 2023-07-30:
Hi! I am using the Orangi pi 5 plus 'official' WIFI/Bluetooth PCIe card with the latest Ubuntu release here (1.21). WIFI is working great, the BT devices on OPi is detected as a device but shows no devices in the list, it shows non-stop 'scanning' icon in the BT menu.

Joshua-Riek, 2023-07-30:
I have not noticed an official WIFI/Bluetooth card. Could you point me to it? I would be happy to add support. My guess is the Bluetooth firmware needs to be added.

ndrewpj, 2023-07-30:
> I have not noticed an official WIFI/Bluetooth card. Could you point me to it? I would be happy to add support. My guess is the Bluetooth firmware needs to be added.

Sure, it is RTL8852BE M.2 :
https://sl.aliexpress.ru/p?key=WWPDZUc

Joshua-Riek, 2023-07-30:
Hmm it looks like firmware is installed, could you provide me with dmesg logs? You can type `dmesg > dmesg.txt` in console and then upload the file here.

Joshua-Riek, 2023-07-30:
I looked at the kernel source and the Bluetooth driver is not included, I will need to port it. This will take some time because I need to order the hardware for testing.

dongxuanlb, 2023-07-31:
@ndrewpj you may try this:

root@rock-5b:~# cat /etc/modprobe.d/blacklist.conf 
blacklist pgdrv
blacklist btusb
blacklist btrtl
blacklist btbcm
blacklist btintel

root@rock-5b:~# reboot

which from the rock5b or rock5a

Joshua-Riek, 2023-07-31:
I have a patch https://github.com/Joshua-Riek/linux-rockchip/commit/aa0c32e0113936996908ec3431fc359182f9b1d8 that is not tested but should work for rtl8852be.

ndrewpj, 2023-08-01:
> I have a patch [Joshua-Riek/linux-rockchip@aa0c32e](https://github.com/Joshua-Riek/linux-rockchip/commit/aa0c32e0113936996908ec3431fc359182f9b1d8) that is not tested but should work for rtl8852be.

How can I apply it?

Joshua-Riek, 2023-08-01:
> > I have a patch [Joshua-Riek/linux-rockchip@aa0c32e](https://github.com/Joshua-Riek/linux-rockchip/commit/aa0c32e0113936996908ec3431fc359182f9b1d8) that is not tested but should work for rtl8852be.
> 
> How can I apply it?

You will need to build the kernel, I can't give instructions right now. But I can build it later today and give you a link to download and install it with the patch.

Joshua-Riek, 2023-08-01:
My dev build is finished. Please try installing it with the below steps, then reboot your system. Let me know if Bluetooth is working after reboot.
```
wget https://launchpadlibrarian.net/679939334/linux-image-5.10.160-rockchip_5.10.160-8_arm64.deb
sudo dpkg -i linux-image-5.10.160-rockchip_5.10.160-8_arm64.deb
sync
sudo reboot now
```

ndrewpj, 2023-08-02:
Nope, this did not help

Joshua-Riek, 2023-08-02:
I see. Thanks for the feedback. I have the  card ordered. Once it's delivered, I can debug.

Joshua-Riek, 2023-08-10:
I did some testing, and it looks like adding the below text to `/etc/modprobe.d/blacklist.conf` and rebooting will allow Bluetooth and WiFi to work correctly.

```
blacklist pgdrv
blacklist btusb
blacklist btrtl
blacklist btbcm
blacklist btintel
```

closed 2023-08-10 by Joshua-Riek

TuTuc0, 2023-08-10:
Thanks a lot. Bluetooth is now fixed. However wifi is still unable to find any networks.
I have also noticed, that there are two wifi devices displayed:  PCI Wifi and Wifi.

RobertTutaev, 2023-08-10:
Me too.
Sometimes when you press the power button twice consecutively, wi-fi turns on.

Joshua-Riek, 2023-08-10:
Did you both do as described in https://github.com/Joshua-Riek/ubuntu-rockchip/issues/286#issuecomment-1672414608? It should be working fine after a reboot.

RobertTutaev, 2023-08-10:
I did as recommended.

Joshua-Riek, 2023-08-10:
Weird, I'm not sure why that would happen, it works on my end.

RobertTutaev, 2023-08-10:
Experimented with turning on the device.

Wifi failed startup log:
`23:40:41 kernel: hid-generic 0003:09DA:0024.0005: input,hiddev97,hidraw4: USB HID v1.01 Mouse [A4tech A4tech 2.4G Wireless Device] on usb-fc8c0000.usb-1/input1
23:40:41 kernel: input: A4tech A4tech 2.4G Wireless Device Consumer Control as /devices/platform/fc8c0000.usb/usb4/4-1/4-1:1.1/0003:09DA:0024.0005/input/input15
23:40:41 kernel: hid-generic 0003:09DA:0024.0004: input,hidraw3: USB HID v1.01 Keyboard [A4tech A4tech 2.4G Wireless Device] on usb-fc8c0000.usb-1/input0
23:40:41 kernel: input: A4tech A4tech 2.4G Wireless Device as /devices/platform/fc8c0000.usb/usb4/4-1/4-1:1.0/0003:09DA:0024.0004/input/input13
23:40:41 kernel: usb 4-1: Manufacturer: A4tech
23:38:39 kernel: Bluetooth: RFCOMM ver 1.11
23:38:39 kernel: dwhdmi-rockchip fdea0000.hdmi: Rate 185580000 missing; compute N dynamically
23:38:29 kernel: vcc_mipicsi1: disabling
23:38:29 kernel: vcc_mipicsi0: disabling
23:38:21 kernel: rtk_btusb: btusb_notify: hci0 evt 2
23:38:13 kernel: ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
23:38:10 kernel: mali fb000000.gpu: Mali firmware git_sha: a9b9d413d30169284dfbc5c68e8c65fdaeca28b7 
23:38:10 kernel: mali fb000000.gpu: Mali firmware git_sha: a9b9d413d30169284dfbc5c68e8c65fdaeca28b7 
23:38:10 kernel: mali fb000000.gpu: Loading Mali firmware 0x1010000
23:38:10 kernel: dwhdmi-rockchip fdea0000.hdmi: Rate 185580000 missing; compute N dynamically
23:38:07 kernel: dw-apb-uart febc0000.serial: failed to request DMA, use interrupt mode
23:38:07 kernel: of_dma_request_slave_channel: dma-names property of node '/serial@febc0000' missing or empty
23:38:07 kernel: [BT_RFKILL]: ** disable irq
23:38:02 kernel: [WARNING][RF][halrf_config_8852b_radio_b_reg]Invalid RF CR Pkg
23:38:02 kernel: [WARNING][RF][halrf_config_8852b_radio_a_reg]Invalid RF CR Pkg
23:38:02 kernel: enP4p65s0: 0xffff80000bb00000, c0:74:2b:ff:5e:8a, IRQ 149
23:38:02 kernel: enP3p49s0: 0xffff80000bae0000, c0:74:2b:ff:5e:8b, IRQ 139
23:38:02 kernel: rtk_btusb: btusb_notify: hci0 evt 3
23:38:02 kernel: Bluetooth: BNEP socket layer initialized
23:38:02 kernel: rtk_btusb: HCI reset.
23:38:02 kernel: usbcore: registered new interface driver rtk_btusb
23:38:02 kernel: rtk_btusb: HCI reset.
23:38:02 kernel: Process accounting resumed
23:38:02 kernel: usb 5-1.1: SerialNumber: 00e04c000001
23:38:02 kernel: rtl8852be 0002:21:00.0 wlP2p33s0: renamed from wlan0
23:38:02 kernel: eric-tx CALL alloc_txring !!!!
23:38:01 kernel: usb 5-1.1: new full-speed USB device number 3 using xhci-hcd
23:38:01 kernel: FAT-fs (mmcblk0p1): utf8 is not a recommended IO charset for FAT filesystems, filesystem will be case sensitive!
23:38:01 kernel: [BT_RFKILL]: ** disable irq
23:38:01 kernel: pwm-fan pwm-fan: Looking up fan-supply property in node /pwm-fan failed
23:38:01 kernel:  el0_sync+0x1a0/0x1c0
23:38:01 kernel: Call trace:
23:38:01 kernel: Hardware name: Orange Pi 5 Plus (DT)
23:38:01 kernel: CPU: 6 PID: 556 Comm: systemd-udevd Not tainted 5.10.160-rockchip #9
23:38:01 kernel:  lock: 0xffff80000b895080, .magic: 00000000, .owner: <none>/-1, .owner_cpu: 0
23:38:01 kernel: BUG: spinlock bad magic on CPU#6, systemd-udevd/556
23:38:01 kernel: rtl8852be 0002:21:00.0: enabling device (0000 -> 0003)
23:38:01 kernel: rtkm: rtkm_prealloc_init: done ret=0
23:38:01 kernel: Started Journal Service.
23:38:01 kernel: squashfs: version 4.0 (2009/01/31) Phillip Lougher
23:38:01 kernel: Mounting Mount unit for snapd, revision 19459...
23:38:01 kernel: Reached target Preparation for Local File Systems.
23:38:01 kernel: Finished Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling.
23:38:01 kernel: Started Device-Mapper Multipath Device Controller.
23:38:01 kernel: Starting Rule-based Manager for Device Events and Files...
23:38:01 kernel: Finished Create Static Device Nodes in /dev.
23:38:01 kernel: Starting Create Static Device Nodes in /dev...
23:38:01 kernel: Reached target Swaps.`

Wifi successful launch log:
`23:46:11 gdm-session-wor: GLib-GObject: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
23:45:55 gnome-session-b: GLib-GIO-CRITICAL: g_bus_get_sync: assertion 'error == NULL || *error == NULL' failed
23:45:54 wpa_supplicant: bgscan simple: Failed to enable signal strength monitoring
23:45:50 kernel: of_dma_request_slave_channel: dma-names property of node '/serial@febc0000' missing or empty
23:45:45 kernel: rtk_btusb: This is not 8723a, use new patch style!
23:45:44 systemd-udevd: wlP2p33s0: Failed to rename network interface 4 from 'wlan0' to 'wlP2p33s0': File exists
23:45:44 kernel:  lock: 0xffff80000ba1d080, .magic: 00000000, .owner: <none>/-1, .owner_cpu: 0
23:45:44 kernel: BUG: spinlock bad magic on CPU#6, systemd-udevd/529
23:45:43 kernel: rk-pcie fe150000.pcie: failed to initialize host
23:45:43 kernel: RKNPU fdab0000.npu: RKNPU: failed to get dynamic-coefficient
23:45:43 kernel: debugfs: Directory 'fdab0000.npu-rknpu' with parent 'vdd_npu_s0' already present!
23:45:43 kernel: RKNPU fdab0000.npu: can't request region for resource [mem 0xfdad0000-0xfdadffff]
[... 3 lines cut ...]
23:45:43 kernel: debugfs: File 'Capture' in directory 'dapm' already present!
23:45:43 kernel: arm-scmi firmware:scmi: Failed. SCMI protocol 17 not active.
23:45:43 kernel: Goodix-TS 7-0014: I2C communication failure: -6
23:45:43 kernel: .. rk pwm remotectl v2.0 init
23:45:43 kernel: Error: Driver 'Goodix-TS' is already registered, aborting...
[... cut ...]
23:45:43 kernel: rkvdec2_init:1022: No niu aclk reset resource define
23:45:43 kernel: mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_a is not found!
23:45:43 kernel: mpp-iep2 fdbb0000.iep: allocate roi buffer failed
23:45:43 kernel: rk-pcie fe190000.pcie: invalid resource
23:45:43 kernel: debugfs: Directory 'vcc_mipicsi1' with parent 'vcc_mipicsi1' already present!
23:45:43 kernel: fiq_debugger_probe: could not install nmi irq handler
23:45:43 kernel: fiq_debugger fiq_debugger.0: IRQ wakeup not found`

RobertTutaev, 2023-08-10:
Wifi failed startup log 2:
```
01:25:17 gdm-session-wor: GLib-GObject: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
01:25:13 systemd: Failed to start Application launched by gnome-session-binary.
01:25:13 gdm-session-wor: gkr-pam: unable to locate daemon control file
01:24:57 gnome-session-b: GLib-GIO-CRITICAL: g_bus_get_sync: assertion 'error == NULL || *error == NULL' failed
01:24:48 kernel: of_dma_request_slave_channel: dma-names property of node '/serial@febc0000' missing or empty
01:24:43 kernel: rtk_btusb: This is not 8723a, use new patch style!
01:24:43 systemd-udevd: wlP2p33s0: Failed to rename network interface 4 from 'wlan0' to 'wlP2p33s0': File exists
01:24:42 kernel:  lock: 0xffff80000ba65080, .magic: 00000000, .owner: <none>/-1, .owner_cpu: 0
01:24:42 kernel: BUG: spinlock bad magic on CPU#7, systemd-udevd/539
01:24:42 kernel: rk-pcie fe150000.pcie: failed to initialize host
01:24:42 kernel: rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy lane locked!
01:24:42 kernel: RKNPU fdab0000.npu: RKNPU: failed to get dynamic-coefficient
01:24:42 kernel: debugfs: Directory 'fdab0000.npu-rknpu' with parent 'vdd_npu_s0' already present!
01:24:42 kernel: RKNPU fdab0000.npu: can't request region for resource [mem 0xfdad0000-0xfdadffff]
01:24:42 kernel: rockchip-dmc dmc: could not find power_model node
01:24:42 kernel: debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
01:24:42 kernel: ES8323 7-0011: ASoC: error at soc_component_write_no_lock on ES8323.7-0011: -5
01:24:42 kernel: debugfs: File 'Capture' in directory 'dapm' already present!
01:24:42 kernel: arm-scmi firmware:scmi: Failed. SCMI protocol 17 not active.
01:24:42 kernel: Goodix-TS 7-0014: I2C communication failure: -6
01:24:42 kernel: .. rk pwm remotectl v2.0 init
01:24:42 kernel: Error: Driver 'Goodix-TS' is already registered, aborting...
01:24:42 kernel: rk806 spi2.0: no dvs-setting pinctrl state
01:24:42 kernel: rockchip-hdptx-phy-hdmi fed70000.hdmiphy: hdptx phy lane locked!
01:24:42 kernel: rk-pcie fe170000.pcie: invalid resource
01:24:42 kernel: rkvenc_init:1814: failed to add venc devfreq
01:24:42 kernel: mpp_rkvenc2 fdbe0000.rkvenc-core: dev_pm_opp_set_regulators: no regulator (venc) found: -19
01:24:42 kernel: rkvenc_init:1814: failed to add venc devfreq
01:24:42 kernel: mpp_rkvenc2 fdbd0000.rkvenc-core: dev_pm_opp_set_regulators: no regulator (venc) found: -19
01:24:42 kernel: rkvdec2_init:1025: No niu hclk reset resource define
01:24:42 kernel: mpp_rkvdec2 fdc48100.rkvdec-core: shared_niu_h is not found!
01:24:42 kernel: rkvdec2_init:1022: No niu aclk reset resource define
01:24:42 kernel: mpp_rkvdec2 fdc48100.rkvdec-core: shared_niu_a is not found!
01:24:42 kernel: rkvdec2_init:1025: No niu hclk reset resource define
01:24:42 kernel: mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_h is not found!
01:24:42 kernel: rkvdec2_init:1022: No niu aclk reset resource define
01:24:42 kernel: mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_a is not found!
01:24:42 kernel: mpp-iep2 fdbb0000.iep: allocate roi buffer failed
01:24:42 kernel: rk-pcie fe180000.pcie: invalid resource
01:24:42 kernel: debugfs: Directory 'vcc_mipicsi1' with parent 'vcc_mipicsi1' already present!
01:24:42 kernel: fiq_debugger_probe: could not install nmi irq handler
01:24:42 kernel: fiq_debugger fiq_debugger.0: IRQ wakeup not found
```

chemradio, 2023-08-17:
Have the same issue. After a clean install - WiFi works as expected, Bluetooth does not work at all.

After adding to `/etc/modprobe.d/blacklist.conf` these lines:

```
blacklist btusb
blacklist btrtl
blacklist btbcm
blacklist btintel
```
but not adding `blacklist pgdrv`
I am able to run both Bluetooth And WiFi. However WiFi works well only if the bluetooth was **disabled** before previous shutdown. If i shut down with bluetooth keyboard connected, WiFi does not see any networks on the next boot up. How can I solve this? Can you help me please?

RobertTutaev, 2023-08-17:
We need a new issue.
The current issue is closed.