Joshua-Riek/ubuntu-rockchip #287: Mitigating CVE-2023-32629 and CVE-2023-2640 Vulnerabilities in OverlayFS

laamalif, 2023-07-31:
Recently, two critical security vulnerabilities, namely CVE-2023-32629 and CVE-2023-2640, have been discovered in the OverlayFS filesystem, affecting Linux kernel versions prior to 5.4. As these vulnerabilities pose significant risks to all images security.

Joshua-Riek, 2023-08-01:
The kernel is a massive mess of security risks and vulnerabilities as it's based on Android and not your typical LTS kernel used for Linux. Currently, no alternative offers total hardware support. Addressing security vulnerabilities is sadly a lost cause, as massive underlying issues exist.

I do not like this, but we are stuck in this situation.

laamalif, 2023-08-01:
Based on the v1.16 changelog, my understanding was that the kernel update from version 5.10.110 to 5.10.160 in the release was aligned with the LTS (Long-Term Support) version 5.10.x.

I came across a fix in the Ubuntu mailing list that addresses CVE-2023-32629. The link to the fix is provided here: 
https://lists.ubuntu.com/archives/kernel-team/2023-July/140920.html

According to the Debian Security tracker, the issue with CVE-2023-32629 is marked as "not-affected" for the "linux" package, indicating that it is an Ubuntu-specific matter.

Reference: https://security-tracker.debian.org/tracker/CVE-2023-32629

Can we safely assume that since this kernel was not based on the Ubuntu kernel, these vulnerabilities may not exist in this kernel version?

Joshua-Riek, 2023-08-01:
I looked at the kernel, and [CVE-2023-32629](https://github.com/advisories/GHSA-5xvw-32xh-2vr5) and [CVE-2023-2640](https://github.com/advisories/GHSA-38f7-vv5r-859m) does not exist in this kernel version.

However, I'm sure this kernel is **filled** with other CVE issues not addressed due to its Android origin. When I learned of this, I made sure to remove any mention of LTS in the project description. It will be a long time before a stable LTS kernel is available for the RK3588 platform.

closed 2023-08-01 by Joshua-Riek