Joshua-Riek/ubuntu-rockchip #288: RADXA-CAMERA-4K gives green picture on ROCK5B

zoldaten, 2023-08-01:
hi!
i try to use camera imx415 like this https://wiki.radxa.com/Rock5/accessories/radxa-camera-4k.
But it gives picture in dark green colors:
[image]

**what i have done.**
Following https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays i have added to 
/boot/firmware/ubuntuEnv.txt overlay:
`overlays=rock-5b-radxa-camera-4k`

Following https://forum.radxa.com/t/how-to-use-radxa-camera-4k/13031/2 i checked camera:
```
rock@rock:~$ gst-inspect-1.0 |grep mpp
rockchipmpp:  mpph264enc: Rockchip Mpp H264 Encoder
rockchipmpp:  mpph265enc: Rockchip Mpp H265 Encoder
rockchipmpp:  mppjpegdec: Rockchip's MPP JPEG image decoder
rockchipmpp:  mppjpegenc: Rockchip Mpp JPEG Encoder
rockchipmpp:  mppvideodec: Rockchip's MPP video decoder
rockchipmpp:  mppvp8enc: Rockchip Mpp VP8 Encoder
rockchipmpp:  mppvpxalphadecodebin: VP8/VP9 Alpha Decoder
typefindfunctions: audio/x-musepack: mpc, mpp, mp+

v4l2-ctl --device /dev/video11 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'UYVY' (UYVY 4:2:2)
		Size: Stepwise 32x32 - 3840x2160 with step 8/8
	[1]: 'NV16' (Y/CbCr 4:2:2)
		Size: Stepwise 32x32 - 3840x2160 with step 8/8
	[2]: 'NV61' (Y/CrCb 4:2:2)
		Size: Stepwise 32x32 - 3840x2160 with step 8/8
	[3]: 'NV21' (Y/CrCb 4:2:0)
		Size: Stepwise 32x32 - 3840x2160 with step 8/8
	[4]: 'NV12' (Y/CbCr 4:2:0)
		Size: Stepwise 32x32 - 3840x2160 with step 8/8
	[5]: 'NM21' (Y/CrCb 4:2:0 (N-C))
		Size: Stepwise 32x32 - 3840x2160 with step 8/8
	[6]: 'NM12' (Y/CbCr 4:2:0 (N-C))
		Size: Stepwise 32x32 - 3840x2160 with step 8/8

```
to work i use this code:
```gst-launch-1.0 v4l2src device=/dev/video11 ! video/x-raw,format=NV12,width=3840,height=2160, framerate=30/1 ! xvimagesink```

may be my problem corresponds with this issue - 
https://forum.radxa.com/t/imx219-rock-3a-rkaiq-3a-server-how-it-made-working-where-get-correct-calibration-files/11961
but how to fix that ?

Joshua-Riek, 2023-08-01:
I don't know much about how cameras work on the system, but it seems related to #254 and #232. From what I have seen is that the camera works best on Debian. 

You can try downloading a different version of the camera engine as talked about here https://github.com/Joshua-Riek/ubuntu-rockchip/issues/232#issuecomment-1612206015. But this has had mixed results. I don't have the radxa camera.

zoldaten, 2023-08-03:
i just installed https://github.com/radxa/debos-radxa/raw/main/rootfs/packages/arm64/camera/camera-engine-rkaiq_3.0.2_arm64.deb
and it starts working:
[image]

but after reboot i see again green:
[image]

rkaiq_3A.service gives following:
```
systemctl status rkaiq_3A.service
● rkaiq_3A.service - rkisp 3A engine
     Loaded: loaded (/lib/systemd/system/rkaiq_3A.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-08-03 11:59:01 MSK; 43min ago
    Process: 813 ExecStart=/etc/init.d/rkaiq_3A.sh start (code=exited, status=0/SUCCESS)
      Tasks: 7 (limit: 4370)
     Memory: 13.8M
     CGroup: /system.slice/rkaiq_3A.service
             ├─816 /usr/bin/rkaiq_3A_server
             └─817 logger -t rkaiq

авг 03 12:38:10 rock rkaiq[817]: DBG: device manager start
авг 03 12:38:10 rock rkaiq[817]: DBG: rkisp_init engine succeed
авг 03 12:38:10 rock rkaiq[817]: DBG: /dev/media1: wait stream stop event...
авг 03 12:38:10 rock rkaiq[817]: [12:38:10.224096][CAMHW]:XCAM ERROR Isp20Params.cpp:4036: get blc params failed!
авг 03 12:38:10 rock rkaiq[817]: 
авг 03 12:38:11 rock rkaiq[817]: DBG: /dev/media1: wait stream stop event success ...
авг 03 12:38:11 rock rkaiq[817]: [12:38:11.104728][CAMHW]:XCAM ERROR SensorHw.cpp:1291: failed to set sync mode 0
авг 03 12:38:11 rock rkaiq[817]: DBG: subscribe events from /dev/video19 success !
авг 03 12:38:11 rock rkaiq[817]: DBG: subscribe events from /dev/video19 success !
авг 03 12:38:11 rock rkaiq[817]: DBG: /dev/media1: wait stream start event...
```

erd-yunmu, 2023-08-24:
Please try to replace the deb, I test imx415 on rk3588s device is ok
[image]

zoldaten, 2023-08-26:
link is dead. but i found it here -
https://github.com/erd-yunmu/ubuntu-rockchip/blob/main/packages/rkaiq/camera-engine-rkaiq_rk3588_arm64.deb

after install i does not work, rka server does not start:
`
rock@rock:~$ systemctl status rkaiq_3A.service
○ rkaiq_3A.service - Enable Rockchip camera engine rkaiq
     Loaded: loaded (/lib/systemd/system/rkaiq_3A.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sat 2023-08-26 20:05:22 MSK; 1min 15s ago
    Process: 869 ExecStart=/etc/init.d/rkaiq_3A.sh start (code=exited, status=0/SUCCESS)
    Process: 964 ExecStop=/etc/init.d/rkaiq_3A.sh stop (code=exited, status=0/SUCCESS)

авг 26 20:05:22 rock rkaiq[871]: 
авг 26 20:05:22 rock rkaiq[871]: W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_ADAPTIVE_SEARCH >
авг 26 20:05:22 rock rkaiq[871]: 
авг 26 20:05:22 rock rkaiq[871]: W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_ADAPTIVE_SEARCH >
авг 26 20:05:22 rock rkaiq[871]: 
авг 26 20:05:22 rock rkaiq[871]: W: iq_parser_v2/j2s/j2s.cpp(895) [_j2s_json_to_obj]: missing len in json for dynamic array 'measiso'
авг 26 20:05:22 rock rkaiq[871]: 
авг 26 20:05:22 rock rkaiq[871]: ERR: rkaiq engine prepare failed !
авг 26 20:05:22 rock rkaiq_3A.sh[966]: rkaiq_3A_server: no process found
авг 26 20:05:22 rock systemd[1]: rkaiq_3A.service: Deactivated successfully.

`

the picture is green.

erd-yunmu, 2023-08-26:
@zoldaten delete rkaiq_3A.service `sudo rm -rf /usr/bin/rkaiq_3A_server`
then install this deb
because deb is dpkg -x + deb + folder

zoldaten, 2023-08-27:
No, not working (

`
rock@5:/usr/bin$ ./rkaiq_3A_server 
media get entity by name: rkisp-isp-subdev is null
media get entity by name: rkisp-input-params is null
media get entity by name: rkisp-statistics is null
media get entity by name: rkisp_mainpath is null
Cound not find rkisp dev names, skipped /dev/media0
ERR: Bad media topology for: /dev/media0
DBG: get rkisp-isp-subdev devname: /dev/v4l-subdev3
DBG: get rkisp-input-params devname: /dev/video19
DBG: get rkisp-statistics devname: /dev/video18
DBG: get rkisp_mainpath devname: /dev/video11
rkaiq log level ff0

************************** VERSION INFOS **************************
version release date: 2023-07-06
         AIQ:       AIQ v5.0x1.3
   IQ PARSER:       Calib v1.4.8,magicCode:1170944
************************ VERSION INFOS END ************************
ERR: Bad media topology for: /dev/media2
ERR: Bad media topology for: /dev/media3
ERR: Bad media topology for: /dev/media4
ERR: Bad media topology for: /dev/media5
ERR: Bad media topology for: /dev/media6
ERR: Bad media topology for: /dev/media7
ERR: Bad media topology for: /dev/media8
ERR: Bad media topology for: /dev/media9
ERR: Bad media topology for: /dev/media10
ERR: Bad media topology for: /dev/media11
ERR: Bad media topology for: /dev/media12
ERR: Bad media topology for: /dev/media13
ERR: Bad media topology for: /dev/media14
ERR: Bad media topology for: /dev/media15
W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: RK_AIQ_SENSOR_HDR_LINE_MODE_STAGGER for rk_aiq_sensor_hdr_line_mode_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: AECV2_STRATEGY_MODE_LOWLIGHT_PRIOR for CalibDb_AeStrategyModeV2_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: AECV2_STRATEGY_MODE_LOWLIGHT_PRIOR for CalibDb_AeStrategyModeV2_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: AECV2_HDR_LONGFRMMODE_NORMAL for CalibDb_AeHdrLongFrmModeV2_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CALIB_AWB_HDR_FRAME_CHOOSE_MODE_AUTO for CalibDbV2_Awb_Hdr_Fr_Ch_Mode_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CALIB_AWB_BLK_STAT_MODE_ALL_V201 for CalibDbV2_Awb_Blk_Stat_V21_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_AF_MODE_CONTINUOUS_PICTURE for CalibDbV2_AF_MODE_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_FSS_ADAPTIVE_RANGE for CalibDbV2_Af_SS_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_ADAPTIVE_SEARCH for CalibDbV2_Af_SearchDir_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_ADAPTIVE_SEARCH for CalibDbV2_Af_SearchDir_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_FSS_ADAPTIVE_RANGE for CalibDbV2_Af_SS_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_ADAPTIVE_SEARCH for CalibDbV2_Af_SearchDir_t

W: iq_parser_v2/j2s/j2s.cpp(128) [j2s_enum_get_value]: unknown enum name: CalibDbV2_CAM_AFM_ADAPTIVE_SEARCH for CalibDbV2_Af_SearchDir_t

W: iq_parser_v2/j2s/j2s.cpp(895) [_j2s_json_to_obj]: missing len in json for dynamic array 'measiso'

ERR: rkaiq engine prepare failed !

`

Joshua-Riek, 2023-09-11:
I updated the camera software packages with commit d206c00d1af117b65d82f7cbf24077423c992c5a and enabled rkisp2 in the kernel. I have the kernel building now on my PPA. When it is finished update your system and enter the below commands to install the new rkaiq and rkisp packages.

```
sudo apt-get update
sudo apt-get upgrade -y
wget https://media.githubusercontent.com/media/Joshua-Riek/ubuntu-rockchip/main/packages/rkaiq/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb?download=true
wget https://media.githubusercontent.com/media/Joshua-Riek/ubuntu-rockchip/main/packages/rkaiq/camera_engine_rkaiq_rk3588_update_arm64.deb?download=true
wget https://media.githubusercontent.com/media/Joshua-Riek/ubuntu-rockchip/main/packages/rkaiq/camera_engine_rkisp-v2.2.0_arm64.deb?download=true
sudo dpkg -i camera_engine_rkaiq_rk3588_1.0.3_arm64.deb
sudo dpkg -i camera_engine_rkaiq_rk3588_update_arm64.deb
sudo dpkg -i camera_engine_rkisp-v2.2.0_arm64.deb
```

Joshua-Riek, 2024-06-24:
Closing see https://github.com/Joshua-Riek/ubuntu-rockchip/pull/690

closed 2024-06-24 by Joshua-Riek

zoldaten, 2026-02-18:
@lucas-muller-1998 hi ! im from radxa 5b writing )
try:
`gst-launch-1.0 v4l2src device=/dev/video11 ! video/x-raw, width=3840, height=2160, framerate=30/1 ! videoconvert ! autovideosink`
should work (just checked).
`autovideosink` should be. NV12 also OK in the pipe.

lucas-muller-1998, 2026-02-25:
hello @zoldaten 
i will give it a try and will reply with the result 
Thank you for your help