Joshua-Riek/ubuntu-rockchip #291: Rock Pi 5B: No sound in v1.21

sunarowicz, 2023-08-02:
Hello,

thank you for your Rock Pi 5B image. Unfortunately I have to report an issue in the latest Rock Pi 5B image. Right after flashing the image, booting up and applying the system updates there is no sound. Everything looks OK. Audio devices are available, sound level meter shows sound is passing through, but nothing is heard. If I change the SD card for another one with some other image, sound works. I checked and compared the pactl and dmesg outputs between those two images, but did not find any differences.

Any idea where the culprit could be hidden?

Used image: https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.21/ubuntu-22.04.2-preinstalled-desktop-arm64-rock-5b.img.xz
[image]

watou, 2023-08-02:
If it's a useful data point, I have a Rock5B running and rebooted to latest apt get upgrade kernel etc. and HDMI0 audio is working in that setup.

sunarowicz, 2023-08-02:
Did a test and can confirm that both HDMI0 and HDMI1 audios work, Only headphone output does not. I cannot test DP output.

Joshua-Riek, 2023-08-02:
I did a test with v1.19, and it also has this issue. I think there is a bug in the device tree.

sunarowicz, 2023-08-03:
Thx for confirming. Is there any possible solution?

Joshua-Riek, 2023-08-03:
I will need to dig deeper into this. I know it was working with the old 5.10.110 kernel.

sunarowicz, 2023-08-05:
Not sure if this is interesting, but I compared the boot messages related to es8316 device between Armbian where sound works and your Ubuntu distribution and found they differ in the first line, see below.

```
rock@rock-5b:~$ neofetch 
                  🭯                    rock@rock-5b 
                 ◢🭫◣                   ------------ 
                ◢◤ ◥◣                  OS: Armbian (23.08.420-monka) aarch64 
               ◢◤   ◥◣                 OS Mod: AfterburnerMOD WIP 
              ◢◤     ◥◣                Host: Radxa ROCK 5B 
             ◢◤  🭉🬹🬾  ◥◣               Kernel: 5.10.160-rk35xx 
            ◢◤   🭅█🭐   ◥◣              Uptime: 7 mins 
           ◢◤    ███    ◥◣             Packages: 2170 (dpkg), 11 (flatpak) 
          ◢◤    🭋███🭀    ◥◣            Shell: bash 5.1.16 
        ◢█◤     🭅███🭐     ◥█◣          Resolution: 2560x1440 
       ◢█◤ 🬏   🭄█████🭏   🬞 ◥█◣         DE: Plasma 5.24.7 
      ◢█◤ 🭋▌🭈🭆🭂███████🭍🭑🬽▐🭀 ◥█◣        WM: kwin 
     ◢█◤  🭅███████████████🭐  ◥█◣       Theme: Breeze [GTK2] 
    ◢█◤   ███🭞🬎███████🬎🭓███   ◥█◣      Terminal: konsole 
   🭮█◤    🭒🭡🭘  🭣🮂🮅🭫🮅🮂🭘  🭣🭖🭝    ◥█🭬     CPU: (8) @ 1.800GHz 
   ◢█            █ █            █◣     Memory: 1181MiB / 15970MiB 
  ◢██◣          ◢🭡 🭖◣          ◢██◣
 ◢████🭍🭑🬭🬭🬭🬭🬭🬭🭆🭂🭠🭗 🭢🭕🭍🭑🬭🬭🬭🬭🬭🬭🭆🭂████◣                           

rock@rock-5b:~$ sudo dmesg | grep 8316
[sudo] password for rock: 
[    7.831648] typec_fusb302 4-0022: Looking up vbus-supply from device tree
[    7.929948] es8316 7-0011: Failed to get IRQ 0: -22
[    7.954631] input: headset-keys as /devices/platform/es8316-sound/input/input5
[    7.955128] rk-multicodecs es8316-sound: ASoC: Property 'rockchip,audio-routing' does not exist or its length is not even
[    7.955134] rk-multicodecs es8316-sound: Audio routing invalid/unspecified
[    8.009847] input: rockchip-es8316 Headset as /devices/platform/es8316-sound/sound/card4/input6
[    8.126938]   #4: rockchip-es8316

===============================================================================================

tux@rockb:~$ neofetch 
            .-/+oossssoo+/-.               tux@rockb 
        `:+ssssssssssssssssss+:`           --------- 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 22.04.2 LTS aarch64 
    .ossssssssssssssssssdMMMNysssso.       Host: Radxa ROCK 5B 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 5.10.160-rockchip 
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 1 min 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 1943 (dpkg) 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 5.1.16 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 2560x1440 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   DE: GNOME 42.9 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   WM: Mutter 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   WM Theme: Adwaita 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Theme: Yaru [GTK2/3] 
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/    Icons: Yaru [GTK2/3] 
  +sssssssssdmydMMMMMMMMddddyssssssss+     Terminal: gnome-terminal 
   /ssssssssssshdmNNNNmyNMMMMhssssss/      CPU: (8) @ 1.800GHz 
    .ossssssssssssssssssdMMMNysssso.       Memory: 925MiB / 15967MiB 
      -+sssssssssssssssssyyyssss+-
        `:+ssssssssssssssssss+:`                                   
            .-/+oossssoo+/-.                                       

tux@rockb:~$ sudo dmesg | grep es8316
[    7.024109] es8316 7-0011: Failed to get IRQ 0: -22
[    7.055095] input: headset-keys as /devices/platform/es8316-sound/input/input5
[    7.055305] rk-multicodecs es8316-sound: ASoC: Property 'rockchip,audio-routing' does not exist or its length is not even
[    7.055467] rk-multicodecs es8316-sound: Audio routing invalid/unspecified
[    7.111388] input: rockchip-es8316 Headset as /devices/platform/es8316-sound/sound/card4/input6
[    7.246311]   #4: rockchip-es8316
```

Joshua-Riek, 2023-08-05:
You used two different grep search patterns, so yes they are different.

Joshua-Riek, 2023-08-05:
Also, I tested with Armbian, and the sound does not work either.

sunarowicz, 2023-08-05:
Ugh, my fault, sorry.

Joshua-Riek, 2023-08-05:
No worries! I will try to debug it later today. Hopefully, I can find a solution, as this also affects the Rock 5A and CM5 from Radxa.

sunarowicz, 2023-08-05:
> Also, I tested with Armbian, and the sound does not work either.

Hmm this is strange, in that Armbian I post the dmesg output from, the headphone audio works to me.

Joshua-Riek, 2023-08-05:
Can you provide a link to the download?

Joshua-Riek, 2023-08-05:
I was able to get the audio working on Armbian after installing `libcanberra-pulse` package, which is weird.

sunarowicz, 2023-08-06:
> Can you provide a link to the download?

Sorry for responding so late, was not at the computer. Sure, here it is : https://monka.systemonachip.net/rock5b/Armbian_23.08.420-monka_rock-5b_jammy_legacy_5.10.160_gnome-desktop-AFM-1.5.img.xz

See here: https://forum.radxa.com/t/guide-best-option-for-armbian-afterburner-image-by-monkablyat/14552

sunarowicz, 2023-08-06:
> I was able to get the audio working on Armbian after installing `libcanberra-pulse` package, which is weird.

Hmm interesting. In the Armbian image I have (and sent the link to you a moment ago) the `libcanberra-pulse` package is not installed, see:

```
rock@rock-5b:~$ neofetch 
                  🭯                    rock@rock-5b 
                 ◢🭫◣                   ------------ 
                ◢◤ ◥◣                  OS: Armbian (23.08.420-monka) aarch64 
               ◢◤   ◥◣                 OS Mod: AfterburnerMOD WIP 
              ◢◤     ◥◣                Host: Radxa ROCK 5B 
             ◢◤  🭉🬹🬾  ◥◣               Kernel: 5.10.160-rk35xx 
            ◢◤   🭅█🭐   ◥◣              Uptime: 3 mins 
           ◢◤    ███    ◥◣             Packages: 2170 (dpkg), 11 (flatpak) 
          ◢◤    🭋███🭀    ◥◣            Shell: bash 5.1.16 
        ◢█◤     🭅███🭐     ◥█◣          Resolution: 2560x1440 
       ◢█◤ 🬏   🭄█████🭏   🬞 ◥█◣         DE: Plasma 5.24.7 
      ◢█◤ 🭋▌🭈🭆🭂███████🭍🭑🬽▐🭀 ◥█◣        WM: kwin 
     ◢█◤  🭅███████████████🭐  ◥█◣       Theme: Breeze [GTK2] 
    ◢█◤   ███🭞🬎███████🬎🭓███   ◥█◣      Terminal: konsole 
   🭮█◤    🭒🭡🭘  🭣🮂🮅🭫🮅🮂🭘  🭣🭖🭝    ◥█🭬     CPU: (8) @ 1.800GHz 
   ◢█            █ █            █◣     Memory: 2059MiB / 15970MiB 
  ◢██◣          ◢🭡 🭖◣          ◢██◣
 ◢████🭍🭑🬭🬭🬭🬭🬭🬭🭆🭂🭠🭗 🭢🭕🭍🭑🬭🬭🬭🬭🬭🬭🭆🭂████◣                           

rock@rock-5b:~$ sudo nala search libcanberra-pulse
libcanberra-pulse 0.30-10ubuntu1.22.04.1 [Ubuntu/jammy main]
└── PulseAudio backend for libcanberra
```

Joshua-Riek, 2023-08-06:
I got the audio working! That took way too long, haha. I need to clean up my changes and do a little testing, I should have a fix posted in a few hours.

Joshua-Riek, 2023-08-08:
I just confirmed that with the commit [97c9230](https://github.com/Joshua-Riek/ubuntu-rockchip/commit/97c9230035fa45792b835769393fa6d2af5d1e49), the sound is working.

sunarowicz, 2023-08-08:
> I just confirmed that with the commit [97c9230](https://github.com/Joshua-Riek/ubuntu-rockchip/commit/97c9230035fa45792b835769393fa6d2af5d1e49), the sound is working.

Great news, thank you!

sunarowicz, 2023-08-12:
I tested the v1.22 on ROCK 5B and can confirm to you that it boots here with no issues and the headphone jack audio works. But on my main system with v1.21 and all updates applied still no audio from jack.:-) Is there anything I could check to provide you more info, or do I have to re-flash to v1.22?