Joshua-Riek/ubuntu-rockchip #295: Orange Pi 5 crashing when watching Youtube or glxgears

portavales, 2023-08-05:
Fresh install flashed into nvme:
https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.21/ubuntu-22.04.2-preinstalled-desktop-arm64-orangepi-5.img.xz

A similar problem happens with the installation on a sdcard. Though it takes longer for it to happen when it is using the sdcard (don't know why).

To reproduce: Open Youtube, notice that some videos show only a [image] (no image), but some videos work. The youtube shorts are the most often to show the green screen.
Might be something related to the codec.
glxgears also trigger the problem but sometimes takes a while... Or maybe glxgears is a coincidence...

Watch a few videos and then suddenly it will reboot without any explanation.
The rebooting seems to be related to a msg in the logs that says:
"libEGL warning: MESA-LOADER: failed to open rknpu:"

Here are the logs:
[journalctl.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12266571/journalctl.log)

Here is the section of the logs that are important for this issue:
Aug 05 10:09:25 orangepi systemd[1007]: Started Application launched by gnome-shell.
Aug 05 10:09:26 orangepi gnome-shell[2159]: (WW) Option "-listen" for file descriptors is deprecated
Aug 05 10:09:26 orangepi gnome-shell[2159]: Please use "-listenfd" instead.
Aug 05 10:09:26 orangepi gnome-shell[2159]: (WW) Option "-listen" for file descriptors is deprecated
Aug 05 10:09:26 orangepi gnome-shell[2159]: Please use "-listenfd" instead.
Aug 05 10:09:26 orangepi gnome-shell[2159]: libEGL warning: MESA-LOADER: failed to open rknpu: /usr/lib/dri/rknpu_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
Aug 05 10:09:26 orangepi gnome-shell[2159]: libEGL warning: MESA-LOADER: failed to open rknpu: /usr/lib/dri/rknpu_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
Aug 05 10:09:26 orangepi systemd[1007]: Reached target GNOME session X11 services.
Aug 05 10:09:26 orangepi systemd[1007]: Starting GNOME XSettings service...
Aug 05 10:09:26 orangepi gnome-shell[2166]: The XKEYBOARD keymap compiler (xkbcomp) reports:
Aug 05 10:09:26 orangepi gnome-shell[2166]: > Warning:          Unsupported maximum keycode 708, clipping.
Aug 05 10:09:26 orangepi gnome-shell[2166]: >                   X11 cannot support keycodes above 255.
Aug 05 10:09:26 orangepi gnome-shell[2166]: Errors from xkbcomp are not fatal to the X server
Aug 05 10:09:27 orangepi systemd[1007]: Started GNOME XSettings service.
Aug 05 10:09:27 orangepi systemd[1007]: Reached target GNOME session X11 services.
Aug 05 10:09:27 orangepi gnome-shell[1132]: ATK Bridge is disabled but a11y has already been enabled.
Aug 05 10:09:27 orangepi gnome-keyring-daemon[1026]: asked to register item /org/freedesktop/secrets/collection/Default_5fkeyring/1, but it's already registered
Aug 05 10:09:27 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100927.645945:ERROR:gpu_init.cc(523)] Passthrough is not supported, GL is egl, ANGLE is
Aug 05 10:09:27 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100927.648527:ERROR:gl_display.cc(504)] EGL Driver message (Error) eglQueryDeviceAttribEXT: eglQueryDeviceAttribEXT
Aug 05 10:09:27 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100927.648637:ERROR:gl_angle_util_vulkan.cc(189)] Failed to retrieve vkGetInstanceProcAddr
Aug 05 10:09:27 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100927.648704:ERROR:vulkan_instance.cc(91)] Failed to get vkGetInstanceProcAddr pointer from ANGLE.
Aug 05 10:09:27 orangepi chromium-browser.desktop[2207]: Failed to query video capabilities: Inappropriate ioctl for device
Aug 05 10:09:27 orangepi chromium-browser.desktop[2207]: Failed to query video capabilities: Inappropriate ioctl for device
Aug 05 10:09:34 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100934.344858:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
Aug 05 10:09:40 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100940.097709:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
Aug 05 10:09:42 orangepi chromium-browser.desktop[2207]: [2207:2207:0805/100942.405119:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!
Aug 05 10:09:44 orangepi tracker-extract[2073]: Could not connect to filesystem miner endpoint: Timeout was reached
Aug 05 10:09:44 orangepi systemd[1007]: tracker-extract-3.service: Main process exited, code=exited, status=1/FAILURE
Aug 05 10:09:44 orangepi systemd[1007]: tracker-extract-3.service: Failed with result 'exit-code'.
Aug 05 10:09:44 orangepi systemd[1007]: Failed to start Tracker metadata extractor.
Aug 05 10:09:45 orangepi systemd[1007]: tracker-extract-3.service: Scheduled restart job, restart counter is at 11.
Aug 05 10:09:45 orangepi systemd[1007]: Stopped Tracker metadata extractor.
Aug 05 10:09:45 orangepi systemd[1007]: Starting Tracker metadata extractor...

Joshua-Riek, 2023-08-05:
Please ensure you use a 5a 4v power supply; if you use one, get another from a different brand. The system should be very stable, and power issues commonly cause crashing or instability.

portavales, 2023-08-07:
Ah, so you think the instability could be caused by the increased power consumption from the increased video processing, which in turn causes it to turn off/reboot ?
I guess that is possible.

BTW, is there any way to monitor for the power consumption ? lmsensors doesn't seem to help...

And kudos to all your efforts, I can't believe how fast you reply to all comments and keep up with everything!
Let me know if I can help in any way.

Joshua-Riek, 2023-08-07:
I don't know enough about how power is managed on this platform. You can measure how many watts the system is using from the wall if you have the right tools.

But in general, it's hard to catch a singular error when there are power issues because so many things start to break and produce errors.

portavales, 2023-08-07:
Makes sense, lets close this issue for now. If I find out the exact cause I can add more info here.
I will get the proper power brick and check later. Thank you

closed 2023-08-07 by portavales

bagong, 2023-12-21:
It think the `tracker-extract-3` related issue always happens. I've installed the image many times now and always get the "system experienced an internal error" in the beginning. Usually one clicks it away (it's first run wizard after all, and I get this on many Ubuntu installs on not fully compatible machines, and one hopes for updates ;-) ), without unclicking "ignore error in future", and then the error isn't reported any more. I think this is easy to reproduce. I have the error report, but it's not possible to copy it or export it into an textfile, so it's hard for me to provide the data. Screenshot isn't a option either, because it's multiple screens long and not all data is visible without exposing it manually.

````