Joshua-Riek/ubuntu-rockchip #296: [Orange Pi 5B][Release 1.21] HDMI eARC support

florentin, 2023-08-05:
Hello all

According to the docs, Orange Pi 5B has HDMI 2.1 eARC audio support.

My OrangePi is connected to an USB Dac (Minidsp flex eight). My hope is to pass the TV's audio to OrangePi through HDMI eARC so the TV's audio is played through the DAC.

Is this possible to achieve? What steps can be taken? 

I was hoping that after connecting the TV there will be a sound input available (like hdmi in) in the settings but that's not the case. If I run something like `sudo cec-ctl -d /dev/cec0 --audio --monitor-all`, the TV detects an eARC device but after selecting it the sound output switches back to the TV speakers and I see a lot of `reason: unrecognized-op` in the `cec-ctl` output.

Thank you

cata02, 2023-08-11:
I was considering the OrangePI for the same use case: get multichannel sound from the TV through eARC, process it on linux then output as usual.

Have you tried another linux distro based on 6.2 (or .3) kernel? I think i remember seeing some PRs that added support for the RK chip there.

Joshua-Riek, 2023-08-11:
The 6.x kernel has many not working features for the RK3588 chipset, and it will be a long time before it's in a "user-friendly" state. I have been looking into the 6.x kernel recently, and it will take a lot of work.

cata02, 2023-08-11:
Do you see a path for earc support there?

Joshua-Riek, 2023-08-20:
I can’t get to this right now, and I do not have plans to investigate eARC. If anyone is able to get eARC working that would be great.

closed 2023-08-20 by Joshua-Riek