Joshua-Riek/ubuntu-rockchip #297: [Orange Pi 5-Plus][Release 1.21] docker macvlan can't run well

aisyun, 2023-08-05:
run well:
`docker run --restart always --name openwrt -d --network macnet --ip 192.168.199.252 --privileged sulinggg/openwrt:armv8 /sbin/init`

can't run well: 
`docker run --restart always --name openwrt -d --network macnet --ip 192.168.1.250 --privileged 
sulinggg/openwrt:armv8 /sbin/init`

I have changed to armbian or orangePi releases are test well.

Joshua-Riek, 2023-08-05:
I need more information here, simply saying that a docker command does not run well gives me little information on what the problem is.

aisyun, 2023-08-05:
docker run --restart always --name openwrt -d --network macnet --privileged sulinggg/openwrt:armv8 /sbin/init
0cd8a0cd3aaa14444ba1366d95eb9df93d3df2b88017c950ec1d529b231dec76
docker: Error response from daemon: failed to create the macvlan port: operation not supported.

aisyun, 2023-08-06:
> I need more information here, simply saying that a docker command does not run well gives me little information on what the problem is.

docker run --restart always --name openwrt -d --network macnet --privileged sulinggg/openwrt:armv8 /sbin/init
0cd8a0cd3aaa14444ba1366d95eb9df93d3df2b88017c950ec1d529b231dec76
docker: Error response from daemon: failed to create the macvlan port: operation not supported.

Joshua-Riek, 2023-08-06:
Try running your commands with sudo? The macvlan driver is enabled in the kernel.

aisyun, 2023-08-06:
> Try running your commands with sudo? The macvlan driver is enabled in the kernel.
thanks! it is runnig with modprobe the macvlan module.

closed 2023-08-11 by Joshua-Riek