Joshua-Riek/ubuntu-rockchip #298: ax210 loses its ip on rock-5b

iamryanchia, 2023-08-06:
Hi, thanks for your help to make the ax210 work on rock-5b.

I have noticed another issue recently. The device  disconnects from wifi sometimes. If checking with `ip addr` command, the wireless network card has no ip address. unfortunately, I haven't found any solutions to recover from this problem other than rebooting the device.

Do you have any suggestions to debugging? Thank you!

iamryanchia, 2023-08-06:
I found it is caused by my router reboot. Is there any way to reconnect after the disconnection?

Joshua-Riek, 2023-08-06:
You should be able to assign a static IP address if you login to your router and find the rock-5b device.

iamryanchia, 2023-08-07:
Yes, i will give it a try. Thank you!

closed 2023-08-20 by Joshua-Riek