Joshua-Riek/ubuntu-rockchip #299: Intel AX200 Bluetooth not working on Ubuntu 22.04 server

Electrocat01, 2023-08-08:
Hello,

I just bought an Orange Pi 5 Plus and installed Ubuntu 22.04 on it. I got a Intel WiFi 6 AX200 and the wifi is perfeclty working but the bluetooth isn't.

The card is detected in `lsusb` and `lspci` but `bluetoothctl list` returns nothing and `bluetoothctl devices` returns `No default controller available`.

I already installed the drivers from Intel (https://www.intel.com/content/www/us/en/support/articles/000005511/wireless.html) but nothing changed.

So I have some questions :
1. Why is my Bluetooth card not working ?
2. Do I need to install some drivers / packages ?
3. Do I need to install the desktop version instead of the server version ?

Thanks for your help in advance !

Pryam

Joshua-Riek, 2023-08-08:
Could you provide the dmesg logs? I assume that the Bluetooth firmware is not loading. 

You can type `dmesg > dmesg.txt` to put the logs into a text file, then attach it to your post here on GitHub.

Electrocat01, 2023-08-08:
Thanks for your answer, here's the log : [dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12292495/dmesg.txt)

Joshua-Riek, 2023-08-08:
I see the below error, it looks like the system can not find the firmware `ibt-20-1-3.sfi`. I don't have time right now to find it, but if you find this file and put it in the path `/usr/lib/firmware/intel` bluetooth should work after a reboot.
```
[   12.147126] bluetooth hci0: Direct firmware load for intel/ibt-20-1-3.sfi failed with error -2
[   12.147134] Bluetooth: hci0: Failed to load Intel firmware file (-2)
```

Electrocat01, 2023-08-08:
Okay, thanks I will search and notify you for progress

Electrocat01, 2023-08-08:
Update : 

I copied this file (https://github.com/pop-os/linux-firmware/blob/master/intel/ibt-20-1-3.sfi) to `/lib/firmware/intel/` and `/usr/lib/firmware/intel/` then rebooted and it works ! The contorller is now detected in `bluetoothctl` and discovered by other devices.

Thanks for your help !

Pryam

closed 2023-08-08 by Electrocat01