Joshua-Riek/ubuntu-rockchip #300: Hello Mr. Joshua, Not booting jammy images on rock 5b .

hemsinghb, 2023-08-11:
I tried all of rock-5b images on my rock-5b board , none of them are booting, only one cli version booted up but not identifying wifi (nmcli) at command prompt iteself. There is some issue with all those images ported for rock 5b. please do need ful, Regards,

hemsinghb@gmail.com

Joshua-Riek, 2023-08-11:
This could be due to updating the bl31 and ddr blobs in the bootloader. Did you ever flash the SPI bootloader on these boards? It might need to be cleared.

hemsinghb, 2023-08-12:
Yes, I cleared all spi booloader from the board and tried a fresh. Still not booting with new jammy xfci and cli etc. There is a bootloder issue on Rock 5b board with new images. Requested to solve the same please.

Joshua-Riek, 2023-08-12:
Are you using a 5v 4A power supply? Does the previous version work?

closed 2023-08-15 by Joshua-Riek

hemsinghb, 2023-08-16:
Im using 3 A 5V power supply  ( 5V/3A,9V/2A ) POWER SUPPLY UNIT SUPPLIED BY
radxa.

hemsinghb, 2023-08-16:
I'm using 3 A 5V power supply  ( 5V/3A,9V/2A ) power supply unit supplied by radxa.

hemsinghb, 2023-08-16:
Please give  a site address /URL (if possible ) a working / tested rock-5b image supporting Jammy 22.04  server image.  
... Regards, Thank you.

Joshua-Riek, 2023-08-16:
I have two Rock 5B, and they work fine. Use a fixed 5v 4a power supply, not the one from Radxa.

hemsinghb, 2023-08-17:
I have used 5v  $A, 5v 3.3A, 5v 10A 
I failed to install at bios itself for the image I have chosen: "rock-5b_ubuntu_jammy_xfce_b36.img.xz"

Joshua-Riek, 2023-08-17:
There is no bios and that image is from radxa. I will lock this issue as this is not reproducible, and you fail to take my recommendations.

hemsinghb, 2023-08-29:
The preinstalled image of rock-5b (jammy -22.04.3) is working perfectly now.
 Thanks to Mr. Joshua Reik. 
With regards.