Joshua-Riek/ubuntu-rockchip #304: HOME ASSISTANT SUPERVISOR

rusudanion, 2023-08-11:
Hello i try to install supervisor Home Assistant on docker container but does not works.Any idea ? Thankyou

Joshua-Riek, 2023-08-11:
Please provide more information on how you set up docker and what the problem is along with any error messages.

rusudanion, 2023-08-12:
hello i use this tutorial https://jamesachambers.com/migrate-home-assistant-to-orange-pi-5/

i have this error
root@OPI-5-PLUS:/home/daniel# dpkg -i homeassistant-supervised.deb
(Reading database ... 157990 files and directories currently installed.)
Preparing to unpack homeassistant-supervised.deb ...
[warn] 
[warn] If you want more control over your own system, run
[warn] Home Assistant as a VM or run Home Assistant Core
[warn] via a Docker container.
[warn] 
[error] Ubuntu 22.04.3 LTS is not supported!
dpkg: error processing archive homeassistant-supervised.deb (--install):
 new homeassistant-supervised package pre-installation script subprocess returned error exit status 1
[info] Undo divert on abort-install
No diversion 'diversion of /etc/NetworkManager/NetworkManager.conf to /etc/NetworkManager/NetworkManager.conf.real by homeassistant-supervised', none removed.
No diversion 'diversion of /etc/NetworkManager/system-connections/default to /etc/NetworkManager/system-connections/default.real by homeassistant-supervised', none removed.
No diversion 'diversion of /etc/docker/daemon.json to /etc/docker/daemon.json.real by homeassistant-supervised', none removed.
No diversion 'diversion of /etc/network/interfaces to /etc/network/interfaces.real by homeassistant-supervised', none removed.
[info] Resetting debconf selections
[info] Removal complete, due to the complexity of this installation method,
[info] you will need to manually remove the containers created by the supervisor
Errors were encountered while processing:
 homeassistant-supervised.deb
root@OPI-5-PLUS:/home/daniel#

closed 2023-08-15 by Joshua-Riek

Al3XKOoL, 2023-08-16:
[error] Ubuntu 22.04.3 LTS is not supported!

Need modify /etc/os-release or install with dpkg and add flag to ignore host OS. For me it works

rusudanion, 2023-08-17:
> [error] Ubuntu 22.04.3 LTS is not supported!
> 
> Need modify /etc/os-release or install with dpkg and add flag to ignore host OS. For me it works

donyou have any errore on home assistant interface ?can you give me the guide line for install like you.thankyou

Al3XKOoL, 2023-08-17:
I have no errors.

sudo BYPASS_OS_CHECK=true dpkg -i homeassistant-supervised.deb

Try it.

rusudanion, 2023-08-17:
> it gives me a "dependency problem" error can I have the guide to install everything from scratch. thanks

Al3XKOoL, 2023-08-22:
Working home assistant supervised on ubuntu rockchip.

All commands need root: `sudo su`
```
apt update
apt upgrade
```

```
apt install \
apparmor \
jq \
wget \
curl \
udisks2 \
libglib2.0-bin \
network-manager \
dbus \
lsb-release \
systemd-journal-remote -y
```

Switch to cgroup v1. Edit `/boot/firmware/ubuntuEnv.txt` and add `apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false` at the end of the first line.

Reboot the system: `reboot`
After reboot
`sudo su`

Install docker
`curl -fsSL get.docker.com | sh`

You need modify `/etc/os-release`
Only PRETTY_NAME:
`PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"`
[image]

Install os agent:
```
wget https://github.com/home-assistant/os-agent/releases/download/1.5.1/os-agent_1.5.1_linux_aarch64.deb
dpkg -i os-agent_1.5.1_linux_aarch64.deb
```

Then install supervisor deb package:
```
wget https://github.com/home-assistant/supervised-installer/releases/latest/download/homeassistant-supervised.deb
sudo BYPASS_OS_CHECK=true dpkg -i -ignore-depends=systemd-resolved ./homeassistant-supervised.deb
```
Select raspberry pi 4 arm64
Before run apt commands you need modify `/var/lib/dpkg/status` and delete `systemd-resolved`  
in the supervisor dependecies (you can search with crtl + w in nano editor)

[image]

Ignore warnings, cgroupv1 is working.

rusudanion, 2023-08-22:
hello thankyou i install with your guide and all works good but i still have error AppArmour on home assistant.any idea?

rusudanion, 2023-08-22:
> hello thankyou i install with your guide and all works good but i still have error AppArmour on home assistant.any idea?

> Working home assistant supervised on ubuntu rockchip.
>
> All commands need root: `sudo su`
>
> ```
> apt update
> apt upgrade
> ```
>
> ```
> apt install \
> apparmor \
> jq \
> wget \
> curl \
> udisks2 \
> libglib2.0-bin \ [...]
hello thankyou i install with your guide and all works good but i still have error AppArmour on home assistant.any idea,?

Al3XKOoL, 2023-08-23:
Add `apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false` to bootargs= (First line) the reboot.
[image]

If not work check apparmor status with systemctl

rusudanion, 2023-08-23:
> Add `apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false` to bootargs= (First line) the reboot. [image]
> 
> If not work check apparmor status with systemctl

hello all works great Thankyou,

i used other script for install home Assistance.deb ,your give me error.

rusudanion, 2023-08-24:
> Add `apparmor=1 security=apparmor systemd.unified_cgroup_hierarchy=false systemd.legacy_systemd_cgroup_controller=false` to bootargs= (First line) the reboot. [image]
> 
> If not work check apparmor status with systemctl

all works good i have only a problem when i try to update software from terminal

Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 homeassistant-supervised : Depends: systemd-resolved but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@OPI-5-PLUS:/home/daniel#

if i try sudo  --fix-broken install  this remove home assistant.

Is any solution for make update without remove home assistant ?

Al3XKOoL, 2023-08-24:
You are not follow full tutorial. Then get problems...

rusudanion, 2023-08-24:
> You are not follow full tutorial. Then get problems...

yes i follow ,i modify only  the line for install home assistant.deb
dpkg -i --ignore-depends=systemd-resolved ./homeassistant-supervised.deb ,because your give me error systemd
i use this

rusudanion, 2023-08-24:
> You are not follow full tutorial. Then get problems...
  hello this script Is not good give me error 

sudo BYPASS_OS_CHECK=true dpkg -i -ignore-depends=systemd-resolved ./homeassistant-supervised.deb
i found the error 

Mis One - in the script.now it's works

sudo BYPASS_OS_CHECK=true dpkg -i --ignore-depends=systemd-resolved ./homeassistant-supervised.deb

rusudanion, 2023-08-24:
all works good now ,no error .thankyou so much

Al3XKOoL, 2023-08-24:
> sudo BYPASS_OS_CHECK=true dpkg -i -ignore-depends=systemd-resolved ./homeassistant-supervised.deb i found the error
>
> Mis One - in the script.now it's works
>
> sudo BYPASS_OS_CHECK=true dpkg -i [...]

The problem is not the command to install supervised.

Before run apt commands you need modify `/var/lib/dpkg/status` and delete `systemd-resolved`  
in the supervisor dependecies (you can search with crtl + w in nano editor)

rusudanion, 2023-08-25:
> The problem is not the command to install supervised.
>
> Before run apt commands you need modify `/var/lib/dpkg/status` and delete `systemd-resolved` in the supervisor dependecies (you can search with [...]

all works now.thankyou

RabHanz, 2024-05-13:
With the release of the new Ubuntu 24.04 cgroup v1 is disabled by default and the file /boot/firmware/ubuntuEnv.txt doesn't exist. So, is there an updated method for this new Ubuntu release to install Home Assistant Supervised? Thanks in advance @Al3XKOoL

Al3XKOoL, 2024-05-13:
Hi.
You have all about in the wiki: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS

Btw when i installed the new version i found a bug with apparmor: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/730
Maybe is fixed now, i don't know.

Check if apparmor is starting "systemctl status apparmor.service"

Joshua-Riek, 2024-05-13:
apparmor is masked by default, I don't want to deal with all the errors and warnings it produces.

Al3XKOoL, 2024-05-13:
No problem, for now i can enable apparmor by disabling mentioned services. 
We need it because is a requeriment for home assistant supervised.

NicholasFlamy, 2024-05-19:
Main problem I have is that http://homeassistant.local doesn't work. Seems that mDNS isn't being advertised properly.

NicholasFlamy, 2024-05-19:
FYI I also tried http://ubuntu.local:8123 and http://orangepi.local:8123 (after changing the host name to orangepi).

NicholasFlamy, 2024-05-19:
And I know it's not my network or my client device because I have confirmed it working with my TrueNAS Scale server.