Joshua-Riek/ubuntu-rockchip #305: Please add support for 13855 camera

ndrewpj, 2023-08-12:
Hi! In the vendor's OS this camera is detected and supported. Tried "Cheeze" app but no output is provided.

https://aliexpress.ru/item/1005005328497702.html?spm=a2g2w.orderdetail.0.0.2db24aa6AlvlUH&sku_id=12000032639616492

Joshua-Riek, 2023-08-12:
This camera is already supported and working fine. Cheese will not work with it, just like on the vendor OS. You must use gstreamer as the vendor documentation shows.

ndrewpj, 2023-08-13:
Hi again! Can you tell which string turns on the camera in the ubuntuEnv.txt file found in boot/firmware?

I see the /dev/videoXX files but when I try any commands on the video0 file the console returns errors like "device /dev/video0" not found

Joshua-Riek, 2023-08-15:
Please see https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays.

closed 2023-08-15 by Joshua-Riek