Joshua-Riek/ubuntu-rockchip #306: HDMI sound not working on Orange Pi 5 Plus

Siliconthaumaturgy, 2023-08-12:
I am unable to get HDMI sound working on Orange Pi 5 Plus.

When I open AlsaMixer, I can change "Card", but no value is displayed for "Chip".

The only "Card" where volume levels show up is "rockchip, es8388."  Other options are "rockchip, dp0", "rockchip-hdmi0", "rockchip-hdmi1", "rockchip-hdmiin"; these only show a setting

Changes to alsamixer settings do not save after exiting. 

I tried uninstalling and reinstalling Alsa and related utility packages, but the issue was not resolved.

Here is a screenshot showing what appears when I enter AlsaMixer.  Apologies for taking a picture of my monitor instead of through SSH.
[image]

Joshua-Riek, 2023-08-12:
Hdmi sound has been working fine for a while. Please check your monitor and that you did not screw up any of the audio config settings.

Joshua-Riek, 2023-08-15:
HDMI sound is working fine for me, closing.

closed 2023-08-15 by Joshua-Riek