Joshua-Riek/ubuntu-rockchip #307: Qt V4L2 test Utility can not work at OrangePi 5 plus

rctzxy, 2023-08-13:
version:1.22
When I open qv4l2 to test the HDMI input interface, its display is as shown in the following picture:
[image]
The bottom right corner of the qv4l2 interface successfully displays the captured Frame and Fps.
I tested using the official image, and it worked fine.
I also noticed that the Graphics option in the About section displays "Unknown Graphics Controller / Mail-G610(Panfrost)". Will this have any impact on usage?

Joshua-Riek, 2023-08-13:
Try using x11 desktop manager?

rctzxy, 2023-08-13:
When I use Xorg, qv4l2 cannot start. Here is the terminal output:
[image]

rctzxy, 2023-11-23:
> Try using x11 desktop manager?

I attempted to temporarily add an environment variable in the terminal with the command "export QT_QPA_PLATFORM="xcb"", and then started qv4l2. It runs well in both Wayland and X11 environments.

I added QT_QPA_PLATFORM="xcb" in /etc/environment, but after restarting, when I checked with echo, I found that it reverted back to Wayland.

closed 2025-02-17 by github-actions[bot]