Joshua-Riek/ubuntu-rockchip #308: Rock 5: No sound in v1.21 ("reopened")

sunarowicz, 2023-08-14:
Hello,

I put new comment to #291, but you probably missed that as it is already closed. I didn't find the way how to reopen, therefore making this new issue. Below is the new comment:

`I tested the v1.22 on ROCK 5B and can confirm to you that it boots here with no issues and the headphone jack audio works. But on my main system with v1.21 and all updates applied still no audio from jack.:-) Is there anything I could check to provide you more info, or do I have to re-flash to v1.22?`

Joshua-Riek, 2023-08-15:
I'm sorry for not getting back to you sooner. I know the audio jack will still have an issue on v1.21. But I currently offer updates regarding kernel, firmware, and bootloader changes between releases, as these are the three major components.

My change to fix the audio was with a systemd service to load the correct audio config for Alsa. The proper way would be to patch Alsa itself and create a fork with the change until it's fixed upstream. This would be a lot of work on top of the packaging and coming up with a different solution that is appropriate for upstream.

sunarowicz, 2023-08-16:
If I understand you well, you fixed it for v1.22 by the alsa-audio-config.service, but not for v1.21. I checked my v1.21 and this service is missing here after updates really. This is OK, not problem at all. I just needed to know whether I have to upgrade to v1.22 or not. So I will, not problem.

Thank you for your support, I appreciate it much!

Joshua-Riek, 2023-08-16:
This would be correct. You need to update to v1.22. I don't have the free time to make this change available for previous versions.

closed 2023-08-19 by Joshua-Riek