Joshua-Riek/ubuntu-rockchip #309: No planes reported by libdrm, despite them being present in /sys/kernel?

joedrago, 2023-08-15:
Hello! -- I've been hacking on RPI4 for a while with some success, but I wanted to try similar code on one of these faster Orange Pi 5 devices (using your great distro) and I don't understand what I'm missing. I'm hoping I'm just failing to load a module or configure something at boot, but I'm pretty lost. I'd appreciate any insight you have.

I'm trying to use libdrm to query/manipulate KMS planes (as I've done on RPI4) using some fairly generic libdrm calls, and if I dump the list of planes available on this opi5 running your distro, I get:

```
jdrago@orangepi5:~/work/planes$ sudo cat /sys/kernel/debug/dri/0/state | grep "plane\["
plane[54]: Cluster0-win0
plane[68]: Esmart0-win0
plane[90]: Cluster1-win0
plane[104]: Esmart1-win0
plane[126]: Cluster2-win0
plane[140]: Esmart2-win0
plane[162]: Cluster3-win0
plane[176]: Esmart3-win0
```

However, if I run a really basic program which just asks how many planes are available on any of the /dev/dri cards, it doesn't return any of them:

```
jdrago@orangepi5:~/work/planes$ uname -a
Linux orangepi5 5.10.160-rk35xx #1 SMP Thu Jul 6 13:36:29 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
jdrago@orangepi5:~/work/planes$ cat /etc/os-release 
PRETTY_NAME="Armbian 23.08.0-trunk jammy"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04 (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.armbian.com"
SUPPORT_URL="https://forum.armbian.com"
BUG_REPORT_URL="https://www.armbian.com/bugs"
PRIVACY_POLICY_URL="https://www.armbian.com"
UBUNTU_CODENAME=jammy
ARMBIAN_PRETTY_NAME="Armbian 23.08.0-trunk jammy"
jdrago@orangepi5:~/work/planes$ cat planes.c 
#include <stdio.h>
#include <fcntl.h>
#include <xf86drm.h>
#include <xf86drmMode.h>

int main(int argc, char *argv[])
{
    char card[1024];
    for(int i = 0; i < 10; ++i) {
        sprintf(card, "/dev/dri/card%d", i);

        int fd = open(card, O_RDWR);
        if(fd) {
            drmModePlaneRes *res = drmModeGetPlaneResources(fd);
            if(res) {
                printf("[%s]: count_planes = %d\n", card, res->count_planes);
                drmModeFreePlaneResources(res);
            }
            drmClose(fd);
        }
    }
    return 0;
}

jdrago@orangepi5:~/work/planes$ gcc -I/usr/include/drm -o planes64 planes.c -ldrm  && ./planes64
[/dev/dri/card0]: count_planes = 0
jdrago@orangepi5:~/work/planes$ 
```

Am I missing a package or driver? Again, Thanks so much in advance.

Joshua-Riek, 2023-08-15:
It looks like you are using Armbian rather than my distro, the kernel and os-release show that clearly. 

But I guess you would need to look at the Rockchip documentation, mostly not in English. However, the current kernel is really an Android kernel with a lot of hacks from Rockchip. This is different from the kernel used on the RPI4 because of its Android origin. 

Documentation: https://gitlab.com/rk3588_linux/linux/bsp/docs

joedrago, 2023-08-15:
Oops, sorry, I have tried this with your distro as well, I was bouncing around on distros to see if any of them behaved to differentl. I will look at that documentation and see if I can get some traction there, thank you.

Joshua-Riek, 2023-08-20:
I can’t look into this right now and I do not have plans to do so. Therefore I will be closing this issue, additionally because this seems to be an edge use case.

closed 2023-08-20 by Joshua-Riek