Joshua-Riek/ubuntu-rockchip #310: Can the default LED mode be changed on Orange PI 5 plus?

CptKKKK, 2023-08-16:
Currently, the default mode of the green and blue LEDs on Orange PI 5 plus is `heartbeat`. In the Ubuntu release by OrangePi, it can be set to `none` (turn off) by the command `orangepi-config`. Since there is no `orangepi-config` command in this Ubuntu release, I wonder how to change the default LED mode directly.

Joshua-Riek, 2023-08-19:
I have some information about this in a previous discussion post, please take a look.
https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/156#discussioncomment-6276122
https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/156#discussioncomment-6276149
https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/156#discussioncomment-6385625

closed 2023-08-19 by Joshua-Riek

CPT-KK, 2023-08-20:
Thanks a lot. I managed to change the default mode of LEDs of orange pi 5 plus by the following steps.

1. Copy  the default dtbo file for disabling the LEDs out: `cp /boot/firmware/dtbs/overlays/orangepi-5-plus-disable-leds.dtbo ~/ `
2. Decompile the dtbo file: `dtc -I dtb -O dts -o ~/my.dts ~/orangepi-5-plus-disable-leds.dtbo`
3. Change the default LED mode: `vim ~/my.dts`
4. Recompile `my.dts` to a new dtbo file: `dtc -I dts -O dtb -o ~/new_leds.dtbo ~/my.dts`
5. Move `new.dtbo` to `/boot/firmware/dtbs/overlays/`
6. Add a new line `overlays=new_leds` in the overlays section in `/boot/firmware/ubuntuEnv.txt`