Joshua-Riek/ubuntu-rockchip #315: Orange Pi 5B Wifi module not found

raquibahmed, 2023-08-17:
Hi there,

Yesterday I had downloaded version 1.22 from the release for my Orange Pi 5B and everything went smooth for the Ubuntu Desktop. During installation I was asked to connect to my Wifi and everything worked out. Later, I had done a `apt update and dist-upgrade`, which worked but my Wifi module was not showing anymore. I could not connect to Wifi. Today I had downloaded the latest release 1.23, and during installation I was not asked to connect to Wifi and again Wifi module not found. Reverted back to 1.22 and I was able to connect to Wifi again. It seems like disabling the software update and any kind of upgrade works to keep the Wifi working on 1.22. I am not sure which Wifi module Orange Pi 5B has on board but I tried `sudo sed -i 's/overlays=/overlays=ap6275p /g' /boot/firmware/ubuntuEnv.txt` from the Wiki and that did not work either to get the Wifi module back on 1.23 or after upgrade from 1.22.

Joshua-Riek, 2023-08-17:
The opi5b does not use a wifi module and does not require an overlay.

raquibahmed, 2023-08-17:
Thanks for the reply. Any suggestions I can try in the meantime to solve the issue? Looking at the kernel changes in the latest release, it seems there were changes to wireless drivers but I am no expert so appreciate your guidance. Thank you

Joshua-Riek, 2023-08-19:
I found the issue and sent a PR to fix this. I will update the kernel once the PR is merged so WiFi will work again.

raquibahmed, 2023-08-19:
Thanks so much. Really appreciate it.

Joshua-Riek, 2023-08-20:
This has been fixed, and I updated the kernel. Users can update the kernel with apt and WiFi will be working once more.

Fix: https://github.com/Joshua-Riek/linux-rockchip/commit/69e97ca96216a5f3557a85cee0bdff0ef14ac5dc

closed 2023-08-20 by Joshua-Riek

TobiPeterG, 2023-09-16:
Hey there,

I have an Orange Pi 5, I updated my system and WIFI still isn't working for me. I tried the overlay fix, but that didn't fix it.
What can I do?

Joshua-Riek, 2023-09-16:
Are you using the official wifi module?

TobiPeterG, 2023-09-16:
> Are you using the official wifi module?

I could connect in the installer, so I think so 😅
However, in the installed OS, I wasn't able to connect to it anymore.

Joshua-Riek, 2023-09-16:
Ensure you use the Orange Pi 5 image, not the Orange Pi 5 B image.
Then, from a command line, enable the overlay for the wifi module with the below command and reboot your device.
```
sudo sed -i 's/overlays=/overlays=ap6275p /g' /boot/firmware/ubuntuEnv.txt
```

TobiPeterG, 2023-09-16:
> Ensure you use the Orange Pi 5 image, not the Orange Pi 5 B image.
> Then, from a command line, enable the overlay for the wifi module with the below command and reboot your device.
> ```
> sudo sed -i 's/overlays=/overlays=ap6275p /g' /boot/firmware/ubuntuEnv.txt
> ```

I'm pretty sure I did both of these.
How can I verify that I installed the right version?

tuftsanti, 2024-09-18:
/boot/firmware/ubuntuEnv.txt doesn't exist, should I create it? No Wifi for me either.