Joshua-Riek/ubuntu-rockchip #317: Tencent Aurora Box 5 Pro may be able to be launched through this project

lee86, 2023-08-18:
By decompiling the boot.img of the official upgrade package, we can obtain the dts file.

This box have added 8150 network cards and rtl8852be wireless network cards (supporting Bluetooth) ,The soc is RK3588s .

Perhaps we can add this device to the support range?

The dts and dtb file
[rk3588s-tencent-5p.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12377247/rk3588s-tencent-5p.zip)

[image]
[image]

I'm sorry for that I don't know how to compile the system. But I tried to replace the DTB of Orangepi-5b and modified some shell scripts, after compilation is completedand it really work

Joshua-Riek, 2023-08-19:
I don't want to support Android TV boxes right now. They have limited use and can be difficult to maintain.

closed 2023-08-19 by Joshua-Riek

avatar21, 2024-03-27:
> By decompiling the boot.img of the official upgrade package, we can obtain the dts file.
>
>
>
> This box have added 8150 network cards and rtl8852be wireless network cards (supporting Bluetooth) ,The [...]

Glad to know that, it's a great news. you may try pushing to armbian community？